hultenvp/solis-sensor #131: Problem with starting integration

triplexxx36, 2022-07-19:
Hello
I have a problem to start the soliscloud integration.
I downloaded the repository, completed the entry in the configuration.yaml file and in the logs I have the following entry:

"Logger: custom_components.solis.service
Source: custom_components / solis / service.py: 103
Integration: solis (documentation, issues)
First occurred: 15:56:27 (3 occurrences)
Last logged: 15:58:27

Failed to discover, scheduling retry. "

Below is an entry in the configuration.yaml file:

sensor:
   - platform: solis
     name: "Solis"
     portal_domain: "www.soliscloud.com:13333"
     portal_username: "xxxxxxx"
     portal_key_id: "API ID"
     portal_secret: "API Secret"
     portal_plant_id: "inverter serial number"

I got the API ID and API Secred from the SolisCloud Plant Monitoring API

Can I ask for help in finding the error ??

Regards
Andrew

hultenvp, 2022-07-19:
Hi @triplexxx36 

You're almost there, pretty sure this fixes your problem:
https://github.com/hultenvp/solis-sensor/discussions/71#discussioncomment-304846

triplexxx36, 2022-07-19:
Hello

And you were right. The problem was Plant ID. I copied from url and everything started.

Thank you very much and all the best

> Hi @triplexxx36
> 
> You're almost there, pretty sure this fixes your problem: [#71 (comment)](https://github.com/hultenvp/solis-sensor/discussions/71#discussioncomment-3046924)

hultenvp, 2022-07-20:
Great to hear and thanks for letting know.

Cheers