hultenvp/solis-sensor #132: Collection of data does not recover from a temporary API error

drsmarsden, 2022-08-01:
Latest version

2022-07-31 23:05:09 DEBUG (MainThread) [custom_components.solis.service] SMSM hour [23] min [5] energy [13.500000]
2022-07-31 23:05:09 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 2 minutes.
2022-07-31 23:07:09 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: 110F5018B140xxx
2022-07-31 23:07:19 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Error from URI (/v1/api/inveterDetail) : 
2022-07-31 23:07:19 INFO (MainThread) [custom_components.solis.soliscloud_api] Unable to fetch details for device with ID: 1308675217944550xxx
2022-07-31 23:07:29 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Error from URI (/v1/api/stationDetail) : 
2022-07-31 23:07:29 INFO (MainThread) [custom_components.solis.soliscloud_api] Unable to fetch details for Station with ID: 1298491919448755xxx
2022-07-31 23:07:29 DEBUG (MainThread) [custom_components.solis.ginlong_base] AttributeError, serial does not exist
  File "/config/custom_components/solis/ginlong_base.py", line 70, in __getattr__
  File "/config/custom_components/solis/service.py", line 158, in async_update
  File "/config/custom_components/solis/service.py", line 131, in update_devices
  File "/config/custom_components/solis/ginlong_base.py", line 73, in __getattr__

restarting HA fixes the problem

interesting failure was late on the last day of the month - Maintenance window ?

Stuart

morts1, 2022-08-06:
I'm seeing the same issue after receiving a message to say the SolisCloud was having a maintenance outage on 4th August, collection stopped just before midday on 5th.
I tried restarting HA but still not collection anything.

Debug logs are not showing anything interesting:

homeassistant    | 2022-08-06 09:00:32 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
homeassistant    | 2022-08-06 09:00:35 INFO (MainThread) [custom_components.solis.sensor] Scheduling discovery
homeassistant    | 2022-08-06 09:00:35 DEBUG (MainThread) [custom_components.solis.service] Scheduling discovery in 1 seconds.
homeassistant    | 2022-08-06 09:00:38 INFO (MainThread) [custom_components.solis.soliscloud_api] Login Successful!
homeassistant    | 2022-08-06 09:00:38 WARNING (MainThread) [custom_components.solis.service] Failed to discover, scheduling retry.
homeassistant    | 2022-08-06 09:00:38 DEBUG (MainThread) [custom_components.solis.service] Scheduling discovery in 60 seconds.
homeassistant    | 2022-08-06 09:01:38 WARNING (MainThread) [custom_components.solis.service] Failed to discover, scheduling retry.

morts1, 2022-08-06:
Settings > System > Restart now seems to of fixed the issue and I see all sensors are now subscribed

blongden, 2022-08-07:
👍 on this one - collection stopped on the 5th just before midday and didn't recover.

drsmarsden, 2022-08-07:
I am a retired, and not actually written any code professionally for 10+ years
while @hultenvp does a proper fix  (after taking a backup) I have added a try/catch to my local function async_update in 

config/custom_components/solis/service.py

   async def async_update(self, *_) -> int:
        """Update the data from Ginlong portal."""
        try:
            update = SCHEDULE_NOK
            # Login using username and password, but only every HRS_BETWEEN_LOGIN hours
            if await self._login():
                inverters = self._api.inverters
                if inverters is None:
                    return update
                for inverter_serial in inverters:
                    data = await self._api.fetch_inverter_data(inverter_serial)
                    if data is not None:
                        # And finally get the inverter details
                        update = SCHEDULE_OK
                        self._last_updated = datetime.now()
                        await self.update_devices(data)
                    else:
                        update = SCHEDULE_NOK
                        # Reset session and try to login again next time
                        await self._logout()
        except:
             _LOGGER.error("async_update failed")
        self.schedule_update(update)

-------------
HA restarts and collects data, I need to wait for the next API outage to see if it works, but probably cannot make the situation  any worse.

drsmarsden, 2022-08-07:
async def async_update(self, *_) -> int:
        """Update the data from Ginlong portal."""
        try:
            update = SCHEDULE_NOK
            # Login using username and password, but only every HRS_BETWEEN_LOGIN hours
            if await self._login():
                inverters = self._api.inverters
                if inverters is None:
                    return update
                for inverter_serial in inverters:
                    data = await self._api.fetch_inverter_data(inverter_serial)
                    if data is not None:
                        # And finally get the inverter details
                        update = SCHEDULE_OK
                        self._last_updated = datetime.now()
                        await self.update_devices(data)
                    else:
                        update = SCHEDULE_NOK
                        # Reset session and try to login again next time
                        await self._logout()
        except:
             _LOGGER.error("SMSM async_update failed")
        self.schedule_update(update)

        if self._logintime is not None:
            if (self._logintime + HRS_BETWEEN_LOGIN) < (datetime.now()):
                # Time to login again
                await self._logout()

        return update

 
with indentation?

hultenvp, 2022-08-07:
Thanks for the workaround @drsmarsden.

Sorry all, I'm on a long summer break at the moment. Will take a few weeks before I'm able to work on the fix.

If anyone has a fix then I can integrate a pr and make a new release. Just cannot test.

hultenvp, 2022-08-07:
Think this is a dup of #125

drsmarsden, 2022-08-07:
Yes I also think it is a dup of #125

I upgraded the firmware on my router, which generated an internet  outage , which I guess is similar to api error

with my try/catch system continued ok

2022-08-07 17:08:39.469 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 2 minutes.
2022-08-07 17:10:39.470 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: 110F5018B140xxx
2022-08-07 17:10:44.481 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Error from URI (/v1/api/inveterDetail) : Cannot connect to host www.soliscloud.com:13333 ssl:default [Try again]
2022-08-07 17:10:44.481 INFO (MainThread) [custom_components.solis.soliscloud_api] Unable to fetch details for device with ID: 1308675217944550xxx
2022-08-07 17:10:49.488 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Error from URI (/v1/api/stationDetail) : Cannot connect to host www.soliscloud.com:13333 ssl:default [Try again]
2022-08-07 17:10:49.488 INFO (MainThread) [custom_components.solis.soliscloud_api] Unable to fetch details for Station with ID: 1298491919448755xxx
2022-08-07 17:10:49.488 DEBUG (MainThread) [custom_components.solis.ginlong_base] AttributeError, serial does not exist
2022-08-07 17:10:49.488 ERROR (MainThread) [custom_components.solis.service] **************  async_update failed ****
2022-08-07 17:10:49.491 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 2 minutes.
2022-08-07 17:12:49.492 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: 110F5018B140xxx
2022-08-07 17:12:50.150 DEBUG (MainThread) [custom_components.solis.soliscloud_api] {'success': True, 'code': '0', 'msg': 'success'........

morts1, 2022-08-09:
Thanks @drsmarsden, I've added the try/catch and will monitor. It has failed the last two mornings @ ~1.30 strangely, so I should be able to confirm if it has fixed the issue tomorrow.

drsmarsden, 2022-08-10:
I have now had lots of occurrences  over the last week or so (in batches) of the try/catch being executed without issue

it also only retires at the normal polling interval, so will not put unnecessary load on the soliscloud servers.

Having been responsible for mission critical systems for years, I would be happy to release this now

drsmarsden, 2022-08-11:
I stopped getting data today, the API was working just returning the same generated power every time.
The soliscloud app also was showing no new data.
I removed and reconnected my ethernet cable into the logger and everything burst back into life

I have added an automation to email me if "Solis Inverter State" becomes 2

hultenvp, 2022-08-23:
The crash is solved and added robustness + logging to failed discovery.
Please open a new issue if problems persist.
Fixed in https://github.com/hultenvp/solis-sensor/releases/tag/v2.3.1

closed 2022-08-23 by hultenvp