hultenvp/solis-sensor #133: The plugin stopped reporting at 5th of august

DdenBraver, 2022-08-07:
The plugin stopped reporting for me on the 5th of august at 13:00 CET.
I tried to reinstall the plugin, and re-enter my configuration but nothing seems to help, it remains "Unavailable"
Restarting HA, or even rebooting the VM HA is running at did not resolve the issue.

Does anybody else have this issue?

[image]

_Originally posted by @DdenBraver in https://github.com/hultenvp/solis-sensor/issues/105#issuecomment-1207364116_

blongden, 2022-08-07:
Yes I have this issue.

merijnbrs, 2022-08-07:
Same issue here. 
Got a different error in logs:
[image]
Likely this is related to the most recent 2022-08 Home Assistant update. As I updated on the same day.

kongjudas, 2022-08-07:
I believe this happened before the 2022.8 update. I have no recording last week, and the update was Wednesday (in Norway).

roperstv, 2022-08-08:
Yes I did have this issue too that the OP is experiencing at core level 2022.08.1 until 
I patched the core to 2022.08.2 then bizzarly everything started working again...

DdenBraver, 2022-08-08:
> Yes I did have this issue too that the OP is experiencing at core level 2022.08.1 until I patched the core to 2022.08.2 then bizzarly everything started working again...

Thank you! I updated from 2022.08.1 to 2022.08.2 today, and everything started working again

DdenBraver, 2022-08-08:
Fix was to update to HA core 2922.08.2

merijnbrs, 2022-08-08:
> Same issue here. Got a different error in logs: [image] Likely this is related to the most recent 2022-08 Home Assistant update. As I updated on the same day.

This morning at around 7:30 in the morning I ran an update to 2022.08.2. But at 17:03 I again saw above error in the logs.
Also energy stopped working from the same time as the error.
So I dont think an update of HASS fixes it properly :)
Didn't try rebooting yet, but I figure it should just work without it (as it worked since the update)

drsmarsden, 2022-08-08:
Hi

Look at ticket 132 

I did a frig to the code a couple of days ago, my ha has been fine since , despite a couple of api errors

Requires simple code edit

Stuart

DdenBraver, 2022-08-09:
> This morning at around 7:30 in the morning I ran an update to 2022.08.2. But at 17:03 I again saw above error in the logs. Also energy stopped working from the same time as the error. So I dont [...]

I think you are right, I now see similar kind of behavior. However a restart if HA seems to solve this one.

hultenvp, 2022-08-21:
Working on it.

hultenvp, 2022-08-23:
The crash is solved and added robustness + logging to failed discovery.
Please open a new issue if problems persist.
Fixed in https://github.com/hultenvp/solis-sensor/releases/tag/v2.3.1

closed 2022-08-23 by hultenvp