hultenvp/solis-sensor #135: Integration won't work if Lattitude and Longitude aren't set

michaelenstone, 2022-08-12:
**Describe the bug**
When downloading plant data if the latitude or longitude are Null then the integration fails with a string to float error.  I fixed by commenting out those lines in code but couldn't work out how to make it fail more elegantly.  

I'm sure there is a bug with Solis cloud setting locations in UK/near zero longitude but would hope that this data is not key for integrating with HA.

t0nyb, 2022-08-16:
Hi

I’m in the uk. Where are you supposed to set the lat and long?

I'm struggling with this too. 

Thanks

Tony

michaelenstone, 2022-08-17:
The Solis cloud derives the lat and long from the plant info you set in the cloud interface, however it seems that the cloud doesn't like my address and so sets long to null.  

In order to correct I commented out the lines for lat and long in the soliscloud_api.py file in my home cloud install but you could also just set them manually in the same file.

hultenvp, 2022-08-23:
Hi @michaelenstone , @t0nyb 

I assume you are using SolisCloud?
Could you drop a log with the exception?

hultenvp, 2022-08-23:
Catching missing lat/long now for both m.ginlong.com and SolisCloud.com.

Result should be that lat/lon no longer exist as sensors if they aren't available.

closed 2022-08-23 by hultenvp