hultenvp/solis-sensor #136: Failed to discover, scheduling retry.

blackwhitebear8, 2022-08-18:
**Describe the bug**
i m getting yhis inside my logs and there are no sensors created:

2022-08-18 14:50:19.300 INFO (MainThread) [custom_components.solis.sensor] Scheduling discovery
2022-08-18 14:50:19.300 DEBUG (MainThread) [custom_components.solis.service] Scheduling discovery in 1 seconds.
2022-08-18 14:50:20.894 INFO (MainThread) [custom_components.solis.soliscloud_api] Login Successful!
2022-08-18 14:50:21.081 WARNING (MainThread) [custom_components.solis.service] Failed to discover, scheduling retry.
2022-08-18 14:50:21.083 DEBUG (MainThread) [custom_components.solis.service] Scheduling discovery in 60 seconds.

**Config**
added this to the config file

sensor:
  - platform: solis
    name: "My Solis Inverter"
    portal_domain: "www.soliscloud.com:13333"
    portal_username: "my username"
    portal_key_id: "my portal key"
    portal_secret: "portal secret"
    portal_plant_id: "plant id"

**Versions**
HA Version an deployment 2022.8.5
Integration version latest

**Debug logs**
Add to configuration.yaml:
```
logger:
  default: warning
  logs:
    custom_components.solis: debug
```
And attach the resulting log to the issue. If yo do not know where to find logs you missed an important step in the install of HACS. ;-)

**Screenshots**
If applicable, add screenshots to help explain your problem.

kyuzumaki, 2022-08-20:
Having the same issue getting 
WARNING (MainThread) [custom_components.solis.service] Failed to discover, scheduling retry.

I also get a config error when I enter the plantID 
portal_plant_id: "plant id"

Anything but an integer in the "plant id" causes it to report an invalid config file

kyuzumaki, 2022-08-21:
> Having the same issue getting WARNING (MainThread) [custom_components.solis.service] Failed to discover, scheduling retry.
> 
> I also get a config error when I enter the plantID portal_plant_id: "plant id"
> 
> Anything but an integer in the "plant id" causes it to report an invalid config file

So I fixed it and I think my issue is different from **blackwhitebear8** but I have now solved it by simply using the correct plant id and latest version of HA. 

Would be useful to add to the documents how to get plant id. 
1: open solis cloud and select the plant you want to use
2: look at the url it should say "https://www.soliscloud.com/#/station/stationdetail_1?id=" 
3: After id= is a long number which is the plant ID needed for the integration

Also thanks for the integration great work!

hultenvp, 2022-08-21:
I'm working on an update. It will solve the exception people are observing, but I have so far been unable to reproduce the failing discovery. The new version will provide more logging detail which hopefully helps fixing it.

blackwhitebear8, 2022-08-21:
> g on a

this is indeed what i did and fixed the problem

closed 2022-08-21 by blackwhitebear8