hultenvp/solis-sensor #137: No updates after installing integration

StoricU, 2022-08-21:
I installed the integration using HACS and set it up in the config file.

So far everything seemed to be working great. "All"(a lot) entities where created and the serialnumber of the inverter was showing correct.
However, the entities have never updated.
If I restart HA, all values are unknown for a few minutes, but it is then updated to the same values as before.
[image]

yaml input:
[image]

Logging into ginlong.com shoes the correct values:
[image]

Is there some basic config I have forgot?

Home Assistant 2022.8.6
Supervisor 2022.08.3
Operating System 8.4
Frontend 20220802.0 - latest

hultenvp, 2022-08-21:
Hi,

The config looks fine and you wouldn't get any sensors if it was wrong. I think the integration crashes on unexpected data. Please check the logs to see if there's an error or warning for Solis. 
If necessary run with debug on:
```
logger:
  default: warning
  logs:
    custom_components.solis: debug
```

StoricU, 2022-08-21:
Thanks for your quick reply.

I enabled debugging, but that did not give me much.
However I think I figured out the problem.

After I had the system installed, I switched from a wifi to a LAN adapter.
Browsing the solis site now I see that both adapters are registered with the same SN. I don't know why..

However I noticed that the calues in HA is the same as the last values the old adapter registered.
HA:
[image]
[image]
Solis:
[image]

Even though there is a delete button for the adapter, it seems that it doesn't work.
Will contact support and see if they can help me and then see what values that shows up in HA ☺

StoricU, 2022-08-21:
Correction..
The SN is offcourse the same, since it is the SN of the inverter.
My problem is that there are 2 loggers registered to the device.
The integration automatically fetches the data from the first one, that is not connected any more.

Have sent support a e-mail asking them to delete the old logger. Hopefully that would fix it! :)

hultenvp, 2022-08-22:
I indeed iterate over the list starting with the newest which is then overwritten by the older one.

If you know how to you can change line 251 in ginlong_api.py from:
```
for record in result_json['result']['paginationAjax']['data']:
```
to
```
for record in reversed(result_json['result']['paginationAjax']['data']):
```

That should fix it and always return the most recent inverter. If it works I'll take this along an an update I'm planning.

StoricU, 2022-08-22:
Hi,

Thanks for your reply.
However, I was able to delete the logger using the phone app. It was only in the browser the buttons did not work.
So, I'm not able to try the change, sorry.

Will close the issue.

closed 2022-08-22 by StoricU