hultenvp/solis-sensor #138: Remove DAILY_ON_GRID_ENERGY

hultenvp, 2022-08-23:
Soliscloud appears to have fixed the bug on the server side.

LucidityCrash, 2022-08-24:
Hate to say it but not on my api calls ... 
[image]

[image]

hultenvp, 2022-08-24:
Crap, will revert. Planning another fix, will take this along.

Thanks for the heads-up.

LucidityCrash, 2022-08-25:
> Crap, will revert. Planning another fix, will take this along.
> 
> Thanks for the heads-up. 

Thing that worries me is the guy that did the original pull request to take out the *10 must have done so for a reason.

hultenvp, 2022-08-25:
True.

@michaelenstone : It looks like what works for you does not work for @LucidityCrash.

I find it difficult to believe you're getting different responses from the same server, but this is Ginlong we're talking about so anything is possible. ;-)

Would you mind running @LucidityCrash's test tool (found [here](https://github.com/hultenvp/solis-sensor/tree/master/test)) and posting the results here?
Let's hope we can find a solution that works for both.

Thanks

hultenvp, 2022-08-26:
@michaelenstone?

michaelenstone, 2022-08-26:
Sorry for the delay.  I'm now not getting anything from Solis either in HA or with the tool above (valid login just zero for all values..)

michaelenstone, 2022-08-26:
Worked it out, I put the wrong id in but still getting nothing for energy today and this month.  Results:

{
    "code": "0",
    "data": {
        "acInType": 0,
        "acOutputType": 0,
        "apparentPower": 520.0,
        "apparentPowerStr": "VA",
        "backupMonthEnergy": 0.0,
        "backupMonthEnergyStr": "kWh",
        "backupTodayEnergy": 0.0,
        "backupTodayEnergyStr": "kWh",
        "backupTotalEnergy": 0.0,
        "backupTotalEnergyStr": "kWh",
        "backupYearEnergy": 0.0,
        "backupYearEnergyStr": "kWh",
        "backupYesterdayEnergy": 0.0,
        "backupYesterdayEnergyStr": "kWh",
        "batteryAcvSet": 0,
        "batteryAlarm": "0",
        "batteryCDEnableSet": 0,
        "batteryCDISet": 0,
        "batteryCDSet": 0,
        "batteryCMaxiSet": 0,
        "batteryCapacitySoc": 100.0,
        "batteryChargeEnergy": 0,
        "batteryChargeEnergyStr": "kWh",
        "batteryChargingCurrent": 0.0,
        "batteryChargingCurrentStr": "A",
        "batteryDMaxiSet": 0,
        "batteryDischargeEnergy": 0,
        "batteryDischargeEnergyStr": "kWh",
        "batteryDischargeLimiting": 50.0,
        "batteryDischargeLimitingStr": "A",
        "batteryFailureInformation01": "0",
        "batteryFailureInformation02": "0",
        "batteryFcvSet": 0,
        "batteryHealthSoh": 100.0,
        "batteryLaTemp": 0.0,
        "batteryMonthChargeEnergy": 0,
        "batteryMonthChargeEnergyStr": "kWh",
        "batteryMonthDischargeEnergy": 0,
        "batteryMonthDischargeEnergyStr": "kWh",
        "batteryOlvEnableSet": 0,
        "batteryOvpSet": 0,
        "batteryPower": 0.153,
        "batteryPowerBms": 0,
        "batteryPowerBmsStr": "kW",
        "batteryPowerFu": 0,
        "batteryPowerPec": "1",
        "batteryPowerStr": "kW",
        "batteryPowerZheng": 153.0,
        "batteryTodayChargeEnergy": 0.0,
        "batteryTodayChargeEnergyStr": "kWh",
        "batteryTodayDischargeEnergy": 0.0,
        "batteryTodayDischargeEnergyStr": "kWh",
        "batteryTotalChargeEnergy": 171.0,
        "batteryTotalChargeEnergyStr": "kWh",
        "batteryTotalDischargeEnergy": 233.0,
        "batteryTotalDischargeEnergyStr": "kWh",
        "batteryType": "1",
        "batteryUvpSet": 0,
        "batteryVoltage": 50.52,
        "batteryVoltageStr": "V",
        "batteryYearChargeEnergy": 0,
        "batteryYearChargeEnergyStr": "kWh",
        "batteryYearDischargeEnergy": 0,
        "batteryYearDischargeEnergyStr": "kWh",
        "batteryYesterdayChargeEnergy": 5.1,
[... 157 lines cut ...]
        "tempName": "inverter internal operating ambient temperature",
        "timeStr": "2022-08-26 15:56:42",
        "timeZone": 0.0,
        "totalLoadPower": 0,
        "totalLoadPowerStr": "kW",
        "uAc1": 236.6,
        "uAc1Str": "V",
        "uAc2": 0.0,
        "uAc2Str": "V",
        "uAc3": 0.0,
        "uAc3Str": "V",
        "uInitGnd": 0,
        "uInitGndStr": "V",
        "uPv1": 301.0,
        "uPv1Str": "V",
        "uPv2": 150.1,
        "upvTotal": 0,
        "upvTotalStr": "V",
        "userId": "",
        "version": "3C0032",
        "warningInfoData": 0
    },
    "msg": "success",
    "success": true
}

hultenvp, 2022-08-26:
There are indeed some inconsistencies.

"gridSellEnergy": 0,
"gridSellEnergyStr": "kWh",
"gridSellMonthEnergy": 0,
"gridSellMonthEnergyStr": "kWh",
"gridSellTodayEnergy": 0.0,
"gridSellTodayEnergyStr": "kWh",
"gridSellTotalEnergy": 314.0,
"gridSellTotalEnergyStr": "kWh",
"gridSellYearEnergy": 0,
"gridSellYearEnergyStr": "kWh",
"gridSellYesterdayEnergy": 8.6,
"gridSellYesterdayEnergyStr": "kWh",

It could be possible you did not sell energy today, but you sold yesterday, so the year figure can never be right.
But did you see figures on daily grid sell energy that were correct and did not require any *10 correction?

michaelenstone, 2022-08-26:
I had the correct figures for about 2 weeks, basically since I posted the
pull request. One day the grid sell energy was 10x bigger than the day
before. So I looked at your code, made the change and pushed it through.

I stopped getting data on the 24th in the afternoon.

I've now updated to the latest version and I'm getting some data but it's
inconsistent...

hultenvp, 2022-08-27:
I don't see how the changes in my update could have contributed and the output of the test script also indicates the issue is server side.
SolisCloud is still work in progress and this kind of confusing behaviour has been observed before. I can only assume their devops processes are not really state of the art yet.
If you want a stable experience I recommend you go to m.ginlong.com and make that account your primary (you can run them in parallel) until SolisCloud stabilizes. 

I'm going to see if I can make a yaml file to switch workarounds on/off. If not, then I propose I revert the change for now.

Thanks for your help.
Cheers

closed 2022-08-27 by hultenvp