hultenvp/solis-sensor #141: Entity is non-numeric: sensor.my_solis_inverter_ac_output_total_power

bartgabriels, 2022-08-23:
Since the last update of today (?), my gauges aren't working anymore.

The entity is non-numeric.

hultenvp, 2022-08-23:
@bartgabriels: Can you please provide debug logging? I do not see the issue in test or production.
I'll have a look once I have the log.

bartgabriels, 2022-08-23:
Logger: custom_components.solis.service
Source: custom_components/solis/service.py:103
Integration: solis ([documentation](https://github.com/hultenvp/solis-sensor/), [issues](https://github.com/hultenvp/solis-sensor/issues))
First occurred: 16:51:24 (1 occurrences)
Last logged: 16:51:24

Failed to discover, scheduling retry.

Where can I find better logs?

bartgabriels, 2022-08-23:
Rolled back to 3.2.0 and issue is gone.

hultenvp, 2022-08-23:
Add the following to your configuration.yaml and restart (with 2.3.1)
```
logger:
  default: warning
  logs:
    custom_components.solis: debug
```

It will provide you with more detailed logging.

Are you using SolisCloud or the old ginlong portal (m.ginlong.com)?

bartgabriels, 2022-08-23:
m.ginlong.com

Will update with logs.

bartgabriels, 2022-08-23:
2022-08-23 17:06:50.261 INFO (MainThread) [custom_components.solis.sensor] Scheduling discovery
2022-08-23 17:06:50.261 DEBUG (MainThread) [custom_components.solis.service] Scheduling discovery in 1 seconds.
2022-08-23 17:07:05.127 DEBUG (MainThread) [custom_components.solis.ginlong_api] Error:
2022-08-23 17:07:05.128 WARNING (MainThread) [custom_components.solis.service] Failed to discover, scheduling retry.
2022-08-23 17:07:05.138 DEBUG (MainThread) [custom_components.solis.service] Scheduling discovery in 60 seconds.

hultenvp, 2022-08-23:
Odd, login seems to fail on timeout (log is admittedly a bit vague), so something prevents you from communicating with the server.
The question is why, because I did not change anything in that part of the code. Will dive into it.

hultenvp, 2022-08-23:
What version of HA are you on?

hultenvp, 2022-08-23:
I've released v2.3.2 which will give some more logging. Could you please run that version and post the debug log output?

bartgabriels, 2022-08-23:
It works now with the new version ...

bartgabriels, 2022-08-23:
Gauge is still not updated.

[image]
[image]

bartgabriels, 2022-08-23:
2022-08-23 18:19:06.022 INFO (MainThread) [custom_components.solis.ginlong_api] Login Successful!
2022-08-23 18:19:07.615 DEBUG (MainThread) [custom_components.solis.ginlong_api] Fetching data for serial: 0003700171073020
2022-08-23 18:19:08.562 DEBUG (MainThread) [custom_components.solis.ginlong_api] Fetching data for serial: 000370017107302
2022-08-23 18:19:08.563 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 1 minutes.
2022-08-23 18:20:18.568 DEBUG (MainThread) [custom_components.solis.ginlong_api] Error: Exception: <class 'asyncio.exceptions.TimeoutError'>
2022-08-23 18:20:18.570 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 1 minutes.
2022-08-23 18:21:19.280 INFO (MainThread) [custom_components.solis.ginlong_api] Login Successful!
2022-08-23 18:21:20.416 DEBUG (MainThread) [custom_components.solis.ginlong_api] Fetching data for serial: 0003700171073020
2022-08-23 18:21:23.537 DEBUG (MainThread) [custom_components.solis.ginlong_api] Fetching data for serial: 000370017107302
2022-08-23 18:21:23.539 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 1 minutes.

hultenvp, 2022-08-23:
Do you have 2 inverters with almost identical serial numbers? One has a zero more.
I did fix a scenario where the same Inverter is registered multiple times, but only one is active. Maybe there's something wrong in that part of the code.
> [custom_components.solis.ginlong_api] Fetching data for serial: 000370017107302**0**
> [custom_components.solis.ginlong_api] Fetching data for serial: 000370017107302

bartgabriels, 2022-08-23:
Must have been there since they installed it ... Removed it now (it contained no data):

2022-08-23 19:10:58.835 INFO (MainThread) [custom_components.solis.ginlong_api] Login Successful!
2022-08-23 19:10:59.111 DEBUG (MainThread) [custom_components.solis.ginlong_api] Fetching data for serial: 000370017107302
2022-08-23 19:10:59.465 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 2 minutes.

bartgabriels, 2022-08-23:
And then, the issue is gone:
[image]
[image]

Thanks

hultenvp, 2022-08-23:
Great to see it is gone now. I'll see if I can find a different way to ignore obsolete inverters. Apparently what solves it for one, creates an issue for the other.

Cheers

closed 2022-08-27 by hultenvp