hultenvp/solis-sensor #157: Battery percentage random spikes

kyuzumaki, 2022-09-14:
**Describe the bug**
I occasionally get a battery charge reading of 18,433%. I've had 4 of these readings in the last week. This unfortunatley messes up the graphs totally as its so far off the normal range. It's always the same value so maybe just some sort of rejection for values that are out of range would sort it?

**Config**
Your anonymized config entry

**Versions**
Using the Latest version 2.3.4

hultenvp, 2022-09-15:
Hi @kyuzumaki,

The integration just passes on the data that it retrieves from the server. It is very unlikely that the integration messes up the data, so without any additional info I have to assume it's a server problem.

If you want me to look into it, then please provide information so I can analyze as I do not have a hybrid system with battery myself:
- Log files
- Configuration
- Compare between server and HA showing anomalies in HA, but not on server.

hultenvp, 2022-10-21:
Close, no answer from submitter

closed 2022-10-21 by hultenvp