hultenvp/solis-sensor #158: Battery power value positive for both charge and discharge

corvus2606, 2022-09-16:
I can't see any value in the available sensors for my Solis inverter on the Solis Cloud that indicates the rate at which the battery is charging or discharging.

I can see the inverter_battery_power value, which does show the rate of charge/discharge, but the value is always positive.

spetrie13, 2022-09-16:
I suspect you would have to combine it with “Battery Status” that indicates Charge / Discharge

corvus2606, 2022-09-16:
unfortunately I only get battery power and remaining capacity as a measurement.  I can't see a battery status value in the API documentation either

lynxus, 2022-09-16:
if you figure this out, please let me know. Ive actually been trying to get that info all day myself and no closer :)

corvus2606, 2022-09-16:
I've hacked together a workaround for now until I can find something better.
I created a Derivative helper based on my battery remaining charge sensor.  Then used that value to determine if the battery power value should be negative with the below template sensor:
```

template:
  sensor:
    - name: "Battery Rate"
      state: >
            {% if states("sensor.battery_delta") | int > 0 -%}  
              {{ states("sensor.my_solis_inverter_battery_power") | int }}
            {%- else -%}
              {{ states("sensor.my_solis_inverter_battery_power") | int * -1}}
            {%- endif -%}
      state_class: measurement
      unit_of_measurement: W
      device_class: power
```

hultenvp, 2022-09-17:
I don't see anything explicit in the soliscloud API indicating charging or discharging status.

What you could do is check with the [test app](https://github.com/hultenvp/solis-sensor/blob/master/test/soliscloud_test.py) if battery current might be useful, but I'm afraid it's just like power only absolute. 

I don't have an inverter with battery myself, so I'm relying on the community here ;-). If you find how I can recreate battery state on soliscloud then I'm happy to add it.

corvus2606, 2022-09-17:
Yeah, I spent the better part of yesterday going through the API documentation and there's nothing explicit as you say.

Honestly, the derivative sensor with the template is proving pretty good.  there's about a 2 minute lag on it, but the solis api isn't realtime anyway.

I'll give the test app a go and try to find anything further that might indicate status... Either way, thank you for your work on this!  The total Charged/Discharged is accurate for the power dashboard, the battery flow is just to satisfy my need for instant gratification :)

lynxus, 2022-09-17:
So...
It looks like the "inverte" page doesnt help much.
However the ( "/v1/api/stationDetail" )  endpoint does..

Inverter just gives power.

But the stationDetail gives a negative value when outputting to the house..

[image]

So, i now use that endpoint to give me the "total" in/out . I guess some  math needs to be used if you query and inverter endpoint directly..

hultenvp, 2022-09-17:
Interesting! So if we know the total of the station then we should be able to figure out whether an individual inverter battery is charging or discharging. Just a matter of making sure the sum adds up.

hultenvp, 2022-09-18:
I'm going to see if I can use the info and make an improvement by adding a battery state.

corvus2606, 2022-09-26:
With @lynxus' comment above seeming to be the answer to where we can get the direction of flow.  Is this just a case of moving BAT_POWER: and BAT_POWER_STR: from the INVERTER_DETAIL endpoint into the PLANT_DETAIL one?

all of the other power sensors in hass seem to operate on negative meaning export and positive meaning import, so this would be in-line with general practise so far as I can tell.

I considered trying this, but I am not a python guy and don't have a test instance of hass available to mess with.

hultenvp, 2022-09-26:
Haha.

Well, you could exchange the inverter_detail values with the plant details if you only have one inverter/battery in your station.
Most people will have that, but stations can have multiple inverters registered and I can only assume that the BAT_POWER is then the sum of all inverters' BAT_POWER.
And since they have different precision (see -0.015 vs 0.01) it will be interesting to come up with a generic algorithm to figure out which sum of positive and negative inverter BAT_POWER values makes up the station BAT POWER.

hultenvp, 2022-09-26:
By the way, it should be as simple as to move the lines BAT_POWER and BAT_POWER_STR from INVERTER_DETAIL to PLANT_DETAIL.

liveware-problems, 2022-09-27:
I simply created a `template` sensor for this - whereby if the solar output is greater than the house consumption then it's a positive value (charging), and if the solar output is less than the house consumption then it's a negative value (discharging)

`
## Solis battery charge / discharge
      battery_power:
        friendly_name: Battery Power
        unit_of_measurement: W
        value_template: >
          {% if states('sensor.solis_ac_output_total_power')| int > states('sensor.solis_total_consumption_power')| int %}
          {% set battery_power = states('sensor.solis_battery_power')|int %}
          {% elif states('sensor.solis_ac_output_total_power')| int < states('sensor.solis_total_consumption_power')| int %}
          {% set battery_power = states('sensor.solis_battery_power')|int*-1 %}
          {% endif %}
          {{ battery_power }}
          `

dhoogfr, 2022-09-27:
It is strange that you see a positive value for both the charging and discharging.
Could it be something with the integration between your battery system and the inverter?

On my system (pylontech batteries), the solis_battery_power entity is negative when discharging.

[image]

corvus2606, 2022-09-27:
> It is strange that you see a positive value for both the charging and discharging. Could it be something with the integration between your battery system and the inverter?
> 
> On my system (pylontech batteries), the solis_battery_power entity is negative when discharging.
> 
> [image]

out of curiosity, are you using solis cloud, or the older m.ginlong.com portal?  I expect this will only affect you if you're on the solis cloud one.  It may be moot at this point, as the solis cloud api is now non-functional as of 4am this morning (hopefully by mistake)

spetrie13, 2022-09-27:
I raised a ticket some time ago and just received this message to my many
questions

 So we have an article which shows a way to access API, however we stopped
it because of GDPR

It would appear they may have shutdown the API access on GDPR grounds.

hultenvp, 2022-10-21:
Let's wait till they get back online with a new API version and see if the problem persists

jmason, 2022-11-23:
hey folks -- just looking in the current SolisCloud dashboard and if this is exposed through the API, this value looks like it could give an indication of charging vs discharging:

[image]

I'll try running the test script tomorrow (assuming my battery gets some use!) and see if that value goes negative when discharging...

jmason, 2022-11-23:
And here's what it looks like while discharging:

```
    "storageBatteryCurrent": -2.7,
    "storageBatteryCurrentStr": "A",
    "storageBatteryVoltage": 51.7,
    "storageBatteryVoltageStr": "V",
```

LucidityCrash, 2022-11-27:
this is really weird ... my battery power is signed (positive for charging and negative for discharging) and always has been as far as I can remember. (using soliscloud)

viking2010, 2022-11-27:
Mine is the same @LucidityCrash, negative for discharging and positive for charging. I am looking to create a sensor that says if it's zero, then battery state is idle, if it's -0.1 or lower then it's discharging, if it's greater than 0 then it's charging

jmason, 2022-11-28:
Yeah that's really weird. solis_battery_power (and the source data, "Battery Power" in the soliscloud.com UI) are always positive for me -- "Battery Current" goes positive/negative.

viking2010, 2022-11-29:
Just checked tonight and weirdly, now I'm getting a positive value for a discharge when it should be a negative one. Bit late for me to start investigating but will update tomorrow. Need to see what the api is spitting out in it's raw form.

LucidityCrash, 2022-11-30:
Curiouser and Curiouser,  mine is still negative.

viking2010, 2022-12-01:
ok, so tested today and if I look back at my history for the battery sensor, it clearly shows that it was going negative at one point. 

[image]

...but since sometime on the 28th, that's when it stopped providing the negative number and now shows it as positive when charging or discharging. Looking over the output of the .py test script, the number in there is positive too. As @lynxus mentioned further back in this thread, the Station Detail shows a negative number: "batteryPower": -0.375, whereas the inverter detail shows: "batteryPower": 0.375.

@LucidityCrash, if the inverter detail for you is providing a negative number, do you see the same in the station detail?

I presume we're all using the newer Solis Cloud site and not the original Ginlong one? Could that be the difference?

LucidityCrash, 2022-12-02:
StationDetail : 
       "batteryPower": -0.093,

InverterDetail ; 
       "batteryPower": -0.093,

viking2010, 2022-12-02:
Mind blown! 🤯

lynxus, 2022-12-02:
> @lynxus & @LucidityCrash just another thought, do you have the newer or older style Data Logging stick? I have recently swapped to the new one with the LED's on the front, and it occurred to me that the dates when my stats changed from showing the negative number was on the same day I swapped to the new logger. That's the only other change I can think of that could be why things are different. If not, I give up lol!!

I do. My install is very new.

I've found so many oddities within the Solis API it's painful.

LucidityCrash, 2022-12-02:
OOoooh possibly the cause ... you two are using the new logger ... I'm
using the older logger.

viking2010, 2022-12-02:
Bingo! That's odd then. Although I think from what I've been told/read/investigated, the data from the old logger goes via Solarman first, and there's some information that doesn't get pushed from Solarman to Solis, Eg the battery Over Discharge SOC and Force Charge SOC, and who knows what else is done differently!  The new data logger sends the data direct to Solis, but the downside is, it doesn't have a separate server or an open port that can be queried like the old one can. It's therefore not compatible with the SolarmanPV integration. The reason I've had to change the logger is because I want the remote control  functionality from the inverter via the Solis App/Portal and that doesn't seem to work with the old data logger. 

So seems we've fathomed out why the negative battery figure is showing as positive. Now all I need to do is figure out where to change the code to grab it from the station detail!

viking2010, 2022-12-03:
I'll take a look later today and let you know.

viking2010, 2022-12-04:
@hultenvp When running that script I get a NameError Date2 is not defined. There is a "Date" field defined, and when running with that, the output is as follows:

`POST /v1/api/inverterDay
Content-MD5: PblYr09bx4mcygX4khzXhw==
Content-Type: application/json
Date: Sun, 04 Dec 2022 11:29:07 GMT
Authorization: API 130xxxxxxxxxxxxxxxxxxxxxxxxxxxx=
Body：{"id":"13xxxxxxx","sn":"16xxxxxxxxx","money":"EUR","timezone":0,"time":"Sun, 04 Dec 2022 11:29:07 GMT"}

{
    "code": "1",
    "data": null,
    "msg": "\u6570\u636e\u5f02\u5e38 \u8bf7\u8054\u7cfb\u7ba1\u7406\u5458",
    "success": true
}
`

I've anonymised the above for obvious reasons lol!

finalbillybong, 2022-12-04:
> @hultenvp When running that script I get a NameError Date2 is not defined. There is a "Date" field defined, and when running with that, the output is as follows:
> 
> ```
> POST /v1/api/inverterDay
> Content-MD5: PblYr09bx4mcygX4khzXhw==
> Content-Type: application/json
> Date: Sun, 04 Dec 2022 11:29:07 GMT
> Authorization: API 130xxxxxxxxxxxxxxxxxxxxxxxxxxxx=
> Body：{"id":"13xxxxxxx","sn":"16xxxxxxxxx","money":"EUR","timezone":0,"time":"Sun, 04 Dec 2022 11:29:07 GMT"}
> 
> {
>     "code": "1",
>     "data": null,
>     "msg": "\u6570\u636e\u5f02\u5e38 \u8bf7\u8054\u7cfb\u7ba1\u7406\u5458",
>     "success": true
> }
> ```
> 
> I've anonymised the above for obvious reasons lol! Not sure if that's the result you expected or not.
> 
> UPDATE: I've just moved the following lines from Inverter detail to Plant Detail in Soliscloud_api.py and all is right with the world once again:
> 
> ```
> BAT_POWER:                     ['batteryPower', float, 3],
> BAT_POWER_STR:                 ['batteryPowerStr', str, None]
> ```
> 
> ....of course, if anyone else does this who has the new data logging stick, don't forget to comment them out in the Inverter Detail section.

I have made this change and am now seeing a negative value when my battery is discharging however it's off by a factor of 100. My Solis app shows 0.238kW but the sensor in Home Assistant shows -0.238W.

[image]
[image]

viking2010, 2022-12-05:
@finalbillybong yeah I get the same. Need to alter the units to kWh which you can either do manually from within the entity window.....

[image]

or I will look at altering the unit type in the relavent .py file

viking2010, 2022-12-05:
Just to update, in Const.py, I've updated the section starting "From homeassistant.const import" to include "POWER_KILO_WATT".  Then against 'batpower' changed the unit type to POWER_KILO_WATT. That then brings it through as kW (not kWh as I put in the screen shot above!). 

Hope that helps. I'm keeping a record of all the changes and may submit them as a pull request.....when I figure out exactly how to do that! lol :-D

viking2010, 2022-12-05:
> @hultenvp When running that script I get a NameError Date2 is not defined. There is a "Date" field defined, and when running with that, the output is as follows:
>
> ```
> POST /v1/api/inverterDay [...]

@hultenvp I've found why I got no information from the "/v1/api/inverterDay", it was to do with how the date format was coming through in the $body section. I've specified the date explicitly as "2022-12-04" and I now get over 37,000 lines appear! Judging by the Epoch timestamp, it's the data gathered at 5 min intervals throughout the whole of the 4th Dec.

hultenvp, 2022-12-05:
From the unicode it looks like the orignal message was something in Chinese.

@viking2010 Good that you managed to fix it. My idea was that if inverterDay would give back the battery power signed that it might be easy to get it out of that one. StationDetail is signed, but officially it's the sum of all batteries/inverters installed in the station, so will only substitute if you have only one. If you have more than one you need to figure out the sign for each individually installed battery in the station's scope. 

@finalbillybong I'll take along the fix in unit.

jmason, 2022-12-06:
@viking2010 if you send me the changed files, I'll create a pull request from them for you :)

viking2010, 2022-12-06:
@hultenvp 
The output from Inverter Day is below. Does this give you the info you need? If not, how do I tell if it's signed?  The date format in the soliscloud_test.py I have is set as `Date = now.strftime("%a, %d %b %Y %H:%M:%S GMT")` It didn't like this format and I know the code you originally sent me had Date2 in it, so I presume the format needs to be different as for me I could only get the output with the format of yyyy-mm-dd. 

I've found how to sort out a pull request so I can sort that after looking over the info below as I've added a few other entities not discussed here.

```
POST /v1/api/inverterDay
Content-MD5: LyYuxixka7CEONgWdUkCMQ==
Content-Type: application/json
Date: Tue, 06 Dec 2022 19:37:02 GMT
Authorization: API XXXREMOVEDXXX
Body：{"id":"XXXREMOVEDXXX","sn":"XXXREMOVEDXXX","money":"GBP","timezone":0,"time":"2022-08-04"}

{
    "code": "0",
    "data": [
        {
            "acOutputType": 1,
            "batteryCapacitySoc": 65.0,
            "batteryChargingCurrent": 74.0,
            "batteryDischargeLimiting": 74.0,
            "batteryHealthSoh": 100.0,
            "batteryPower": -377.0,
            "batteryTodayChargeEnergy": 0.0,
            "batteryTodayDischargeEnergy": 0.4,
            "batteryTotalChargeEnergy": 416.0,
            "batteryTotalDischargeEnergy": 467.0,
            "batteryType": 0.0,
            "batteryVoltage": 49.23,
            "bstteryCurrent": 6.0,
            "bypassAcCurrent": 0.0,
            "bypassAcVoltage": 0.0,
            "bypassLoadPower": 0.0,
            "dataTimestamp": "1659571251000",
            "dcBus": 0.0,
            "dcBusHalf": 0.0,
            "dcInputType": 1,
            "eToday": 0.0,
            "eTotal": 1975.0,
            "epmFailSafe": 0.0,
            "fac": 49.96,
            "familyLoadPower": 204.0,
            "gridPurchasedTodayEnergy": 0.0,
            "gridPurchasedTotalEnergy": 0,
            "gridSellTodayEnergy": 0.0,
            "gridSellTotalEnergy": 0,
            "homeLoadTodayEnergy": 0.4,
            "homeLoadTotalEnergy": 1242.0,
            "iAc1": 2.7,
            "iAc2": 0.0,
            "iAc3": 0.0,
            "iPv1": 0.0,
            "iPv10": 0.0,
            "iPv11": 0.0,
            "iPv12": 0.0,
            "iPv13": 0.0,
            "iPv14": 0.0,
            "iPv15": 0.0,
            "iPv16": 0.0,
            "iPv17": 0.0,
            "iPv18": 0.0,
            "iPv19": 0.0,
            "iPv2": 0.1,
            "iPv20": 0.0,
            "iPv21": 0.0,
            "iPv22": 0.0,
            "iPv23": 0.0,
            "iPv24": 0.0,
            "iPv25": 0.0,
            "iPv26": 0.0,
            "iPv27": 0.0,
            "iPv28": 0.0,
            "iPv29": 0.0,
            "iPv3": 0.0,
            "iPv30": 0.0,
            "iPv31": 0.0,
            "iPv32": 0.0,
            "iPv4": 0.0,
            "iPv5": 0.0,
            "iPv6": 0.0,
            "iPv7": 0.0,
            "iPv8": 0.0,
            "iPv9": 0.0,
            "inverterTemperature": 30.4,
            "llcBusVoltage": 0.0,
            "pSum": 18.0,
            "pac": 0.0,
            "pacPec": "0.001",
            "pacStr": "kW",
            "pepm": 0.0,
            "pepmSet": 0.0,
            "pfactorLimitSet": 0.0,
            "plimitSet": 0.0,
            "powerFactor": 1.0,
            "preactiveLimitSet": 0.0,
            "socChargingSet": 0.0,
            "socDischargeSet": 0.0,
            "state": 1,
            "storageBatteryCurrent": -7.6,
            "storageBatteryVoltage": 49.7,
            "time": "08:00:51",
            "timeStr": "2022-08-04 00:00:51",
            "timeZone": 8,
            "uAc1": 242.5,
            "uAc2": 0.0,
            "uAc3": 0.0,
            "uPv1": 2.4,
            "uPv10": 0.0,
            "uPv11": 0.0,
            "uPv12": 0.0,
            "uPv13": 0.0,
            "uPv14": 0.0,
            "uPv15": 0.0,
            "uPv16": 0.0,
            "uPv17": 0.0,
            "uPv18": 0.0,
            "uPv19": 0.0,
            "uPv2": 3.1,
            "uPv20": 0.0,
            "uPv21": 0.0,
            "uPv22": 0.0,
            "uPv23": 0.0,
            "uPv24": 0.0,
            "uPv25": 0.0,
            "uPv26": 0.0,
            "uPv27": 0.0,
            "uPv28": 0.0,
            "uPv29": 0.0,
            "uPv3": 0.0,
            "uPv30": 0.0,
            "uPv31": 0.0,
            "uPv32": 0.0,
            "uPv4": 0.0,
            "uPv5": 0.0,
            "uPv6": 0.0,
            "uPv7": 0.0,
            "uPv8": 0.0,
            "uPv9": 0.0,
            "uinitGnd": 0.0
        },
```

viking2010, 2022-12-06:
To add to the above, I figured out the data/time stamp formatting (still learning!!) but I think the only format the InverterDay will accept, (according to the API documentation) is YYYY-MM-DD only. It doesn't accept a time so you can't pull just an individual record out. It produces around 177k's worth of lines of output from an entire 24hrs.

jmason, 2022-12-06:
@hultenvp I can confirm it's signed in the `inverterDay` output.  For example, here's the battery charging:

```
      "storageBatteryCurrent": 3.1,
      "storageBatteryVoltage": 53,
      "time": "19:50:41",
      "timeStr": "2022-12-06 11:50:41",
```

and here's my battery discharging:

```
      "storageBatteryCurrent": -3.4,
      "storageBatteryVoltage": 51.6,
      "time": "22:40:41",
      "timeStr": "2022-12-06 14:40:41",
```

jmason, 2022-12-08:
I've added those 2 fields (storageBatteryCurrent and storageBatteryVoltage) for SolisCloud in this PR https://github.com/hultenvp/solis-sensor/pull/227

jmason, 2022-12-15:
https://github.com/hultenvp/solis-sensor/pull/227/files now contains a fix for this issue.

hultenvp, 2022-12-16:
Fixed in rel 3.2.0

closed 2022-12-16 by hultenvp

barthoefs, 2022-12-16:
Hello @hultenvp ,
[image]
After the update discharge is now negative and charge is positive.
You can see in the graph that during the night it was charging (negative block) and now it is using battery power (after update to 3.2.0) it is discharging (also negative).
Is this expected behaviour?

viking2010, 2022-12-16:
@barthoefs this is now the intended behaviour. Some people using the older data loggers already had positive and negative figures being displayed, but those with newer data loggers were only getting positive numbers whether it was charging or discharging. This has now been separated to provide a positive charge figure and negative discharge figure regardless of what data logger you have.

hultenvp, 2022-12-16:
Thanks @viking2010 for chipping in!