hultenvp/solis-sensor #159: No data in Energy Dashboard

GillesC, 2022-09-18:
**Describe the bug**
While the logs (in debug mode) indicate that it successfully subscribed to all sensors in the SolisCloud, it does not show anything in the dashboard. I added the integration yesterday, so it has been active for more than 2 hours.

**Config**
```yaml
sensor:
  - platform: solis
    name: "Solis Inverter"
    portal_domain: "www.soliscloud.com:13333"
    portal_username: "****"
    portal_key_id: "*****"
    portal_secret: "******"
    portal_plant_id: ****

logger:
  default: warning
  logs:
    custom_components.solis: debug
```

**Versions**
HA Version an deployment
HACS version
Integration version

**Debug logs**
```
2022-09-18 09:41:18.194 DEBUG (MainThread) [custom_components.solis.soliscloud_api] workarounds: {'correct_daily_on_grid_energy_enabled': True}
2022-09-18 09:41:18.195 INFO (MainThread) [custom_components.solis.sensor] Scheduling discovery
2022-09-18 09:41:18.195 DEBUG (MainThread) [custom_components.solis.service] Scheduling discovery in 1 seconds.
2022-09-18 09:41:20.613 INFO (MainThread) [custom_components.solis.soliscloud_api] Login Successful!
2022-09-18 09:41:20.728 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Found inverters: ['1031042226290163']
2022-09-18 09:41:20.728 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: 1031042226290163
2022-09-18 09:41:20.835 DEBUG (MainThread) [custom_components.solis.sensor] Creating inverterpowerstate (1031042226290163)
2022-09-18 09:41:20.835 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute powerState for inverter 1031042226290163
2022-09-18 09:41:20.836 DEBUG (MainThread) [custom_components.solis.sensor] Creating inverterstate (1031042226290163)
2022-09-18 09:41:20.836 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute state for inverter 1031042226290163
2022-09-18 09:41:20.836 DEBUG (MainThread) [custom_components.solis.sensor] Creating timestampmeasurement (1031042226290163)
2022-09-18 09:41:20.836 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute timestampUpdate for inverter 1031042226290163
2022-09-18 09:41:20.836 DEBUG (MainThread) [custom_components.solis.sensor] Creating temperature (1031042226290163)
2022-09-18 09:41:20.836 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute inverterTemperature for inverter 1031042226290163
2022-09-18 09:41:20.837 DEBUG (MainThread) [custom_components.solis.sensor] Creating dcinputvoltagepv1 (1031042226290163)
2022-09-18 09:41:20.837 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute dcVoltage1 for inverter 1031042226290163
2022-09-18 09:41:20.837 DEBUG (MainThread) [custom_components.solis.sensor] Creating dcinputvoltagepv2 (1031042226290163)
2022-09-18 09:41:20.838 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute dcVoltage2 for inverter 1031042226290163
2022-09-18 09:41:20.838 DEBUG (MainThread) [custom_components.solis.sensor] Creating dcinputcurrentpv1 (1031042226290163)
2022-09-18 09:41:20.838 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute dcCurrent1 for inverter 1031042226290163
2022-09-18 09:41:20.838 DEBUG (MainThread) [custom_components.solis.sensor] Creating dcinputcurrentpv2 (1031042226290163)
2022-09-18 09:41:20.838 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute dcCurrent2 for inverter 1031042226290163
2022-09-18 09:41:20.839 DEBUG (MainThread) [custom_components.solis.sensor] Creating dcinputpowerpv1 (1031042226290163)
2022-09-18 09:41:20.839 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute dcPower1 for inverter 1031042226290163
2022-09-18 09:41:20.839 DEBUG (MainThread) [custom_components.solis.sensor] Creating dcinputpowerpv2 (1031042226290163)
2022-09-18 09:41:20.839 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute dcPower2 for inverter 1031042226290163
2022-09-18 09:41:20.839 DEBUG (MainThread) [custom_components.solis.sensor] Creating acoutputvoltage1 (1031042226290163)
2022-09-18 09:41:20.839 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute acVoltage1 for inverter 1031042226290163
2022-09-18 09:41:20.840 DEBUG (MainThread) [custom_components.solis.sensor] Creating acoutputcurrent1 (1031042226290163)
2022-09-18 09:41:20.840 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute acCurrent1 for inverter 1031042226290163
2022-09-18 09:41:20.840 DEBUG (MainThread) [custom_components.solis.sensor] Creating actualpower (1031042226290163)
2022-09-18 09:41:20.840 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute acPower for inverter 1031042226290163
2022-09-18 09:41:20.840 DEBUG (MainThread) [custom_components.solis.sensor] Creating acfrequency (1031042226290163)
2022-09-18 09:41:20.841 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute acFrequency for inverter 1031042226290163
2022-09-18 09:41:20.842 DEBUG (MainThread) [custom_components.solis.sensor] Creating energytoday (1031042226290163)
2022-09-18 09:41:20.842 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute energyToday for inverter 1031042226290163
2022-09-18 09:41:20.842 DEBUG (MainThread) [custom_components.solis.sensor] Creating energythismonth (1031042226290163)
2022-09-18 09:41:20.842 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute energyThisMonth for inverter 1031042226290163
2022-09-18 09:41:20.842 DEBUG (MainThread) [custom_components.solis.sensor] Creating energythisyear (1031042226290163)
2022-09-18 09:41:20.842 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute energyThisYear for inverter 1031042226290163
2022-09-18 09:41:20.843 DEBUG (MainThread) [custom_components.solis.sensor] Creating energytotal (1031042226290163)
2022-09-18 09:41:20.843 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute energyTotalLife for inverter 1031042226290163
2022-09-18 09:41:20.843 DEBUG (MainThread) [custom_components.solis.sensor] Creating batpower (1031042226290163)
[... 16 lines cut ...]
2022-09-18 09:41:20.844 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute dailyEnergyUsed for inverter 1031042226290163
2022-09-18 09:41:20.844 DEBUG (MainThread) [custom_components.solis.sensor] Creating gridmonthlyenergypurchased (1031042226290163)
2022-09-18 09:41:20.844 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute monthlyEnergyPurchased for inverter 1031042226290163
2022-09-18 09:41:20.845 DEBUG (MainThread) [custom_components.solis.sensor] Creating gridyearlyenergypurchased (1031042226290163)
2022-09-18 09:41:20.845 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute yearlyEnergyPurchased for inverter 1031042226290163
2022-09-18 09:41:20.845 DEBUG (MainThread) [custom_components.solis.sensor] Creating gridtotalongridenergy (1031042226290163)
2022-09-18 09:41:20.845 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute totalOnGridEnergy for inverter 1031042226290163
2022-09-18 09:41:20.845 DEBUG (MainThread) [custom_components.solis.sensor] Creating gridpowergridtotalpower (1031042226290163)
2022-09-18 09:41:20.845 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute totalPower for inverter 1031042226290163
2022-09-18 09:41:20.845 DEBUG (MainThread) [custom_components.solis.sensor] Creating gridtotalconsumptionpower (1031042226290163)
2022-09-18 09:41:20.845 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute totalConsumptionPower for inverter 1031042226290163
2022-09-18 09:41:20.845 DEBUG (MainThread) [custom_components.solis.sensor] Creating gridtotalenergyused (1031042226290163)
2022-09-18 09:41:20.845 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute totalEnergyUsed for inverter 1031042226290163
2022-09-18 09:41:20.846 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 1 minutes.
2022-09-18 09:42:20.849 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: 1031042226290163
[... cut ...]
2022-09-18 09:44:21.071 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: 1031042226290163
2022-09-18 09:44:21.273 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 2 minutes.
2022-09-18 09:46:21.276 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: 1031042226290163
2022-09-18 09:46:21.507 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 2 minutes.
```

**Screenshots**
[image]
[image]

hultenvp, 2022-09-18:
Hi @GillesC,

What probably happened is that somehow you have still sensors from a previous configuration attempt registered in home assistant. If you look in the history you'll probably find a stale Solis.energy_today and an active Solis.energy_today_2 (similar for all sensors). Switch to the active sensor and delete the stale one. Optionally you can rename the sensor if you're like me and can't stand the _2 suffix :-)

GillesC, 2022-09-18:
Hi @hultenvp,

Thanks! I removed the sensor sources in the energy dashboard, rebooted, and it worked. So, I can't confirm if this was the case. Given that I didn't change the entities, my guess is that it was not a stale sensor.

I'll close the issue now.

closed 2022-09-18 by GillesC