hultenvp/solis-sensor #162: No Data after Soliscloud update AM 27/09

corvus2606, 2022-09-27:
After Soliscloud was updated early this morning, there is no data coming through on the sensors.  on a HASS restart, I am seeing the below errors:

**Config**
  - platform: solis
    name: "My Solis Inverter"
    portal_domain: "www.soliscloud.com:13333"
    portal_username: ******
    portal_key_id: ******
    portal_secret: !secret Solis
    portal_plant_id: ********

**Versions**
HA Docker install, 2022.9.7
HACS version: 1.27.2
Integration version: 2.3.4

**Debug logs**

2022-09-27 10:20:39.686 INFO (MainThread) [custom_components.solis.soliscloud_api] Login Successful!
2022-09-27 10:20:39.918 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Found inverters: ['************************']
2022-09-27 10:20:39.918 WARNING (MainThread) [custom_components.solis.service] Failed to discover, scheduling retry in 180 seconds.
2022-09-27 10:20:39.919 DEBUG (MainThread) [custom_components.solis.service] Scheduling discovery in 180 seconds.

**Screenshots**

damianharbr, 2022-09-27:
Exactly the same issue this morning

defeatedbythecat, 2022-09-27:
Same issue this morning

drsmarsden, 2022-09-27:
This is what I saw

2022-09-27 03:45:04.516 DEBUG (MainThread) [custom_components.solis.service] SMSM hour [3] min [45] energy [0.000000]
2022-09-27 03:45:04.516 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 2 minutes.
**2022-09-27 03:47:04.611 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Error from URI (/v1/api/addUser) : 0, message='Attempt to decode JSON with unexpected mimetype: text/html', url=URL('https://www.soliscloud.com:13333/v1/api/addUser')**
2022-09-27 03:47:04.611 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 1 minutes.
2022-09-27 03:48:04.716 INFO (MainThread) [custom_components.solis.soliscloud_api] Login Successful!

corvus2606, 2022-09-27:
so it seems they've probably made an undocumented change to their API, this is why i prefer to avoid cloud based services...

liveware-problems, 2022-09-27:
This was working for me before the update to Solis Cloud, and this morning there is no data. However, I've checked the apiManage page and it now shows Nonactivated whereas before I had an active key. Trying to re-register for a key now results in the 'No Authority' error discussed on the Solis Cloud comment thread:
[image]

defeatedbythecat, 2022-09-27:
Mine showing as active, but still having the issue above

corvus2606, 2022-09-27:
I'm getting the same error when trying to reactivate the API access

corvus2606, 2022-09-27:
I've logged a support ticket with their team, awaiting a response.

From what i can tell, the API has now been locked behind some extra permissions, even the API management option is missing from the basic settings page, we have to type the address directly.

Hopefully this is just an error and not a permanent move by them.

liveware-problems, 2022-09-27:
I also logged a ticket with Solis Support about this.

The issue with using m.ginlong.com is that I'm not sure the newer datalogger supports this platform - I think older hardware registered on ginlong.com and then migrated to soliscloud, whereas the newer S3-WiFI-ST stick seems to only register on soliscloud and there is no option to migrate from there to ginlong.com

I'll have a try anyway 😄

corvus2606, 2022-09-27:
this has been the prompt i needed to give in to my urge to avoid cloud based integrations.  I'll be going down the route of an RS435 ethernet bridge and monitoring over modbus.

I never was super fond of the idea of freely giving away all of my electricity usage habits to a chinese entity anyway.

We'll see what they say on the support ticket and I'll feed back here regardless

GillesC, 2022-09-27:
> this has been the prompt i needed to give in to my urge to avoid cloud based integrations. I'll be going down the route of an RS485 ethernet bridge and monitoring over modbus.
> 
> I never was super fond of the idea of freely giving away all of my electricity usage habits to a chinese entity anyway.
> 
> We'll see what they say on the support ticket and I'll feed back here regardless

I agree. Could you share the procedure when you are able to get this work?

abbamust, 2022-09-28:
Mine also have the same issue. 
But do this have anything to do with the upgrade to HA 2022.9.7?
Or is this a problem the SolisCloud have caused?

INFO (MainThread) [custom_components.solis.soliscloud_api] Login Successful!
DEBUG (MainThread) [custom_components.solis.soliscloud_api] Found inverters: ['***********', '***********']
WARNING (MainThread) [custom_components.solis.service] Failed to discover, scheduling retry in 60 seconds.
 DEBUG (MainThread) [custom_components.solis.service] Scheduling discovery in 60 seconds.

corvus2606, 2022-09-28:
this is not caused by Home assistant, this is a problem on the Solis Cloud side.  I've had indications that they may have withdrawn API access citing GDPR issues.  I'm awaiting confirmation of that

dhoogfr, 2022-09-28:
Curious what kind of GDPR issue that could be.
I mean, the data is still being send to Solis Cloud, the API only makes it available for the user via a different method than the website.
Unless they gather statistics on the API usage and additional approval is needed for that (or it isn't about GDPR, but about security...)

corvus2606, 2022-09-28:
yeah, until we get an official answer, it's all rumour and speculation unfortunately.  Either way, the end result for us is the same...

hultenvp, 2022-09-28:
> this has been the prompt i needed to give in to my urge to avoid cloud based integrations. I'll be going down the route of an RS485 ethernet bridge and monitoring over modbus.
>
> I never was super [...]

There was a link in one of the issues or in the discussion thread to an integration that would potentially work.

Regardless of what will happen to the SolisCloud I will continue to maintain this integration. Mostly because it's fun and a great learning experience ;-).
For now I'll monitor the situation and leave the ticket open, not much I can do to deal with a disabled/broken service.
Please post any new info anyone might have here.

corvus2606, 2022-09-28:
@hultenvp Of course, nothing against you or the work you've done, this integration would have been the perfect solution had solis chosen to be a reliable data provider.

Thank you very much for the work you've done on this!

abbamust, 2022-09-28:
It looks like the datalogger I have is not working with ginlong, so for now I am a little stuck.
Anyone tried to read from the new datalogges via IP adress?

Moondevil-ha, 2022-09-28:
I have a datalogger with serial starting "5A" which will not allow me to set up any local access. I was going to attempt it, but can see no way forward without building an MQTT2RS485 interface?

tobes49, 2022-09-28:
> I also logged a ticket with Solis Support about this.
>
> The issue with using m.ginlong.com is that I'm not sure the newer datalogger supports this platform - I think older hardware registered on [...]

I've got the same problem so fingers crossed they fix it soon.  
I too have the S3 stick and have looked at direct access but cant seem to find a username or password to let me access the stick (tried admin/admin also admin/S/N etc) - would anyone have this?

corvus2606, 2022-09-28:
> It looks like the datalogger I have is not working with ginlong, so for now I am a little stuck. Anyone tried to read from the new datalogges via IP adress?

the newer model of datalogger is locked in to Solis Cloud and also doesn't publish any data that can be read using the solismon integrations, you can't even scrape the data off the page thanks to ajax.  The only option with those is to replace them with an older model, or use an RS485 module and interface directly with modbus

corvus2606, 2022-09-28:
> I've got the same problem so fingers crossed they fix it soon. I too have the S3 stick and have looked at direct access but cant seem to find a username or password to let me access the stick (tried [...]

the username is 'admin' and the password is your wifi password.

but as i said above, it doesn't support rest api requests like the old ones did, so you can't use it to pull data

corvus2606, 2022-09-28:
> I have a datalogger with serial starting "5A" which will not allow me to set up any local access. I was going to attempt it, but can see no way forward without building an MQTT2RS485 interface?

I am waiting to receive a RS485 ethernet bridge.  you can then integrate with it using the modbus integration in home assistant without the need to have MQTT in the middle.

I'm working off a lot of the things in this thread: https://community.home-assistant.io/t/getting-data-from-solis-inverter/302189/70?u=corvus2606

spetrie13, 2022-09-28:
S3 stick.

Not found a way to get data from it.

Once configured and connected to your network, login with admin password is
your wifi password

rmhomecouk, 2022-09-28:
I was loving the simplicity of this integration but the outage has pushed me to stick an Ethernet bridge in place of the stick. So far so good (the bridge is a PE-11 unit which seems really solid). 

[image]

[image]

liveware-problems, 2022-09-28:
Is that an AC powered PE-11 you are using @rmhomecouk ? So the yellow cable is just the RS485 data connection? I'm looking at these options, but think the port on the inverter (as used by the stick) only supplies +5v DC:

Protoss-PE11-M is RS485 to Ethernet server, no wifi, power input 9V to 48VDC;

Protoss-PE11-H is RS485 to Ethernet server, no wifi, power input 100V to240VAC;

Moondevil-ha, 2022-09-28:
I am looking at something like this: Serial Port RS485 to WiFi Device Server Module Elfin-EW11 f TCP/IP Protocol Data: 
[https://bit.ly/3UIQh1I](url)
I'm ordering and going to get my thinking cap on.

corvus2606, 2022-09-28:
I ordered this unit: https://www.amazon.co.uk/dp/B08NW79RNK?psc=1&ref_=chk_typ_imgToDp

It takes 5v power from the port, so hopefully it'll be self contained.  I'll report back when it arrives

rmhomecouk, 2022-09-28:
> net server, no wifi, power input 9V to 48VDC;

Yeah I went for the 240VAC option as I didn't realise that I could has used the internal 5V from the invertor. got a spare 2way consumer unit that I'll wire up for it though. Yellow cable is indeed the data cable

so far so good - data flowing through and just sorting out some scaling issues.

liveware-problems, 2022-09-28:
Looks like a good find @corvus2606 - but a month for delivery! Isn't everyone used to next-day Prime? 😄

corvus2606, 2022-09-28:
> Looks like a good find @corvus2606 - but a month for delivery! Isn't everyone used to next-day Prime? 😄
I got it on prime, it just arrived and I ordered it about 2pm yesterday.  unfortunately the plug from that one guy on [ebay](https://www.ebay.co.uk/itm/234686080710) will be a little longer...

joernschumacher0001, 2022-09-28:
Just a note, you don't really need additional hardware to access the Modbus data - you can add your own server to send it to in the dongle setup page. I've got this https://community.home-assistant.io/t/getting-data-from-solis-inverter/302189 working with some adjustments (seems the data format is also different between the dongles and/or firmware versions).

RIA-Ingo, 2022-09-28:
> > Just a note, you don't really need additional hardware to access the Modbus data - you can add your own server to send it to in the dongle setup page. I've got this https://community.home-assistant.io/t/getting-data-from-solis-inverter/302189 working with some adjustments (seems the data format is also different between the dongles and/or firmware versions).
> 
> Unfortunately this is only true if you have a version of the stick pre-V3. V3 has removed all local access, hence everyone's consternation.

Unfortunatly on my Solis wifi stick i canot set a SERVER B IP Adresse to use this way

hescher, 2022-09-28:
I got an solis s3-wifi-st and am able to login in the webpage of the datalogger.
(Firmware version4B0030)
So temporarly I just get the values via rest sensors:
  - platform: rest
    resource: http://IP/inverter.cgi?t=0.8156536889125297
    authentication: basic
    username: admin
    password: wifipassword
    scan_interval: 10
    value_template: '{{ value.split(";")[4] }}'
    name: solaris_test_current_power
    unit_of_measurement: "W"
    device_class: power

The device password changed from the inital 123456789 to my wifi password. t=0.8156536889125297 is just a random number.
I use 6 rest sensors for: Inverter temprature, Current power, Yield today, Total yield, Alerts, Wifi-Quality.
The only downside is, only one update of current power every 5 minutes.

rmhomecouk, 2022-09-28:
I have the same stick - I took the cover off and connected to the internal headers which provides serial. It appears to be read only and only provides info on the wifi strength and available heap. There was also an option to upload a new firmware via ymodem. Maybe if we can get hold of a firmware bin we can modify it to provide server b again?

corvus2606, 2022-09-28:
> 

> * platform: rest
>   resource: http://IP/inverter.cgi?t=0.8156536889125297
>   authentication: basic
>   username: admin
>   password: wifipassword
>   scan_interval: 10
>   value_template: '{{ value.split(";")[4] }}'
>   name: solaris_test_current_power
>   unit_of_measurement: "W"
>   device_class: power

this is really handy as a stop gap to at least see what our solar panels are doing.  it's a pity it doesn't provide battery and other stats :(

Moondevil-ha, 2022-09-28:
> I got an solis s3-wifi-st and am able to login in the webpage of the datalogger. (Firmware 0001012F, Serial 5A12...) So temporarly I just get the values via rest sensors:
> 
> * platform: rest
>   resource: http://IP/inverter.cgi?t=0.8156536889125297
>   authentication: basic
>   username: admin
>   password: wifipassword
>   scan_interval: 10
>   value_template: '{{ value.split(";")[4] }}'
>   name: solaris_test_current_power
>   unit_of_measurement: "W"
>   device_class: power
> 
> The device password changed from the inital 123456789 to my wifi password. t=0.8156536889125297 is just a random number. I use 6 rest sensors for: Inverter temprature[3], Current power[4], Yield today[5], Total yield[6], Alerts[7], Wifi-Quality[8]. The only downside is, only one update of current power every 5 minutes.

Great find, great stopgap

corvus2606, 2022-09-28:
not sure if this is affecting everyone, but the yield today value is off by a factor of 10, It's telling me I produced 47kWh (i wish) when it is in-fact 4.7

corvus2606, 2022-09-29:
> I was loving the simplicity of this integration but the outage has pushed me to stick an Ethernet bridge in place of the stick. So far so good (the bridge is a PE-11 unit which seems really solid).
> 
> [image]
> 
> [image]

any chance you could share your configs for the modbus registers for HASS?

DdenBraver, 2022-09-29:
> this is really handy as a stop gap to at least see what our solar panels are doing. it's a pity it doesn't provide battery and other stats :(

Thank you for this. Its enough for me to use for my energie reporting and monitoring.
I will probably be using this as a local solution, until SolisCloud becomes more stable.

rmhomecouk, 2022-09-29:
> > I was loving the simplicity of this integration but the outage has pushed me to stick an Ethernet bridge in place of the stick. So far so good (the bridge is a PE-11 unit which seems really solid).
> > [image]
> > [image]
> 
> any chance you could share your configs for the modbus registers for HASS?

Certainly - Sharing is caring :)

I've just posted the files in here:

https://community.home-assistant.io/t/solis-inverter-modbus-integration/292553/54?u=rico365

roperstv, 2022-09-30:
Confirming my integration also stopped working from the 27th.
Will try Dden's temp fix until this is resolved, thank you.

up2datecyborg, 2022-09-30:
> 

Sounds very interesting, do you think it's possible to bridge the TX and RX cables and have at the same time the WIFI S3 data logger to still have data on soliscloud, and have a RS485 to ethernet/wifi device?

rmhomecouk, 2022-09-30:
[image]

rmhomecouk, 2022-09-30:
> @rmhomecouk 
> 
> 
> 
> Sounds very interesting, do you think it's possible to bridge the TX and RX cables and have at the same time the WIFI S3 data logger to still have data on soliscloud, and have a RS485 to ethernet/wifi device?

It might work but you'll end up with collisions in the data line if both devices send a query. I did this with the meter and removed the 120ohm resister on a rs485 dongle which was hooked to an arduino. I had decent readings most of the time but sometimes they were out of sync.

I've now decided just to sell the wifi stick and stay away from the Solis cloud.

hultenvp, 2022-10-01:
The comment of @LucidityCrash in the discussion thread allowed me to find at least a partial way around the current API limitations. Apparently only some API calls are blocked and those are not critical for the actual data retrieval.
People that do have a valid key and secret should now again be able to get data from SolisCloud. 
The fix is in 2.3.5 and is available on HACS.
If you have no valid key and secret (anymore) then sorry, this will not help. Let me know if you see crashes, I had to fix this on my "production" HA instance with limited means to test.
Keeping the issue open to monitor what happens to the API.

Dickovski01, 2022-10-01:
Working for me again :-)

jeroenbal, 2022-10-01:
> Working for me again :-)

Here also back online!

roperstv, 2022-10-01:
working for me too with the new update, thank you hultenvp!

liveware-problems, 2022-10-01:
Working again for me too, many thanks @hultenvp for this!

sabsteef, 2022-10-02:
still unstable. I just created a NR solution to read out the Wifi module on the Solis and update HA. but tnx for the work

usbrit, 2022-10-02:
Working again for me too, many thanks @hultenvp
<https://github.com/hultenvp> for this!

liveware-problems, 2022-10-03:
> still unstable. I just created a NR solution to read out the Wifi module on the Solis and update HA. but tnx for the work

Care to share @sabsteef ? Is it reading all values from SolisCloud, or just scraping the basic data from the stick itself?

LucidityCrash, 2022-10-03:
> > still unstable. I just created a NR solution to read out the Wifi module on the Solis and update HA. but tnx for the work
> 
> Care to share @sabsteef ? Is it reading all values from SolisCloud, or just scraping the basic data from the stick itself?

Depending on which datalogger you have you can use https://github.com/StephanJoubert/home_assistant_solarman, it may be able to talk direct to your datalogger and you can use that instead of this integration.  It has to be said this is what I've switched to even after putting in the effort to work out how the soliscloud api works.  I prefer not relying on cloud solutions and in particular the soliscloud API seems especially poor.

liveware-problems, 2022-10-03:
> Depending on which datalogger you have you can use https://github.com/StephanJoubert/home_assistant_solarman, it may be able to talk direct to your datalogger and you can use that instead of this [...]

I had a look at that a while ago, and eventually figured out my S3 WiFi stick just wasn't compatible with that approach :( I've purchased an RS485 to WiFi bridge now, so I can skip relying on the cloud too. Thanks for both your work on this integration and the comment!

LucidityCrash, 2022-10-03:
> Hi after two days, still having issues with talking direct to WiFi stick - current issue: there is no "today_battery_discharge" , there is "today_battery_charge".
> 
> I will switch back to the cloud when I can

is that from the solarman one I linked above ... I'm seeing both today_battery_charge and today_battery_discharge.  which parameter lookup file are you using and what inverter do you have ... you may need to tweek the config.

DominicStenzel, 2022-10-05:
Working for me also, thanks hultenvp!

drsmarsden, 2022-10-07:
> is that from the solarman one I linked above ... I'm seeing both today_battery_charge and today_battery_discharge. which parameter lookup file are you using and what inverter do you have ... you may [...]

Thanks I was using https://github.com/NosIreland/solismon3
I have fixed up the definition file and appears to be good now - thanks for your help

LucidityCrash, 2022-10-07:
Ah ... that is something completely different, more for setting values than
getting values :)

liveware-problems, 2022-10-20:
> I ordered this unit: https://www.amazon.co.uk/dp/B08NW79RNK?psc=1&ref_=chk_typ_imgToDp
> 
> It takes 5v power from the port, so hopefully it'll be self contained. I'll report back when it arrives

Did you get this up and running @corvus2606  ? Any issues with it, or straightforward replacement and modbus is working with quicker updates?

[27 more replies]

closed 2022-12-06 by hultenvp