hultenvp/solis-sensor #163: Integration not available

Dtrotmw, 2022-09-29:
I've installed solis-sensor via HACS and all the files are correct in my installation.  I've restarted HA (several times) but "New Integration" does not have Ginlong or Solis available.  I've got this in the home.assistant.log
2022-09-29 11:44:08.696 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solis which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2022-09-29 11:44:08.699 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration localtuya which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2022-09-29 11:44:08.701 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration myenergi which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2022-09-29 11:44:29.660 WARNING (MainThread) [custom_components.solis.service] Failed to discover, scheduling retry in 60 seconds.
2022-09-29 11:45:33.433 WARNING (MainThread) [custom_components.solis.ginlong_api] Unknown payload received

Moondevil-ha, 2022-09-29:
I'm not an api specialist, but there appear to have been significant changes made in the recent update, comparing the two API documents.

hultenvp, 2022-09-29:
See issue #162 . Closing as duplicate

closed 2022-09-29 by hultenvp