hultenvp/solis-sensor #166: crash after upgrade 2.3.5

drsmarsden, 2022-10-02:
looks like I get invalid/missing data

Stuart

2022-10-02 10:31:28.250 INFO (MainThread) [custom_components.solis.sensor] Scheduling discovery
2022-10-02 10:31:28.250 DEBUG (MainThread) [custom_components.solis.service] Scheduling discovery in 1 seconds.
2022-10-02 10:31:41.250 INFO (MainThread) [custom_components.solis.soliscloud_api] Login successful
2022-10-02 10:31:41.250 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Found inverters: ['110F5018B1400XX']
2022-10-02 10:31:41.250 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: 110F5018B1400XX
2022-10-02 10:31:41.652 DEBUG (MainThread) [custom_components.solis.soliscloud_api] {'success': True, 'code': '0', 'msg': 'success', 'data': {'fullHour': 0.0, 'fullHourStr': 'h', 'reactivePower': 0, 'apparentPower': 0, 'uInitGnd': 0, 'uInitGndStr': 'V', 'dcBus': 0, 'dcBusStr': 'V', 'dcBusHalf': 0, 'dcBusHalfStr': 'V', 'power': 0, 'powerStr': 'kWp', 'powerPec': '1', 'porwerPercent': 0, 'pac': 0, 'pacStr': 'kW', 'pacPec': '1', 'oneSelf': 0, 'eToday': 0, 'eTodayStr': 'kWh', 'eMonth': 0, 'eMonthStr': 'kWh', 'eYear': 0, 'eYearStr': 'kWh', 'eTotal': 0, 'eTotalStr': 'kWh', 'uPv1': 0, 'uPv1Str': 'V', 'iPv1': 0, 'iPv1Str': 'A', 'uPv2': 0, 'uPv2Str': 'V', 'iPv2': 0, 'iPv2Str': 'A', 'uPv3': 0, 'uPv3Str': 'V', 'iPv3': 0, 'iPv3Str': 'A', 'uPv4': 0, 'uPv4Str': 'V', 'iPv4': 0, 'iPv4Str': 'A', 'uPv5': 0, 'uPv5Str': 'V', 'iPv5': 0, 'iPv5Str': 'A', 'uPv6': 0, 'uPv6Str': 'V', 'iPv6': 0, 'iPv6Str': 'A', 'uPv7': 0, 'uPv7Str': 'V', 'iPv7': 0, 'iPv7Str': 'A', 'uPv8': 0, 'uPv8Str': 'V', 'iPv8': 0, 'iPv8Str': 'A', 'uPv9': 0, 'uPv9Str': 'V', 'iPv9': 0, 'iPv9Str': 'A', 'uPv10': 0, 'uPv10Str': 'V', 'iPv10': 0, 'iPv10Str': 'A', 'uPv11': 0, 'uPv11Str': 'V', 'iPv11': 0, 'iPv11Str': 'A', 'uPv12': 0, 'uPv12Str': 'V', 'iPv12': 0, 'iPv12Str': 'A', 'uPv13': 0, 'uPv13Str': 'V', 'iPv13': 0, 'iPv13Str': 'A', 'uPv14': 0, 'uPv14Str': 'V', 'iPv14': 0, 'iPv14Str': 'A', 'uPv15': 0, 'uPv15Str': 'V', 'iPv15': 0, 'iPv15Str': 'A', 'uPv16': 0, 'uPv16Str': 'V', 'iPv16': 0, 'iPv16Str': 'A', 'uPv17': 0, 'uPv17Str': 'V', 'iPv17': 0, 'iPv17Str': 'A', 'uPv18': 0, 'uPv18Str': 'V', 'iPv18': 0, 'iPv18Str': 'A', 'uPv19': 0, 'uPv19Str': 'V', 'iPv19': 0, 'iPv19Str': 'A', 'uPv20': 0, 'uPv20Str': 'V', 'iPv20': 0, 'iPv20Str': 'A', 'uPv21': 0, 'uPv21Str': 'V', 'iPv21': 0, 'iPv21Str': 'A', 'uPv22': 0, 'uPv22Str': 'V', 'iPv22': 0, 'iPv22Str': 'A', 'uPv23': 0, 'uPv23Str': 'V', 'iPv23': 0, 'iPv23Str': 'A', 'uPv24': 0, 'uPv24Str': 'V', 'iPv24': 0, 'iPv24Str': 'A', 'uPv25': 0, 'uPv25Str': 'V', 'iPv25': 0, 'iPv25Str': 'A', 'uPv26': 0, 'uPv26Str': 'V', 'iPv26': 0, 'iPv26Str': 'A', 'uPv27': 0, 'uPv27Str': 'V', 'iPv27': 0, 'iPv27Str': 'A', 'uPv28': 0, 'uPv28Str': 'V', 'iPv28': 0, 'iPv28Str': 'A', 'uPv29': 0, 'uPv29Str': 'V', 'iPv29': 0, 'iPv29Str': 'A', 'uPv30': 0, 'uPv30Str': 'V', 'iPv30': 0, 'iPv30Str': 'A', 'uPv31': 0, 'uPv31Str': 'V', 'iPv31': 0, 'iPv31Str': 'A', 'uPv32': 0, 'uPv32Str': 'V', 'iPv32': 0, 'iPv32Str': 'A', 'pow1': 0, 'pow1Str': 'W', 'pow2': 0, 'pow2Str': 'W', 'pow3': 0, 'pow3Str': 'W', 'pow4': 0, 'pow4Str': 'W', 'pow5': 0, 'pow5Str': 'W', 'pow6': 0, 'pow6Str': 'W', 'pow7': 0, 'pow7Str': 'W', 'pow8': 0, 'pow8Str': 'W', 'pow9': 0, 'pow9Str': 'W', 'pow10': 0, 'pow10Str': 'W', 'pow11': 0, 'pow11Str': 'W', 'pow12': 0, 'pow12Str': 'W', 'pow13': 0, 'pow13Str': 'W', 'pow14': 0, 'pow14Str': 'W', 'pow15': 0, 'pow15Str': 'W', 'pow16': 0, 'pow16Str': 'W', 'pow17': 0, 'pow17Str': 'W', 'pow18': 0, 'pow18Str': 'W', 'pow19': 0, 'pow19Str': 'W', 'pow20': 0, 'pow20Str': 'W', 'pow21': 0, 'pow21Str': 'W', 'pow22': 0, 'pow22Str': 'W', 'pow23': 0, 'pow23Str': 'W', 'pow24': 0, 'pow24Str': 'W', 'pow25': 0, 'pow25Str': 'W', 'pow26': 0, 'pow26Str': 'W', 'pow27': 0, 'pow27Str': 'W', 'pow28': 0, 'pow28Str': 'W', 'pow29': 0, 'pow29Str': 'W', 'pow30': 0, 'pow30Str': 'W', 'pow31': 0, 'pow31Str': 'W', 'pow32': 0, 'pow32Str': 'W', 'uAc1': 0, 'uAc1Str': 'V', 'iAc1': 0, 'iAc1Str': 'A', 'uAc2': 0, 'uAc2Str': 'V', 'iAc2': 0, 'iAc2Str': 'A', 'uAc3': 0, 'uAc3Str': 'V', 'iAc3': 0, 'iAc3Str': 'A', 'batteryDischargeEnergy': 0, 'batteryDischargeEnergyStr': 'kWh', 'batteryChargeEnergy': 0, 'batteryChargeEnergyStr': 'kWh', 'homeLoadEnergy': 0, 'homeLoadEnergyStr': 'kWh', 'gridPurchasedEnergy': 0, 'gridPurchasedEnergyStr': 'kWh', 'gridSellEnergy': 0, 'gridSellEnergyStr': 'kWh', 'facStr': 'Hz', 'batteryPower': 0, 'batteryPowerStr': 'kW', 'batteryPowerPec': '1', 'batteryPowerZheng': 0, 'batteryPowerFu': 0, 'storageBatteryVoltage': 0, 'storageBatteryVoltageStr': 'V', 'storageBatteryCurrent': 0, 'storageBatteryCurrentStr': 'A', 'batteryVoltage': 0, 'batteryVoltageStr': 'V', 'bstteryCurrent': 0, 'bstteryCurrentStr': 'A', 'batteryPowerBms': 0, 'batteryPowerBmsStr': 'kW', 'batteryChargingCurrent': 0, 'batteryChargingCurrentStr': 'A', 'batteryDischargeLimiting': 0, 'batteryDischargeLimitingStr': 'A', 'batteryTotalChargeEnergy': 0, 'batteryTotalChargeEnergyStr': 'kWh', 'batteryTodayChargeEnergy': 0, 'batteryTodayChargeEnergyStr': 'kWh', 'batteryMonthChargeEnergy': 0, 'batteryMonthChargeEnergyStr': 'kWh', 'batteryYearChargeEnergy': 0, 'batteryYearChargeEnergyStr': 'kWh', 'batteryYesterdayChargeEnergy': 0, 'batteryYesterdayChargeEnergyStr': 'kWh', 'batteryTotalDischargeEnergy': 0, 'batteryTotalDischargeEnergyStr': 'kWh', 'batteryTodayDischargeEnergy': 0, 'batteryTodayDischargeEnergyStr': 'kWh', 'batteryMonthDischargeEnergy': 0, 'batteryMonthDischargeEnergyStr': 'kWh', 'batteryYearDischargeEnergy': 0, 'batteryYearDischargeEnergyStr': 'kWh', 'batteryYesterdayDischargeEnergy': 0, 'batteryYesterdayDischargeEnergyStr': 'kWh', 'gridPurchasedTotalEnergy': 0, 'gridPurchasedTotalEnergyStr': 'kWh', 'gridPurchasedYearEnergy': 0, 'gridPurchasedYearEnergyStr': 'kWh', 'gridPurchasedMonthEnergy': 0, 'gridPurchasedMonthEnergyStr': 'kWh', 'gridPurchasedTodayEnergy': 0, 'gridPurchasedTodayEnergyStr': 'kWh', 'gridPurchasedYesterdayEnergy': 0, 'gridPurchasedYesterdayEnergyStr': 'kWh', 'gridSellTotalEnergy': 0, 'gridSellTotalEnergyStr': 'kWh', 'gridSellYearEnergy': 0, 'gridSellYearEnergyStr': 'kWh', 'gridSellMonthEnergy': 0, 'gridSellMonthEnergyStr': 'kWh', 'gridSellTodayEnergy': 0, 'gridSellTodayEnergyStr': 'kWh', 'gridSellYesterdayEnergy': 0, 'gridSellYesterdayEnergyStr': 'kWh', 'homeLoadTotalEnergy': 0, 'homeLoadTotalEnergyStr': 'kWh', 'homeLoadTodayEnergy': 0, 'homeLoadTodayEnergyStr': 'kWh', 'totalLoadPower': 0, 'totalLoadPowerStr': 'kW', 'homeLoadYesterdayEnergy': 0, 'homeLoadYesterdayEnergyStr': 'kWh', 'familyLoadPower': 0, 'familyLoadPowerStr': 'kW', 'familyLoadPercent': 0, 'homeGridYesterdayEnergy': 0, 'homeGridYesterdayEnergyStr': 'kWh', 'homeGridTodayEnergy': 0, 'homeGridTodayEnergyStr': 'kWh', 'homeGridMonthEnergy': 0, 'homeGridMonthEnergyStr': 'kWh', 'homeGridYearEnergy': 0, 'homeGridYearEnergyStr': 'kWh', 'homeGridTotalEnergy': 0, 'homeGridTotalEnergyStr': 'kWh', 'bypassLoadPower': 0, 'bypassLoadPowerStr': 'kW', 'backupYesterdayEnergy': 0, 'backupYesterdayEnergyStr': 'kWh', 'backupTodayEnergy': 0, 'backupTodayEnergyStr': 'kWh', 'backupMonthEnergy': 0, 'backupMonthEnergyStr': 'kWh', 'backupYearEnergy': 0, 'backupYearEnergyStr': 'kWh', 'backupTotalEnergy': 0, 'backupTotalEnergyStr': 'kWh', 'batteryType': '0', 'psumCalPec': '1', 'dispersionRate': 0, 'upvTotal': 0, 'upvTotalStr': 'V', 'ipvTotal': 0, 'ipvTotalStr': 'A', 'powTotal': 0, 'powTotalStr': 'W', 'batteryCDEnableSet': 0, 'batteryCDSet': 0, 'batteryCDISet': 0, 'batteryCMaxiSet': 0, 'batteryDMaxiSet': 0, 'batteryUvpSet': 0, 'batteryFcvSet': 0, 'batteryAcvSet': 0, 'batteryOvpSet': 0, 'batteryLaTemp': 0, 'psum': 0, 'pepmSet': 0, 'pepm': 0, 'psumCal': 0, 'familyLoadPowerPec': '1', 'pepmStr': 'kW', 'reactivePowerStr': 'Var', 'apparentPowerStr': 'VA', 'pepmSetStr': 'kW', 'psumStr': 'kW', 'psumCalStr': 'kW'}}
  File "/config/custom_components/solis/service.py", line 100, in async_discover
  File "/config/custom_components/solis/service.py", line 120, in _do_discover
  File "/config/custom_components/solis/soliscloud_api.py", line 234, in fetch_inverter_data
  File "/config/custom_components/solis/soliscloud_api.py", line 316, in _post_process

hultenvp, 2022-10-02:
Do you also get the same issues in the test app?

drsmarsden, 2022-10-02:
I get this back

{"code": "1", "data": null, "msg": "\u6570\u636e\u5f02\u5e38 \u8bf7\u8054\u7cfb\u7ba1\u7406\u5458", "success": true}

Not looked at the test app before

hultenvp, 2022-10-02:
Hi Stuart,

Code should be 0 for a successful call, other codes are specified as error. The non-zero values are not specified further. I can only assume your key and/or secret are no longer valid.
The message is Chinese, can get it translated.

hultenvp, 2022-10-02:
I'm still interested in the full stack trace of the error. It's not going to help you, but it shouldn't crash.

drsmarsden, 2022-10-03:
2022-10-03 15:20:14.520 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved                                                                                           
Traceback (most recent call last):                                                                                                                                                                       
  File "/config/custom_components/solis/service.py", line 100, in async_discover                                                                                                                         
    capabilities = await self._do_discover()                                                                                                                                                             
  File "/config/custom_components/solis/service.py", line 120, in _do_discover                                                                                                                           
    data = await self._api.fetch_inverter_data(inverter_serial)                                                                                                                                          
  File "/config/custom_components/solis/soliscloud_api.py", line 234, in fetch_inverter_data                                                                                                             
    self._post_process()                                                                                                                                                                                 
  File "/config/custom_components/solis/soliscloud_api.py", line 316, in _post_process                                                                                                                   
    float(self._data[INVERTER_TIMESTAMP_UPDATE])/1000                                                                                                                                                    
KeyError: 'timestampUpdate'            

I guess it might then fail looking for another key