hultenvp/solis-sensor #173: SolisCloud Invalid Credentials

alienatedsec, 2022-10-22:
My integration worked perfectly fine before moving from 2.x.x to 3.0.2 shortly before opening this issue.

Whenever trying to use existing API credentials (not new, existing that was working via configuration.yaml) I have the below error appearing

[image]

How to troubleshoot or what could be wrong with the URL or credentials?

alienatedsec, 2022-10-22:
@hultenvp I reverted to 2.3.5, and it is working fine again.

[image]

hultenvp, 2022-10-22:
The has been some discussion on the discussions page. The plant_id is handled as an int, but for some values it starts rounding the int for some reason I still do not entirely get. In the configuration.yaml the plant_id is handled as a string.

closed 2022-10-22 by hultenvp

hultenvp, 2022-10-22:
Can't fully validate myself, but should work now

alienatedsec, 2022-12-20:
@hultenvp Again the same problem with the recent version

PR #174 (2022-10-22): 173 soliscloud invalid credentials
Closes #173