hultenvp/solis-sensor #177: Moving to v3.0.5 caused duplicated entity names

snigehere, 2022-10-23:
**Describe the bug**
I moved o 3.05 this evening and on restart got a messaged from the system to remove my solis config from YAML as it was moved to the UI.  I edited the config to comment out the solis details and restarted

All loaded up ok except all integration entity names were duplicated with the new ones having "_2" on the end

This meant that any automations and component integrations that used the entity IDs would all have to be updated 

**Config**
N/A

**Versions**
Windows 10 running Oracle Virtual box Version 6.1.38 r153438 (Qt5.6.2)

Home Assistant 2022.10.5
Supervisor 2022.10.0
Operating System 9.2
Frontend 20221010.0 - latest
Integration version:	1.28.3
Frontend version:	20220906112053
Solis 3.05

**Debug logs**
Add to configuration.yaml:
```
logger:
  default: warning
  logs:
    custom_components.solis: debug
```
And attach the resulting log to the issue. If yo do not know where to find logs you missed an important step in the install of HACS. ;-)

**Screenshots**
If applicable, add screenshots to help explain your problem.

dpgh947, 2022-10-24:
I got hit by this when I took the initial 3.0 update, which fell in a heap due to the "missing https://" problem which was since fixed. I managed to get through this, but it was messy and I ended up with the _2 entities too.  I found you can delete the old entities and then rename the new _2 ones back to those original names, I did this and it seemed to be ok, but I had further trouble with it simply not working at all after a restart of HA, all entities remained "unavailable" until I manually reloaded the integration, after which it worked fine. This was a solid problem after every HA restart and I could not find anything in any logs anywhere to explain why, so I ended up backing out to 2.3.5 again and have not re-attempted the upgrade yet. Maybe caused by the delete/rename, but I don't know why this would break it, or why a reload would make it start working again until the next HA restart. I have been put off trying to upgrade again partly because practically every time I look, another bug has been spotted and fixed (up to 3.0.5 now).

snigehere, 2022-10-24:
Yes.  I did the same as you in deleting the old and renaming the new .. it was just a very untidy change so decided to report as a bug.

Thanks for the comments on restart as I found I had to restart the Integration this morning ... i had wondered if i had forgotten to do a full restart of ha yesterday although I beleived I had, or maybe solus cloud had been down.  Based on your experience I will do some more checking tomorrow to see what I find.

cpuks, 2022-10-24:
Same problem here on 3.0.5, also if you have two or more plants and try to change entity name - entity is moved to other plant. So in the end I have two plants one empty and one with all entities from both - 48 (2 x 24). If this explanation is unclear please leave a msg and I'll provide screenshot as for now I'm using default names to not mess up my stats.

cpuks, 2022-10-25:
Something odd is with version 3.0.5 even as a fresh setup - second-day pooling data hangs around 10 - no update since, reloaded integration - no go. Had to revert back to 2.3.5 - rock solid no issues.

snigehere, 2022-10-25:
I see the integration just stops running ... I restarted it yesterday and it looked like it was running but no solar data at all dring the day ... restarted just now and days solar is then shown all in one hit this evening.

I looked to downgrade but it only offered me other 3.x versions or "master" .. no idea what "master" is ... I am reluctant to delete it because I have a lot of data collected for this year

dpgh947, 2022-10-26:
Oh dear, yes I also don't see 2.3.5 offered under "redownload" any more, glad I did it while it was there. I have been running fine since I backed out, I won't be trying the upgrade again until this is fixed.

cpuks, 2022-10-26:
By default hacs offers 5 releases - go to integrations hacs and change offered versions to 10 or something then go back to 2.3.5

snigehere, 2022-10-27:
> By default hacs offers 5 releases - go to integrations hacs and change offered versions to 10 or something then go back to 2.3.5

Many thanks - that fixed it ... yesterday the integration again stopped during the day messing up my solar generation chart.  I hadn't spotted it had stopped as I was out but it restarted when I restarted HA to install a system update.  Nothing captured from 14:00 to end of day (sunny) then caught up after the restart.

I have now rolled back to 235 until the bugs are sorted out.

[image]

hultenvp, 2022-10-30:
Sorry about this one, this one slipped through my tests. Been working on and off for a few months introducing the config flow and the changes are more significant than expected. 
Will try to reproduce and fix the duplicate entries. The crash that @snigehere is reporting will be handled in issue #180

hultenvp, 2022-10-30:
I was able to reproduce the issue. Found the cause is one underscore.

hultenvp, 2022-10-30:
Fixed in v3.0.7.

closed 2022-10-30 by hultenvp

dpgh947, 2022-10-31:
I've just updated to 3.0.7, upgrade all fine this time, seems to be working, thanks.

dpgh947, 2022-11-03:
I spoke too soon. I only just noticed that the integration had stopped updating on 3.0.7 like it did before, all the sensors were populated rather than unavailable, but with readings from yesterday. I reloaded the integration and it all kicked off again. Nothing obvious in the logs. I think I will be backing out again.

snigehere, 2022-11-03:
> I spoke too soon. I only just noticed that the integration had stopped updating on 3.0.7 like it did before, all the sensors were populated rather than unavailable, but with readings from yesterday. [...]

This issue is covered in #180

dpgh947, 2022-11-03:
Thanks, I thought that had been "fixed" already. BTW I cannot backout to 2.3.5 any more, I now get "Setup failed for custom integration solis: No setup or config entry setup function defined." no matter what I do. Completely removed solis, re-downloaded 2.3.5, put the solis sensor info back in configuration.yaml, rebooted entire machine, it will not start.