hultenvp/solis-sensor #180: The integration stops retrieving data from the server.

heebo1974, 2022-10-28:
**Describe the bug**
Every now and then (daily) integration stops updating data from server.
I have Solar Sofar 4.4KTL-X inverter.

I use these settings:
Sofar solar (home.solarman.cn)
portal version ginlong_v2

**Versions**
HA Version an deployment 2022.10.5
HACS version
Integration version 1.28.3

**core logs**
`...
2022-10-28 08:36:26.070 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.solar_sofar_pv1_power is taking over 10 seconds
2022-10-28 08:37:26.073 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.solar_sofar_pv1_power is taking over 10 seconds
2022-10-28 08:38:26.074 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.solar_sofar_pv1_power is taking over 10 seconds
2022-10-28 08:39:26.076 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.solar_sofar_pv1_power is taking over 10 seconds
2022-10-28 08:40:26.075 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.solar_sofar_pv1_power is taking over 10 seconds
2022-10-28 08:41:26.076 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.solar_sofar_pv1_power is taking over 10 seconds
2022-10-28 08:42:26.077 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.solar_sofar_pv1_power is taking over 10 seconds
2022-10-28 08:43:26.083 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.solar_sofar_pv1_power is taking over 10 seconds
2022-10-28 08:44:26.082 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.solar_sofar_pv1_power is taking over 10 seconds
2022-10-28 08:45:26.084 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.solar_sofar_pv1_power is taking over 10 seconds
2022-10-28 09:20:17.614 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
File "/config/custom_components/solis/service.py", line 182, in async_update
if await self._login():
File "/config/custom_components/solis/service.py", line 92, in _login
if await self._api.login(async_get_clientsession(self._hass)):
File "/config/custom_components/solis/ginlong_api.py", line 225, in login
self._plant_name = getattr(data, INVERTER_PLANT_NAME)
AttributeError: 'NoneType' object has no attribute 'plantName'
`
There is also this log entry in supervisor log:
`22-10-28 09:58:47 WARNING (SyncWorker_3) [supervisor.addons.validate] Add-on config 'startup' with 'before' is deprecated. Please report this to the maintainer of SofarSolarMQTT`

heebo1974, 2022-10-28:
I'll add debug logs later.

metcrl, 2022-10-28:
Same here.  I hope we'll find a solution

donal222022, 2022-10-29:
I am having the same issue since I moved the config from yaml to GUI Integration, stops at least once everyday.

TiemenSoeteman, 2022-10-29:
I have the same issue. Also started when I stopped using the yaml configuration. At first I thought it might have something to do with the inverter going offline when it stops producing electricity, but I also noticed it stopped working during the day. Reloading the integration solves the problem for roughly 24 hours.

hultenvp, 2022-10-30:
I added reading the plant name to the login procedure. Apparently after some point in time the integration is forced to renew the login. This succeeds but reading the plant name doesn't for some reason.
It's easy to avoid the crash, but I cannot reproduce the issue myself, so need your feedback to check it works now.
Do all of you see the issue on the ginlong API or are there also people seeing it on Soliscloud?

metcrl, 2022-10-30:
> I added reading the plant name to the login procedure. Apparently after some point in time the integration is forced to renew the login. This succeeds but reading the plant name doesn't for some [...]

For me the problem is on ginlong API. I'll check if it's working now

donal222022, 2022-10-30:
Its ginlong API for me , will feedback on the results

hultenvp, 2022-10-30:
@donal222022 Fix is in v3.0.6

metcrl, 2022-10-31:
It's happening again, it just stopped receiving data yet

hultenvp, 2022-10-31:
Reopening.
General remark: There are so many different systems out there all resulting in even different behavior from the server it is impossible for me to system test all possible versions. It took me months to even get something simple as Energy Today stable and reliable.
**So if an issue occurs: please add the logs. Without them you can tell me 100x it doesn't work, but I cannot solve it.**

heebo1974, 2022-10-31:
I haven't had any problems since the update so far.

metcrl, 2022-10-31:
I have now updated to 3.0.7 and I have duplicated all sensors, so I need to fix this first.
In any case, I'll keep an eye on the "2" versions of the sensors to see if they get stuck.
Forgive the ignorance where do I find the log specifically?

donal222022, 2022-10-31:
I thought the issue had happened again, but it was the duplicates was my problem this time, will turn on debug and monitor.

hultenvp, 2022-10-31:
> I have now updated to 3.0.7 and I have duplicated all sensors, so I need to fix this first. In any case, I'll keep an eye on the "2" versions of the sensors to see if they get stuck. Forgive the ignorance where do I find the log specifically?

Yeah the remaining dups are a pain as well. Another example of how many different use cases there are. Doesn't help that I need to learn it myself on the fly as well ;-). I hope the issue #186 will give me some leads here.

Logs are under settings->System->logs and then log details. Best if debug is on:
https://github.com/hultenvp/solis-sensor/wiki/FAQ#you-ask-me-for-debug-logs-how-do-i-do-that

metcrl, 2022-10-31:
Just to understand, is it better to fix duplicate sensors by hand or are we waiting for a resolutive update?
If you say we fix them by hand by eliminating the old ones, and see if the new ones still encounter the problem of stuck retrieving data.

Thanks a lot!!!

metcrl, 2022-11-02:
Yes it's stopper again, i can confirm

snigehere, 2022-11-02:
Mine stopped as well ...which I guess suggests a common incident on the cloud may be the trigger ... Unfortunately I don't have the logs for this one as I did a couple of restarts a bit later to update some other stuff before I spotted it had stopped.  I have added the last update timestamp to my main dashboard so I can spot if it stops.
[image]

hultenvp, 2022-11-03:
> Hmm.. Now I think that it stopped again ?
> 
> `2022-11-02 10:18:27.700 DEBUG (MainThread) [custom_components.solis.ginlong_api] Fetching data for serial: xxxxxxxxxxxxxxxxx 2022-11-02 10:18:37.702 DEBUG (MainThread) [custom_components.solis.ginlong_api] Error: Exception: <class 'asyncio.exceptions.TimeoutError'> 2022-11-02 10:18:37.702 INFO (MainThread) [custom_components.solis.ginlong_api] Unable to fetch details for device with ID: 105176242 2022-11-02 10:18:37.702 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 1 minutes. 2022-11-02 10:19:45.978 INFO (MainThread) [custom_components.solis.ginlong_api] Login Successful! 2022-11-02 10:19:55.979 DEBUG (MainThread) [custom_components.solis.ginlong_api] Error: Exception: <class 'asyncio.exceptions.TimeoutError'> File "/config/custom_components/solis/service.py", line 182, in async_update File "/config/custom_components/solis/service.py", line 92, in _login File "/config/custom_components/solis/ginlong_api.py", line 224, in login`

Looks like an outage server side. The timeout error is new, I'll catch it so at least it recovers (I assume it didn't)

snigehere, 2022-11-03:
The last update I saw was at 03-Nov-2022 08:57:25 UTC ... I saw there were timeouts before that time and later a home assistant exception which seems to have killed the integration 
Again I see the double fetch after the timeout ... longer log attached rather than in the page

[solis log.txt](https://github.com/hultenvp/solis-sensor/files/9927277/solis.log.txt)

donal222022, 2022-11-03:
Adding my log which failed earlier:

2022-11-03 09:13:30.947 DEBUG (MainThread)
[custom_components.solis.ginlong_api] Fetching data for serial:
xxxxxxxxxxxxxx
2022-11-03 09:15:31.916 DEBUG (MainThread)
[custom_components.solis.ginlong_api] Fetching data for serial:
xxxxxxxxxxxxxx
2022-11-03 09:17:33.682 DEBUG (MainThread)
[custom_components.solis.ginlong_api] Fetching data for serial:
xxxxxxxxxxxxxx
2022-11-03 09:19:34.342 DEBUG (MainThread)
[custom_components.solis.ginlong_api] Fetching data for serial:
xxxxxxxxxxxxxx
2022-11-03 09:21:38.667 DEBUG (MainThread)
[custom_components.solis.ginlong_api] Fetching data for serial:
xxxxxxxxxxxxxx
2022-11-03 09:23:39.428 DEBUG (MainThread)
[custom_components.solis.ginlong_api] Fetching data for serial:
xxxxxxxxxxxxxx
2022-11-03 09:25:40.142 DEBUG (MainThread)
[custom_components.solis.ginlong_api] Fetching data for serial:
xxxxxxxxxxxxxx
2022-11-03 09:27:55.436 DEBUG (MainThread)
[custom_components.solis.ginlong_api] Fetching data for serial:
xxxxxxxxxxxxxx
2022-11-03 09:29:58.146 DEBUG (MainThread)
[custom_components.solis.ginlong_api] Fetching data for serial:
xxxxxxxxxxxxxx
2022-11-03 09:31:59.172 DEBUG (MainThread)
[custom_components.solis.ginlong_api] Fetching data for serial:
xxxxxxxxxxxxxx
2022-11-03 09:32:09.175 DEBUG (MainThread)
[custom_components.solis.ginlong_api] Error: Exception: <class
'asyncio.exceptions.TimeoutError'>
2022-11-03 09:32:09.175 INFO (MainThread)
[custom_components.solis.ginlong_api] Unable to fetch details for device
with ID: xxxxxxxxxxxxxx
2022-11-03 09:33:10.404 INFO (MainThread)
[custom_components.solis.ginlong_api] Login Successful!
2022-11-03 09:33:14.397 DEBUG (MainThread)
[custom_components.solis.ginlong_api] Fetching data for serial:
xxxxxxxxxxxxxx
2022-11-03 09:33:22.565 DEBUG (MainThread)
[custom_components.solis.ginlong_api] Fetching data for serial:
xxxxxxxxxxxxxx
2022-11-03 09:33:32.568 DEBUG (MainThread)
[custom_components.solis.ginlong_api] Error: Exception: <class
'asyncio.exceptions.TimeoutError'>
2022-11-03 09:33:32.569 INFO (MainThread)
[custom_components.solis.ginlong_api] Unable to fetch details for device
with ID: xxxxxxxxxxxxxx
2022-11-03 09:34:35.026 INFO (MainThread)
[custom_components.solis.ginlong_api] Login Successful!
2022-11-03 09:34:44.543 DEBUG (MainThread)
[custom_components.solis.ginlong_api] Fetching data for serial:
xxxxxxxxxxxxxx
2022-11-03 09:35:00.107 DEBUG (MainThread)
[custom_components.solis.ginlong_api] Error: Exception: <class
'asyncio.exceptions.TimeoutError'>
2022-11-03 09:35:00.108 INFO (MainThread)
[custom_components.solis.ginlong_api] Unable to fetch details for device
with ID: xxxxxxxxxxxxxx
2022-11-03 09:35:00.108 DEBUG (MainThread)
[custom_components.solis.ginlong_api] Could not fetch inverter data, retry
loging attempt
2022-11-03 09:36:00.691 INFO (MainThread)
[custom_components.solis.ginlong_api] Login Successful!
2022-11-03 09:36:00.932 DEBUG (MainThread)
[custom_components.solis.ginlong_api] Fetching data for serial:
xxxxxxxxxxxxxx
2022-11-03 09:36:01.426 DEBUG (MainThread)
[custom_components.solis.ginlong_api] Fetching data for serial:
xxxxxxxxxxxxxx
2022-11-03 09:38:02.038 DEBUG (MainThread)
[custom_components.solis.ginlong_api] Fetching data for serial:
xxxxxxxxxxxxxx
2022-11-03 09:40:03.729 DEBUG (MainThread)
[custom_components.solis.ginlong_api] Fetching data for serial:
xxxxxxxxxxxxxx
2022-11-03 09:42:07.217 DEBUG (MainThread)
[custom_components.solis.ginlong_api] Fetching data for serial:
xxxxxxxxxxxxxx
2022-11-03 09:44:08.522 DEBUG (MainThread)
[custom_components.solis.ginlong_api] Fetching data for serial:
xxxxxxxxxxxxxx
2022-11-03 09:46:09.097 DEBUG (MainThread)
[custom_components.solis.ginlong_api] Fetching data for serial:
xxxxxxxxxxxxxx
2022-11-03 09:48:13.351 DEBUG (MainThread)
[custom_components.solis.ginlong_api] Fetching data for serial:
xxxxxxxxxxxxxx
2022-11-03 09:50:19.253 DEBUG (MainThread)
[custom_components.solis.ginlong_api] Fetching data for serial:
xxxxxxxxxxxxxx
2022-11-03 09:50:29.327 DEBUG (MainThread)
[custom_components.solis.ginlong_api] Error: Exception: <class
'asyncio.exceptions.TimeoutError'>
2022-11-03 09:50:29.328 INFO (MainThread)
[custom_components.solis.ginlong_api] Unable to fetch details for device
with ID: xxxxxxxxxxxxxx
2022-11-03 09:51:37.164 INFO (MainThread)
[custom_components.solis.ginlong_api] Login Successful!
2022-11-03 09:51:47.166 DEBUG (MainThread)
[custom_components.solis.ginlong_api] Error: Exception: <class
'asyncio.exceptions.TimeoutError'>
2022-11-03 09:51:47.167 ERROR (MainThread) [homeassistant] Error doing job:
Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/solis/service.py", line 182, in
async_update
    if await self._login():
  File "/config/custom_components/solis/service.py", line 92, in _login
    if await self._api.login(async_get_clientsession(self._hass)):
  File "/config/custom_components/solis/ginlong_api.py", line 224, in login
    data = await self.fetch_inverter_data(next(iter(self._inverter_list)))
TypeError: 'NoneType' object is not iterable

donal222022, 2022-11-03:
> Adding my log which failed earlier: 2022-11-03 09:13:30.947 DEBUG (MainThread) [custom_components.solis.ginlong_api] Fetching data for serial: xxxxxxxxxxxxxx 2022-11-03 09:15:31.916 DEBUG [...]

In relation to this, I am able to get it working again by Disabling and Enabling the integration again.

Many Thanks,

Donal.

hultenvp, 2022-11-03:
First of all thanks for all the logs, this really helps.

I'll try to get out a fix tonight.

donal222022, 2022-11-03:
Thanks a mill, let me know if you need more logs etc.

snigehere, 2022-11-03:
> In relation to this, I am able to get it working again by Disabling and Enabling the integration again.

I am able to get it going again just by hitting reload (option at the top of the menu where disable is at the bottom)

trialek, 2022-11-03:
Hi all,
I have the same issue and use 3.0.7 v
Stops reviving data.

hultenvp, 2022-11-03:
@snigehere 
> I do see a few times the app times out and wonder if there is some code issue caused by this .. What I notice in the log extract below is two fetch cycles without a schedule between them and also "device id" it is mentioned where al the other messages are "data for serial" .. I don't know if these are pointers to a root cause

What you see there is actually the integration recovering from a timeout. If it fails to retrieve data it will log out and log in again and merrily continue.
Still it is pointing to the root cause. You see that in this part of the error @donal222022 reported:

> [custom_components.solis.ginlong_api] Error: Exception: <class 'asyncio.exceptions.TimeoutError'> 2022-11-03 09:51:47.167 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never [...]

There as part of the login the get inverter list fails (it uses the same low-level service call and in that call the timeout occurs), resulting in an empty list. Next line it trips over trying to iterate a non-existing list while trying to fetch the plant name.

hultenvp, 2022-11-03:
The TypeError should now be handled properly. Feel free to reopen if crashes still occur

PR #193 (2022-11-03): 180 the integration stops retrieving data from the server 2
Closes #180
closed 2022-11-03 by hultenvp