hultenvp/solis-sensor #183: Cannot login with provided url and credentials

mvereecken, 2022-10-30:
**Describe the bug**
When upgraden to 3.0.5 I had the message that I had to remove the integration from configuration.yaml, so I did. I reinstalled HACS integration, but when entering the credentials I get a "Cannot login with provided url and credentials".
Just to make sure I have the right values, I checked again in my Solis account, and the values were still the same as before.

**Versions**
HA version: 2022.10.5
HACS: 1.28.3
Solis-sensor: 3.0.5

mvereecken, 2022-10-30:
I manually reverted to 2.3.5, added my old settings in configuration.yaml and it works again.

hultenvp, 2022-10-30:
What does the log say?

mvereecken, 2022-11-04:
I couldn't find anything in the log about it. Or maybe I looked in the wrong place? I looked into the Home Assistante Core logs.

hultenvp, 2022-11-04:
Enable logs as described [here](https://github.com/hultenvp/solis-sensor/wiki/FAQ#you-ask-me-for-debug-logs-how-do-i-do-that) and then indeed look in the core log details.

Note that there's a 3.0.8 with quite a lot of small improvements and fixes. Also fixes an issue where the UI would still proceed if no platform was specified causing a

hultenvp, 2022-11-04:
Oops accidentally closed

hultenvp, 2022-11-04:
See also #197. You'll probably have so wait for Solis to fix this.

geduxas, 2022-11-22:
Hello i am new here, and just stuck with same problem.. that i have found that timeout for soliscloud response is only 10sec. For me solis cloud responded only in 23sec.. so i prolonged timeout to 100s.

greigallen, 2022-11-27:
@geduxas I'm having the same issue, how did you extend the timeout?

geduxas, 2022-11-27:
> @geduxas I'm having the same issue, how did you extend the timeout?

Just change timout from 10 to 100

https://github.com/hultenvp/solis-sensor/blob/86c087371f5df2688401413adeebf10e5b2015fa/custom_components/solis/soliscloud_api.py#L496-L498

wyliecoyoteuk, 2022-11-27:
Hi, I had this issue, so I changed the timeout, I now get "An unknown error occured"

geduxas, 2022-11-27:
> Hi, I had this issue, so I changed the timeout, I now get "Unknown error occurred" I have the V3 datalogger, should this work with it.

Maybe your problem is different..

wyliecoyoteuk, 2022-11-27:
Tried again, nowgett

> Hi, I had this issue, so I changed the timeout, I now get "Unknown error occurred" I have the V3 datalogger, should this work with it.

Tried again, now "getting cannot log in with supplied credentials" after 100 sec. 
In logs, I am getting "no inverters found".
One question. Is the "inverter name" requested, a generic name (e.g.Solis), the plant name, or the actual inverter name?

geduxas, 2022-11-27:
> Tried again, nowgett
> 
> > Hi, I had this issue, so I changed the timeout, I now get "Unknown error occurred" I have the V3 datalogger, should this work with it.
> 
> Tried again, now "getting cannot log in with supplied credentials" after 100 sec. In logs, I am getting "no inverters found". One question. Is the "inverter name" requested, a generic name (e.g.Solis), the plant name, or the actual inverter name (default is "inverter")?

which system do you use? if you are trying to connect to soliscloud so you need to use plant ID from URL when you are in soliscloud website...

wyliecoyoteuk, 2022-11-27:
> which system do you use? if you are trying to connect to soliscloud so you need to use plant ID from URL when you are in soliscloud website...

I have the plant ID, but what is the inverter name?

Is it the plant name or the inverter name?  
E.g. Id is 466abc,  plant name is Fred, inverter name is inverter1 which is the string needed for login?

wyliecoyoteuk, 2022-11-27:
The problem is that everyone assumes they are using the correct terms, when there are multiple vague options.
I may be doing everything right and still having problems, but incorrect naming causes errors.

geduxas, 2022-11-27:
> The problem is that everyone assumes they are using the correct terms, when there are multiple vague options. I may be doing everything right and still having problems, but incorrect naming causes errors.

For solis cloud, you will need api url, login e-mail, api key, and api secret, also station id. Station id you should login to yours soliscloud, open your inverter.. and in full url you should find your station id:

https://soliscloud.com/#/station/stationdetail_1?id=[your_station_id]

wyliecoyoteuk, 2022-11-28:
> For solis cloud, you will need api url, login e-mail, api key, and api secret, also station id. Station id you should login to yours soliscloud, open your inverter.. and in full url you should find [...]

I have all of those, they are clearly labelled. What I am asking about is the inverter **name** which is required on the first screen. This is prefilled with Solis. Should I use the plant name or the inverter name? I have tried both but still get rejected.

geduxas, 2022-11-28:
> I have all of those, they are clearly labelled. What I am asking about is the **inverter name** which is required on the first screen. This is prefilled with Solis. Should I use the plant name or [...]

My fault :D no, its just label for home assistant device. All you need from first screen server address and protocol type.. inverter name is just for you.. if you have couple of them.

wyliecoyoteuk, 2022-11-28:
Thanks. I didn't realise that.
I have tried to connect several times now, without success, I suppose I was looking for any possible cause for it to fail.

geduxas, 2022-11-28:
> Thanks. I didn't realise that. I have tried to connect several times now, without success, I suppose I was looking for any possible cause for it to fail.

Dou you have api access? Have you enabled it from soliscloud?

wyliecoyoteuk, 2022-11-28:
> > Thanks. I didn't realise that. I have tried to connect several times now, without success, I suppose I was looking for any possible cause for it to fail.
> 
> Dou you have api access? Have you enabled it from soliscloud?

Yes, I have the url, key and the secret, etc.

gutyex, 2022-11-28:
I'm also getting this error when trying to set up the integration with a newly-created API access key (access was enabled over the weekend by Solis support)  
  
Debug logs show:

`2022-11-28 10:07:37.680 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found`

soliscloud_test.py works fine, I'm able to get a response for all 8 of the requests in it. Most of them return what looks like good data, numbers match what I'm seeing in the web interface, but UserStationList gives:

```
{
    "code": "0",
    "data": {
        "mpptSwitch": 0,
        "page": {
            "current": 1,
            "optimizeCountSql": false,
            "orders": [],
            "pages": 0,
            "records": [],
            "searchCount": true,
            "size": 20,
            "total": 0
        },
        "stationStatusVo": {
            "all": 0,
            "building": 0,
            "fault": 0,
            "mppt": 0,
            "normal": 0,
            "offline": 0
        }
    },
    "msg": "success",
    "success": true
}
```

wyliecoyoteuk, 2022-11-28:
2022-11-28 12:32:04.323 ERROR (MainThread) [frontend.js.latest.202211080] :0:0 ResizeObserver loop completed with undelivered notifications.
2022-11-28 12:33:15.092 DEBUG (MainThread) [custom_components.solis.soliscloud_api] workarounds: {'correct_daily_on_grid_energy_enabled': True}
2022-11-28 12:33:23.196 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found

Nolb, 2022-11-28:
Been struggling with this since getting the api key last week, kept getting the 'No inverter' warning, found the solution here https://github.com/hultenvp/solis-sensor/discussions/71

Just drop the / from the API URL after 13333

gutyex, 2022-11-28:
I tried setting it up without the trailing / on the URL as @Nolb suggested, logs now show:

```
2022-11-28 10:07:37.680 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found
2022-11-28 12:47:26.257 DEBUG (MainThread) [custom_components.solis.soliscloud_api] workarounds: {'correct_daily_on_grid_energy_enabled': True}
2022-11-28 12:47:26.667 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Error from URI (/v1/api/inveterList) : 0, message='Attempt to decode JSON with unexpected mimetype: text/html', url=URL('https://www.soliscloud.com:13333/v1/api/inveterList')
```

wyliecoyoteuk, 2022-11-28:
could it be the typo inveterlist?

Nolb, 2022-11-28:
The only other things I did were on the first setup page where it asks for inverter name I used the exact name I have it set as on the Solis cloud, not inverter I'd (don't know if it matters or not) and used my email address rather than user name on the other page, not sure what else it could be sorry

wyliecoyoteuk, 2022-11-28:
One thing, if i look at the station ID using the URL above, I get an 18 digit decimal number. whereas I was using the one on the web page which is a six digit hexadecimal number.
Neither seems to work, but which one should I use?

Nolb, 2022-11-28:
I used the 18 digit ID in the URL for the station ID, when it connected it then returned the 6 digit hexadecimal value as the inverter name against the integration which matches the one in Solis cloud

Nolb, 2022-11-28:
The url above is the inverter detail, although it does have the station at the end as well as the inverter id, it should be the same as on this url (https://www.soliscloud.com/#/station/stationdetail_1?id=)

wyliecoyoteuk, 2022-11-28:
Working!
The number needed is actually the inverter ID, which is 19 digits long (mine was the same as the station ID with one extra digit)
Thank you for your patience.

hultenvp, 2022-11-29:
Nice to see it works, I've updated the readme to make the registration and configuration more transparent.
To answer some more of the discussions in this thread:

> Just change timeout from 10 to 100

I will add this to the code. I notice that the current version of HA processes a lot asynchronically during startup and needs more slack on timeouts.

> Just drop the / from the API URL after 13333

I want to add some robustness to the code to remove any trailing slashes.

> could it be the typo **inveterlist**?

No, they typo is in their endpoint.

> I used the 18 digit ID (actually mine is 19 digits) in the URL for the station ID, when it connected it then returned the 6 digit hexadecimal value as the inverter name against the integration which matches the one in Solis cloud

Yes, you need the 19 digit ID, I made that more clear in the readme. On the inverter name: Before the API lockdown the API would return the Station/Plant name. Now it no longer does, so in the device registry I now have to use the 6-digit HEX value instead of a real name.

> The problem is that everyone assumes they are using the correct terms, when there are multiple vague options.
I may be doing everything right and still having problems, but incorrect naming causes errors.

@wyliecoyoteuk: Please check the new [readme](https://github.com/hultenvp/solis-sensor#readme). Would be helpful if you could let me know if this is now more clear or still needs further improvement.

wyliecoyoteuk, 2022-11-29:
That's great. I have now noticed that the number is sometimes truncated on some screens in soliscloud, not sure why.
I did a brief how-to on my blog. 
https://wyliecoyoteuk.wordpress.com/2022/11/28/home-assistant-with-a-solis-hybrid-inverter/

jjeuriss, 2022-12-05:
@hultenvp A suggestion for the [readme](https://github.com/hultenvp/solis-sensor#readme): it would would be to clarify that if you select `soliscloud` as PV Portal version, you also need to specify the API URL. By default https://m.ginlong.com was filed in and I didn't realize I needed to change it to https://www.soliscloud.com:13333 (without slash) after selecting `soliscloud`.

jmason, 2022-12-08:
> > could it be the typo **inveterlist**?
> 
> No, they typo is in their endpoint.

BTW, I think this is fixed -- it looks like the endpoints are now duplicated with and without the typo.

matthewbyrne, 2022-12-09:
Having this issue too.
PV URL: https://www.soliscloud.com:13333
Copying the secret and key from the API settings, and the station ID from https://www.soliscloud.com/#/station/stationdetail_1?id=xxxxxxxxxxx

Result:
Cannot login with provided URL and credentials

jonathanmandrews2022, 2022-12-11:
Seem to get this strange error when attempt to set up (put * on serial no)

2022-12-11 12:12:48.688 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Found inverters: ['6031023228200***']
2022-12-11 12:12:48.688 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: 6031023228200***
2022-12-11 12:12:48.809 INFO (MainThread) [custom_components.solis.soliscloud_api] /v1/api/stationDetail responded with error: 1:数据异常 请联系管理员
2022-12-11 12:12:48.809 DEBUG (MainThread) [custom_components.solis.ginlong_base] AttributeError, plantName does not exist
2022-12-11 12:12:48.809 INFO (MainThread) [custom_components.solis.soliscloud_api] Failed to acquire plant name, login failed

roshanabady, 2022-12-13:
I'm a beginner, and I had the same issue, as there are up to 3 different IDs in the URL.
And the correct one is actually what most everyone has suggested, which is: https://www.soliscloud.com/#/station/stationdetail_1?id=xxxxxxxxxxx
But I believe a visual guide is always the best(My apologies for the Pictoral guide):

1. log in to your Solis Cloud Account
2. In the **Plant Overview** tab, under the **Plant Name** column, Click on your actual _plant name_, per the screenshot below:
[image]
3. Find the 19-digit ID in the new screen at the end of the page URL, and take the screenshot as a guide:
[image]
4. For the rest of the process, follow the main guide
 @hultenvp cheers, mate, You Rock!

matthewbyrne, 2022-12-14:
Still no success here!
https://www.soliscloud.com:13333 without the trailing `/`
Email address that I use to login to solis cloud
API Secret  = KeySecret from soliscloud
API Key ID = KeyID from soliscloud
Station ID = As shown above from https://www.soliscloud.com/#/station/stationdetail_1?id=19digits

Any ideas!?

geduxas, 2022-12-15:
@matthewbyrne 

Try to increase timeout from 10 to 100

I have noticed that solis cloud response very delayed for me.. so increase of timeout solved problem.. 

https://github.com/hultenvp/solis-sensor/blob/86c087371f5df2688401413adeebf10e5b2015fa/custom_components/solis/soliscloud_api.py#L496-L498

matthewbyrne, 2022-12-15:
> @matthewbyrne
> 
> Try to increase timeout from 10 to 100
> 
> I have noticed that solis cloud response very delayed for me.. so increase of timeout solved problem..
> 
> https://github.com/hultenvp/solis-sensor/blob/86c087371f5df2688401413adeebf10e5b2015fa/custom_components/solis/soliscloud_api.py#L496-L498

Just tried that, but the integration never takes near 10 seconds to fail. The 100 seconds timeout made no difference. 
No errors in the homeassistant logs either.

geduxas, 2022-12-15:
@matthewbyrne just double check, do you reloaded home asistant after change?

hultenvp, 2022-12-16:
People are reporting various connectivity issues in different tickets.
Let me summarize:
1. Sometimes the server throws a HTTP 408 error, this happens when there's time drift between client and server (i.e. your system time is off more than 15 minutes). ==> I have an update ready that at least will give you a nice warning if that's the case.
2. Sometimes the server throws a HTTP 502 error. This is a server side error that comes and goes, but some people seem to have more trouble with it than others. ==> I have an update ready that throttles the amount of calls made per second to avoid overloading the server. I'm not sure this solves the issue, because I cannot reliably reproduce. 
3. Some people report a debug output error in Chinese that translates into "The data is abnormal, please contact the administrator", see [here](https://github.com/hultenvp/solis-sensor/discussions/189) ==> There's no solution yet.

@matthewbyrne: Maybe it's caused by problem number 2, but I'm not sure. If the coming update does not solve it, please try to run the test app, in discussion [189](https://github.com/hultenvp/solis-sensor/discussions/189) is manual how to do that. The login sequence makes the calls in the following order: InverterList, InverterDetail, StationDetail. You might want to try using the test app if the individual calls do report correct data.

hultenvp, 2022-12-16:
I'll close this issue, further issues after release 3.2.0 please report in #232

closed 2022-12-16 by hultenvp

hultenvp, 2022-12-16:
By the way, @roshanabady : I used your screenshots and description to improve the readme. Thanks!

matthewbyrne, 2022-12-20:
> People are reporting various connectivity issues in different tickets. Let me summarize:
>
> 1. Sometimes the server throws a HTTP 408 error, this happens when there's time drift between client and [...]

Thanks for the info. That soliscloud_test.py test app should be in a troubleshooting readme. It's very handy.
I get Chinese characters unfortunately. Or specifically unicode of chinese characters:
`{"code": "1", "data": null, "msg": "\u6570\u636e\u5f02\u5e38 \u8bf7\u8054\u7cfb\u7ba1\u7406\u5458", "success": true}`
[Site to translate unicode](https://r12a.github.io/app-conversion/)

How to use test app: [link](https://github.com/hultenvp/solis-sensor/discussions/189#discussioncomment-4383696)
Test app code: [soliscloud_test.py](https://github.com/hultenvp/solis-sensor/blob/master/test/soliscloud_test.py)

hultenvp, 2022-12-21:
Yup, that's the "abnormal data" issue. 

I'm working on a new test app to get more insights if there are more commonalities between the various people reporting the issue.
Could you check what response inverterDetail and stationAll endpoints give? They should respond with data, and I'm interested what data they give

matthewbyrne, 2022-12-23:
Anything in particular you're looking for? Nothing jumping out at me as overly important. i've redacted a few personal ids and serial numbers etc.

stationsAll
```
{
  "code": "0",
  "data": [
    {
      "batteryChargeEnergy": 0,
      "batteryDischargeEnergy": 0,
      "consumeEnergy": 1146.3,
      "energy": 202,
      "energyPec": "1",
      "energyStr": "kWh",
      "errorFlag": 0,
      "fullHour": 40.4,
      "gridPurchasedEnergy": 1018.01,
      "gridPurchasedIncome": 0.52,
      "gridSellEnergy": 73.71,
      "gridSellIncome": 20.894,
      "homeLoadEnergy": 198,
      "id": "<id here>",
      "money": 56.962,
      "moneyPec": "1",
      "moneyStr": "GBP",
      "offSetEnergy": 0,
      "offSetIncome": 0,
      "oneSelf": 128.29,
      "produceEnergy": 202,
      "timeZone": 8,
      "year": 2022
    }
  ],
  "msg": "success",
  "success": true
}
```

inverterDetails:
```
{
  "code": "0",
  "data": {
    "acInType": 0,
    "acOutputType": 0,
    "apparentPower": 0,
    "apparentPowerStr": "VA",
    "backupMonthEnergy": 0,
    "backupMonthEnergyStr": "kWh",
    "backupTodayEnergy": 0,
    "backupTodayEnergyStr": "kWh",
    "backupTotalEnergy": 0,
    "backupTotalEnergyStr": "kWh",
    "backupYearEnergy": 0,
    "backupYearEnergyStr": "kWh",
    "backupYesterdayEnergy": 0,
    "backupYesterdayEnergyStr": "kWh",
    "batteryAcvSet": 0,
    "batteryAlarm": "0",
    "batteryCDEnableSet": 0,
    "batteryCDISet": 0,
    "batteryCDSet": 0,
    "batteryCMaxiSet": 0,
    "batteryCapacitySoc": 0,
    "batteryChargeEnergy": 0,
    "batteryChargeEnergyStr": "kWh",
    "batteryChargingCurrent": 0,
    "batteryChargingCurrentStr": "A",
    "batteryDMaxiSet": 0,
    "batteryDischargeEnergy": 0,
[... 387 lines cut ...]
    "uPv32": 0,
    "uPv32Str": "V",
    "uPv3Str": "V",
    "uPv4": 0,
    "uPv4Str": "V",
    "uPv5": 0,
    "uPv5Str": "V",
    "uPv6": 0,
    "uPv6Str": "V",
    "uPv7": 0,
    "uPv7Str": "V",
    "uPv8": 0,
    "uPv8Str": "V",
    "uPv9": 0,
    "uPv9Str": "V",
    "upvTotal": 0,
    "upvTotalStr": "V",
    "userId": "<REDACTED>",
    "version": "3D0037",
    "warningInfoData": 0
  },
  "msg": "success",
  "success": true
}
```

wyliecoyoteuk, 2022-12-23:
I am having the same problem. It worked until a few days ago.
the test script gives {"code": "1", "data": null, "msg": "\u6570\u636e\u5f02\u5e38 \u8bf7\u8054\u7cfb\u7ba1\u7406\u5458", "success": true}.
I was also getting HTPS: failures in the Home assistant logs. I tried to reinstall and got the login error.

hultenvp, 2022-12-23:
@matthewbyrne : No nothing in particular, just the fact it does report data ;-)

Thanks for the logs, they helped me confirming what I thought.

I've released a 3.2.2, it should solve your issues. Also valid for @wyliecoyoteuk 
If you experience further issues, please report them in #229 or #232 whichever is most appropriate.

Cheers

matthewbyrne, 2022-12-30:
@hultenvp , looks like 3.2.2 has solved my issues. Just configured now, and got a "configuration created successfully for station xxxx".
Thanks!

wyliecoyoteuk, 2022-12-30:
For me also :)

ilmazzo87, 2023-11-09:
> Still no success here! https://www.soliscloud.com:13333 without the trailing `/` Email address that I use to login to solis cloud API Secret = KeySecret from soliscloud API Key ID = KeyID from [...]

Hi,
I am absolute beginner so I beg your pardon if the question is trivial.
I managed to get the 19 digits station ID.
However, what's the difference between: "solis cloud API Secret" and "soliscloud API Key ID" ?
Thank you!

PR #233 (2022-12-16): 216 solis energy today entity became unavailable
Closes #183 , closes #216 , closes #230