hultenvp/solis-sensor #186: New entity names created when installing new release 3.07 rather than reuse of existing

snigehere, 2022-10-31:
This is a follow up to [https://github.com/hultenvp/solis-sensor/issues/177](https://github.com/hultenvp/solis-sensor/issues/177) related to duplicate names

**Some background:**
In this issue I described an problem where the entity names were duplicated when the new release was installed.
In recovering from that I deleted the old names and renamed the new ones to match the old which worked around the issue.  

In addition there was a further issue where it seemed to either stop on reload of HA or just stop during the day.  As a result I tried to revert back to the last 2.x release but, maybe due to user error, I ended up with 3.05 again ....  To solve this I removed the integration and removed it from HACS.  I tried to re-install 2.5 but because of the level of HA core I could not do that because the change to how credentials were stored ... in the end I opted for a fresh install of 3.05  ... keeping up?

**A fresh install**
After installing the fresh version of 3.05 I found that the entity names had changed from "my_solis_inverter_xxx" to "solis_xxx" and as a result the history of generation was not visible in the energy dashboard.  Having installed I could not find where I could change this name via the integration - I tried the rename option but that called something completely different.

To work around the I decided to delete the old sensors and rename the new sensors back to the old format ... for example - 
sensor.solis_energy_today was renamed to sensor.my_solis_inverter_energy_today meaning I could again see the history and the solar generation continued to accumulate in the energy dashboard so now its working again.

**Release 3.07 install**
I now see that 3.07 is released so opt to install it ... and now I have a whole bunch of new sensors again in the format "solis_xxx" and again the energy dashboard is compromised

To try and find how I could change the names in the integration I deleted the integration and reinstalled it and see the root cause of the issue I am having.

In the old configuration file I see I had platform and name
`#solis logon details    
  - platform: solis
    name: "My Solis Inverter"
    portal_domain: xxxx <snip>
`
In the new configuration I only have "platform"  .. it seems the "name" component has been dropped and so my entities all got renamed based on the platform value... 

To solve this in the short term I gave my integration the platform name as "My Solis Inverter" and now I have the entities correctly named via the integration. 

Hop this is all clear .. 

I don't know how many people will have set different values for platform compared to name but maybe there will be a few who will be impacted by this ...

Thanks for your time with this integration - Please let me know if I you need any additional information

metcrl, 2022-10-31:
When updated to 3.0.7 there is again duplicate sensors

hultenvp, 2022-10-31:
Thanks for the elaborate description @snigehere.

From your description I do read that you are referring to the platform name description in the config UI
[image]

This entry is what is the actual "name" field in the config.yaml.
It allows you to give your platform (inverter) a non-default name that's used in the entity names.

Would the following phrasing make things clearer and actually solve your issue?
[image]

snigehere, 2022-11-01:
Thanks for the clarification - yes, for me the change would have helped - it implies its my choice rather than something more formal within HA.

Jonnehs, 2022-11-01:
Sensors have duplicated for me as well, is there a fix?

snigehere, 2022-11-01:
@Jonnehs 
> Sensors have duplicated for me as well, is there a fix?
I documented the issue of duplicate names in ticket #177 which got closed as resolved .. perhaps that needs re-opening again if the issue you have is as described in that ticket - This ticket was more about names changing (eg different name) rather than names duplicated (eg added a _2 to the existing names)

hultenvp, 2022-11-01:
#177 does fix the duplicate issue, but harm done can unfortunately not be automatically corrected.
The duplicate can be removed manually. 
Look which ones of the duplicates is active and then delete the inactive ones. If the active ones have the _2 postfix and you don't like that then you can rename them after you deleted the inactive ones.
Always backup before doing this. Energy dashboard should be fine, but best to double check

PR #192 (2022-11-03): 186 new entity names created when installing new release 307 rather than reuse of existing
closes #186
closed 2022-11-03 by hultenvp