hultenvp/solis-sensor #190: Error setting up entry Station XXXXXX for solis

noharm1010, 2022-11-02:
**Describe the bug**
SolisCloud Plant Monitoring API was activated yesterday.
I configured Ginlong Solis PV portal integration v3.0.7

I am abled to connect to the portal but get this error message:

Deze fout is ontstaan door een aangepaste integratie.

Logger: custom_components.solis.soliscloud_api
Source: custom_components/solis/soliscloud_api.py:193
Integration: Solis Inverter (documentation, issues)
First occurred: 15:06:10 (4 occurrences)
Last logged: 15:09:06

No inverters found

-----
Deze fout is ontstaan door een aangepaste integratie.

Logger: homeassistant.config_entries
Source: custom_components/solis/__init__.py:70
Integration: Solis Inverter (documentation, issues)
First occurred: 10:58:58 (2 occurrences)
Last logged: 15:11:49

Error setting up entry Station XXXXXX for solis
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 365, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/solis/__init__.py", line 70, in async_setup_entry
    portal_version = config[CONF_PORTAL_VERSION]
KeyError: 'portal_version'

**Config**
HA

**Versions**
HA Version an deployment 9.2
HACS version  1.28.3
--

<br class="Apple-interchange-newline">
Integration version

**Debug logs**

Logger: homeassistant.config_entries
Source: custom_components/solis/__init__.py:70
Integration: Solis Inverter (documentation, issues)
First occurred: 10:58:58 (2 occurrences)
Last logged: 15:11:49

Error setting up entry Station XXXXX for solis
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 365, in async_setup
    result = await component.async_setup_entry(hass, self)
  File "/config/custom_components/solis/__init__.py", line 70, in async_setup_entry
    portal_version = config[CONF_PORTAL_VERSION]
KeyError: 'portal_version'

hultenvp, 2022-11-03:
Hey @noharm1010 

Did you configure via the UI or did you upgrade from an older version and import from configuration.yaml?
It looks like the config misses the selection what service to use.

closed 2022-11-03 by hultenvp

hultenvp, 2022-11-03:
Apparently it is possible to proceed without selection one of the options. Now defaulting to Ginlong if that happens