hultenvp/solis-sensor #197: Credentials not accepted by the integration UI (but are correct)

terppatyyppi, 2022-11-04:
Hello all,

I' having trouble getting past the Soliscloud credentials on this Home assistant HACS integration for Solis PV Inverters.

I do have the API key and secret, as well as the station ID and the portal URL...but despite all of that being in the right place in the UI install, it will not get me past this -> "Cannot login with provided URL and credentials". The UI only asks for a username or e-mail, but no password is ever asked for.

Can someone help me please? Would really appreciate it.

terppatyyppi, 2022-11-04:
As I see this is not a bug, so I'll close my comment so it won't bother anyone here. In any case, I still have this issue and would still appreciate some help for dummies. Apologies.

closed 2022-11-04 by terppatyyppi

hultenvp, 2022-11-04:
This is unfortunately not a bug. The API simply rejects your credentials.
Either you made a mistake somewhere in entering username, key and secret or you are victim of the current API lockdown:
https://github.com/hultenvp/solis-sensor#soliscloud
Many people are locked out. Only if you had a working key and secret before mid September then the API likely still returns some data. Others are less lucky.

If you want to test with some more insights then enable debug or run the test app:   https://github.com/hultenvp/solis-sensor/blob/master/test/soliscloud_test.py

terppatyyppi, 2022-11-04:
Thanks so much for your e-mail.
Well, that explains this then. I had misunderstood that they had just
stopped from delivering new API keys and secrets based on requests....but
if it really is so that you had to have your key and secret already IN USE
before mid Sept, then I'm propably locked out.
I have my key and secret, it's just they were never used before this.

Thanks again.

hultenvp, 2022-11-04:
All the info we have is anecdotal. 
The people that report they can still get data seem to be people who had it working before they locked down the API. So I really can't tell you for sure whether you're part of the unlucky ones or whether there's a small mistake in your config.

If you want to spend time on it and do know how to run a python script then try to run the test app. Fill in your credentials there and give it a spin. It will output JSON data. That data will tell you for sure what the problem is.

terppatyyppi, 2022-11-04:
Hmm, it seems on Oct 26 they have reopened the access for end users:
https://solis-service.solisinverters.com/support/solutions/articles/44002212561-api-access-soliscloud

Maybe they have at the same time changed something?

Sent from mobile

Du7chManiac, 2022-11-18:
> All the info we have is anecdotal. The people that report they can still get data seem to be people who had it working before they locked down the API. So I really can't tell you for sure whether [...]

Test script works for me, returns data perfectly fine. Yet I can't setup the integration because of the error: Cannot login with provided URL and credentials