hultenvp/solis-sensor #198: Total Energy only increases in steps of 10

cvroque, 2022-11-06:
**Describe the bug**

Ever since a few days ago, the integration stopped increasing the energy total immediately and instead is just increasing it in steps of 10:

[image]

It seems to have changed on November 3, I also have the latest updates installed. Anyone else having the same problem?

**Config**
UI configuration

**Versions**
Home Assistant 2022.11.1 on HA OS x86
HACS version: 1.28.3
Integration version: v3.0.8

**Screenshots**
If applicable, add screenshots to help explain your problem.

Many steps on Nov 3:

[image]

And after that, only increases of 10.

hultenvp, 2022-11-06:
Can you add debug logs?
Are you using the Ginlong platform or the soliscloud?

cvroque, 2022-11-07:
> Can you add debug logs?
> Are you using the Ginlong platform or the soliscloud?

Hello! I'm using the soliscloud and below are the the logs. I also noticed the daily and monthly energy is increasing normally, while total and year are in steps of 10:
[image]

After checking, I believe it is an API change.

[DEBUG LOGS.txt](https://github.com/hultenvp/solis-sensor/files/9954070/text_53.txt)

hultenvp, 2022-11-10:
My panels don't receive much sunlight lately, but today I collected enough energy to see indeed  energy_total making a 10 kWh step. So confirmed, this is how the API now apparently reports it.

hultenvp, 2022-11-10:
Closing as not a bug

closed 2022-11-10 by hultenvp