hultenvp/solis-sensor #205: Still having trouble with integration not updating on 3.0.10

dpgh947, 2022-11-07:
I am still having issues with 3.0.10. I am using ginlong login, NOT soliscloud.

I am trying to upgrade from 2.3.5 (which is 100% solid), I have gone through this process several times on various levels, now with 3.0.10 and still not working properly:

Running on 2.3.5 (backed out to this countless times now), choose upgrade to latest level.

Reboot - get message saying old config stuff has been imported in to UI, integration says it needs configuring.
Without updating or configuring anything, reboot again - integration now comes up configured, and works, entities seem to update ok. Get message saying old stuff needs removing from configuration.yaml.
Edit config to remove old stuff, reboot again, now the integration is not working. Entities remain "unavailable", nothing happens, nothing in normal logs. Stays like this until I reload the integration, after which it fires up and starts updating.

Every time I restart HA, it comes up "stuck" until reloaded. Backout to 2.3.5 and it works perfectly again. I turned on debug log and I see these messages (edited):

2022-11-07 13:48:29.715 INFO (MainThread) [custom_components.solis.sensor] Scheduling discovery
2022-11-07 13:48:29.715 DEBUG (MainThread) [custom_components.solis.service] Scheduling discovery in 1 seconds.
2022-11-07 13:48:34.213 DEBUG (MainThread) [custom_components.solis.ginlong_api] Resp = <ClientResponse(https://m.ginlong.com/cpro/login/validateLogin.json?userName=xxxxxxxxxxxxxxxxxxxx&password=xxxxxxxxxxxxxxx&lan=2&domain=https://m.ginlong.com&userType=C) [200 OK]>
...
2022-11-07 13:48:34.215 INFO (MainThread) [custom_components.solis.ginlong_api] Login Successful!
2022-11-07 13:48:34.473 DEBUG (MainThread) [custom_components.solis.ginlong_api] Fetching data for serial: xxxxxxx
2022-11-07 13:48:44.898 DEBUG (MainThread) [custom_components.solis.ginlong_api] Error: Exception: <class 'asyncio.exceptions.TimeoutError'>
2022-11-07 13:48:44.898 INFO (MainThread) [custom_components.solis.ginlong_api] Unable to fetch details for device with ID: xxxx
2022-11-07 13:48:44.898 DEBUG (MainThread) [custom_components.solis.ginlong_api] Could not fetch inverter data, retry loging attempt

When I reload the integration, I then see these debug messages:

...
2022-11-07 14:07:25.171 INFO (MainThread) [custom_components.solis.ginlong_api] Login Successful!
2022-11-07 14:07:25.418 DEBUG (MainThread) [custom_components.solis.ginlong_api] Fetching data for serial: xxx
2022-11-07 14:07:25.879 DEBUG (MainThread) [custom_components.solis.ginlong_api] Fetching data for serial: xxx
2022-11-07 14:07:26.485 DEBUG (MainThread) [custom_components.solis.sensor] Creating inverterstate (xxx)
2022-11-07 14:07:26.486 INFO (MainThread) [custom_components.solis.service] Subscribing sensor to attribute state for inverter xxx
... 
and so on for all the sensors. It then seems to work.

**Versions**
HA Version an deployment 2022.11.1
HACS version 1.28.3
Integration version 3.0.10

dpgh947, 2022-11-08:
Been doing lots of testing on this, it is definitely a timeout problem. Very occasionally, it will come up ok after an HA restart. Most of the time, I get the "DEBUG (MainThread) [custom_components.solis.ginlong_api] Error: Exception: <class 'asyncio.exceptions.TimeoutError'>" message, often this is the last message it puts out, sometimes I also get the "could not fetch inverter data, retry loging attempt" as in my initial report, but after that it does nothing until I reload it. Whatever the cause of the timeout is (I get lots of 10 second timeout type messages during startup), it is not retrying after that, it just gets stuck.

hultenvp, 2022-11-09:
Confirmed, could reproduce. Running tests now

closed 2022-11-10 by hultenvp

dpgh947, 2022-11-11:
Installed 3.1.0, rebooted with debug on, got the timeout, the retry after 1 min was all good, seems fine, thanks.

PR #209 (2022-11-10): 205 still having trouble with integration not updating on 3010
Closes #205