hultenvp/solis-sensor #212: Solis Inverter Energy Today starts the day with an erroneous offset

GillesC, 2022-11-20:
**Describe the bug**
As can be seen from the screenshot, the `Solis Inverter Energy Today` entity always outputs a wrong energy production value at the start of the day (after booting up). The issue seems somewhat related to #128. 

**Config**
N.A.

**Versions**
Home Assistant 2022.11.3
Supervisor 2022.10.2
Operating System 9.3
Frontend 20221108.0 - latest

m.ginlong.com integration

**Screenshots**
[image]

hultenvp, 2022-11-29:
Yes, the ginlong API only resets energy today in the morning when the device comes back online.
The general rule I apply to filter this out is: If it's AM and the inverter is offline return "0".
If it's AM and the inverter is online then return the actual value. If it's PM, return actual value.

What I see in the screenshot is that indeed "0" is returned as long as the sun is not up. But it looks like the inverter merrily continues adding to the energy today of the previous day after it comes back online until _exactly_ 12 o'clock. 
I suspect your time or timezone settings are wrong. I'd check configuration of both the portal and your inverter, but my guess is the inverter is incorrectly configured because of an AM/PM misconfiguration

PieterVandenBerge, 2022-11-29:
So do you create a new entity then to make this work (resetting daily
solar)?
Sorry about the noob question but would be great if you could share the
code for this.
Thanks!

PieterVandenBerge, 2022-11-30:
[image]
I do see the same happening on SOLIS cloud as well. It seems they don't reset immediately at start-up after begin OFF during the night..

[image]
If the battery is powered on, the issue isn't happening.
(My battery is currently switched off as it's charging from the net which is very undesired behavior and that is something that SOLIS is also currently investigating (at least according to my installer)).

hultenvp, 2022-12-01:
> So do you create a new entity then to make this work (resetting daily solar)? Sorry about the noob question but would be great if you could share the code for this. Thanks!

Nope, I simply filter the Energy today data received from the server before updating the sensor entity.
The filtering code can be found here: 
https://github.com/hultenvp/solis-sensor/blob/master/custom_components/solis/service.py#L153-L176

hultenvp, 2022-12-01:
> [image] I do see the same happening on SOLIS cloud as well. It seems they don't reset immediately at start-up after begin OFF during the night..
>
> [image] If the battery is powered on, the issue [...]

Correct. That's why I filter, in the code you also see a 5 min grace period to avoid still catching the final measurement from previous day. 
Behaviour with battery on is different because the battery typically keeps the inverter on past midnight.

GillesC, 2022-12-01:
> Nope, I simply filter the Energy today data received from the server before updating the sensor entity. The filtering code can be found here: [...]

So, as it is already implemented, I'll have to check the timezone to resolve the issue I am seeing now?

GillesC, 2022-12-01:
> Yes, check time and locale settings both on the server and in your inverter. I've seen both scenarios.

Both are filled in correctly. I have checked the HA host and the soliscloud show the same wall clock as expected.

hultenvp, 2022-12-01:
Did you also check the timezone? It can't be coincidence your server data resets at 12 noon instead of 12 midnight.

GillesC, 2022-12-08:
> Did you also check the timezone? It can't be coincidence your server data resets at 12 noon instead of 12 midnight.

Hi, yes, both are set to UTC + 1. It is indeed weird....

At solis Cloud:
[image]
[image]

At HA:
[image]

Any pointers, where to look?

GillesC, 2022-12-08:
I have maybe a bold suggestion. Could we do the following:
- if first value of the day of `INVERTER_ENERGY_TODAY`,  `initial_value=value`
-` value = value - initial_value`, as long as the day is not pasted (or till inverter goes offline)
- if `value` is below previous value (or 0?) and the inverter has not gone offline in the mean time,  `initial_value=value`

GillesC, 2022-12-08:
I was unaware that the timing shown on the soliscloud did not reflect the time used by the inverter.
The inverter time was indeed 12h off.

**For other having this problem: check the manual of your inverter to see how to adjust the time and time zone.**

Thank you for your patience @hultenvp!

closed 2022-12-08 by GillesC

hultenvp, 2022-12-08:
Good to see it's solved! I indeed wanted to ask you to also double check the inverter itself.

PieterVandenBerge, 2022-12-09:
I looked into the manual to see how I could change the time zone on my inverter:
[image]
However, there's only mention of time. Not timezone.

When I check then the datalogger on soliscloud:
[image]
It has the same timezone (I'm on Brussels' time).

So not really sure what I still need to change...