hultenvp/solis-sensor #213: Cannot login - Unknown error occurred"

jaypikay, 2022-11-20:
When trying to set up the integration the login fails with "Unknown error occurred". If I use a "wrong" password it actually says its wrong, to rule out that my credentials were not correct.

I have configured the integration through HACS, so no more yaml entries.

**Versions**
HA version: **2022.11.3**
HACS version: **1.28.3**
Integration version: **v3.1.0**

**Logs**
```
homeassistant  | 2022-11-20 15:48:59.278 DEBUG (MainThread) [custom_components.solis.ginlong_api] Resp = <ClientResponse(https://m.ginlong.com/cpro/login/validateLogin.json?userName=*****************@*****.***&password=******************&lan=2&domain=https://m.ginlong.com&userType=C) [200 OK]>
homeassistant  | <CIMultiDictProxy('Server': 'Tengine', 'Content-Type': 'application/json;charset=UTF-8', 'Content-Length': '579', 'Connection': 'keep-alive', 'Date': 'Sun, 20 Nov 2022 14:48:59 GMT', 'Set-Cookie': 'JSESSIONID=*********-****-****-****-************; Path=/; HttpOnly', 'Set-Cookie': 'rememberMe=deleteMe; Path=/; Max-Age=0; Expires=Sat, 19-Nov-2022 14:48:59 GMT', 'Accept-Charset': 'big5, big5-hkscs, cesu-8, euc-jp, euc-kr, gb18030, gb2312, gbk, ibm-thai, ibm00858, ibm01140, ibm01141, ibm01142, ibm01143, ibm01144, ibm01145, ibm01146, ibm01147, ibm01148, ibm01149, ibm037, ibm1026, ibm1047, ibm273, ibm277, ibm278, ibm280, ibm284, ibm285, ibm290, ibm297, ibm420, ibm424, ibm437, ibm500, ibm775, ibm850, ibm852, ibm855, ibm857, ibm860, ibm861, ibm862, ibm863, ibm864, ibm865, ibm866, ibm868, ibm869, ibm870, ibm871, ibm918, iso-2022-cn, iso-2022-jp, iso-2022-jp-2, iso-2022-kr, iso-8859-1, iso-8859-13, iso-8859-15, iso-8859-2, iso-8859-3, iso-8859-4, iso-8859-5, iso-8859-6, iso-8859-7, iso-8859-8, iso-8859-9, jis_x0201, jis_x0212-1990, koi8-r, koi8-u, shift_jis, tis-620, us-ascii, utf-16, utf-16be, utf-16le, utf-32, utf-32be, utf-32le, utf-8, windows-1250, windows-1251, windows-1252, windows-1253, windows-1254, windows-1255, windows-1256, windows-1257, windows-1258, windows-31j, x-big5-hkscs-2001, x-big5-solaris, x-compound_text, x-euc-jp-linux, x-euc-tw, x-eucjp-open, x-ibm1006, x-ibm1025, x-ibm1046, x-ibm1097, x-ibm1098, x-ibm1112, x-ibm1122, x-ibm1123, x-ibm1124, x-ibm1166, x-ibm1364, x-ibm1381, x-ibm1383, x-ibm300, x-ibm33722, x-ibm737, x-ibm833, x-ibm834, x-ibm856, x-ibm874, x-ibm875, x-ibm921, x-ibm922, x-ibm930, x-ibm933, x-ibm935, x-ibm937, x-ibm939, x-ibm942, x-ibm942c, x-ibm943, x-ibm943c, x-ibm948, x-ibm949, x-ibm949c, x-ibm950, x-ibm964, x-ibm970, x-iscii91, x-iso-2022-cn-cns, x-iso-2022-cn-gb, x-iso-8859-11, x-jis0208, x-jisautodetect, x-johab, x-macarabic, x-maccentraleurope, x-maccroatian, x-maccyrillic, x-macdingbat, x-macgreek, x-machebrew, x-maciceland, x-macroman, x-macromania, x-macsymbol, x-macthai, x-macturkish, x-macukraine, x-ms932_0213, x-ms950-hkscs, x-ms950-hkscs-xp, x-mswin-936, x-pck, x-sjis_0213, x-utf-16le-bom, x-utf-32be-bom, x-utf-32le-bom, x-windows-50220, x-windows-50221, x-windows-874, x-windows-949, x-windows-950, x-windows-iso2022jp', 'Via': 'cache8.l2ot7-1[49,0], cache10.de3[192,0]', 'Timing-Allow-Origin': '*', 'EagleId': '4f85b19e16689557390905736e')>
homeassistant  | 
homeassistant  | 2022-11-20 15:48:59.280 INFO (MainThread) [custom_components.solis.ginlong_api] Login Successful!
homeassistant  | 2022-11-20 15:48:59.570 ERROR (MainThread) [aiohttp.server] Error handling request
homeassistant  | Traceback (most recent call last):
homeassistant  |   File "/config/custom_components/solis/ginlong_api.py", line 224, in login
homeassistant  |     data = await self.fetch_inverter_data(next(iter(self._inverter_list)))
homeassistant  | StopIteration
homeassistant  | 
homeassistant  | The above exception was the direct cause of the following exception:
homeassistant  | 
homeassistant  | Traceback (most recent call last):
homeassistant  |   File "/usr/local/lib/python3.10/site-packages/aiohttp/web_protocol.py", line 435, in _handle_request
homeassistant  |     resp = await request_handler(request)
homeassistant  |   File "/usr/local/lib/python3.10/site-packages/aiohttp/web_app.py", line 504, in _handle
homeassistant  |     resp = await handler(request)
homeassistant  |   File "/usr/local/lib/python3.10/site-packages/aiohttp/web_middlewares.py", line 117, in impl
homeassistant  |     return await handler(request)
homeassistant  |   File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 60, in security_filter_middleware
homeassistant  |     return await handler(request)
homeassistant  |   File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 100, in forwarded_middleware
homeassistant  |     return await handler(request)
homeassistant  |   File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
homeassistant  |     return await handler(request)
homeassistant  |   File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 82, in ban_middleware
homeassistant  |     return await handler(request)
homeassistant  |   File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 236, in auth_middleware
homeassistant  |     return await handler(request)
homeassistant  |   File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 136, in handle
homeassistant  |     result = await result
homeassistant  |   File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 180, in post
homeassistant  |     return await super().post(request, flow_id)
homeassistant  |   File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 73, in wrapper
homeassistant  |     result = await method(view, request, data, *args, **kwargs)
homeassistant  |   File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 110, in post
homeassistant  |     result = await self._flow_mgr.async_configure(flow_id, data)
homeassistant  |   File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 280, in async_configure
homeassistant  |     result = await self._async_handle_step(
homeassistant  |   File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 367, in _async_handle_step
homeassistant  |     result: FlowResult = await getattr(flow, method)(user_input)
homeassistant  |   File "/config/custom_components/solis/config_flow.py", line 80, in async_step_credentials_password
homeassistant  |     if await api.login(async_get_clientsession(self.hass)):
homeassistant  | RuntimeError: coroutine raised StopIteration
```

[image]

pappajj, 2022-11-23:
Same issue here

Was working fine through the yaml until I was forced to change!

hultenvp, 2022-11-24:
Thanks for the bug report, clear and very complete. Helps me a lot.
The introduction of the UI based config meant some code has moved to a coroutine. Apparently it triggers now the following situation:
>PEP 479 is enabled for all code in Python 3.7, meaning that StopIteration exceptions raised directly or indirectly in coroutines and generators are transformed into RuntimeError exceptions. (Contributed by Yury Selivanov in bpo-32670.)

My guess is that you see this and others not because you have more than one inverter registered. 
@jaypikay @pappajj : Can you confirm?
If not, can you tell me how you run homeassistant and if you use home assistant OS or another configuration?

Thanks

jaypikay, 2022-11-24:
I am running home assistant as docker service. The installation itself is a relatively fresh install (not much customized, yet).

I only have one inverter. If you need more information or details just ask, I will see what I can safely provide.

pappajj, 2022-11-25:
I have two inverters, but the other is a Goodwe via a different integration.
I had two Solis inverters setup on m.ginlong.com but I have deleted one, same issue

My installation is on a RPI4

jaypikay, 2022-11-25:
Oh my, I just figured out what my problem was. The datalogger failed to start up properly and did not report data. When I checked it, I have only verified the green LED, but not if it is correctly connected and transmitting data.

It seems, that the datalogger has not updated the remote data for an X amount of time, that this causes the first initialization of the integration to fail?

tl:dr: unlugged/replugged the datalogger it started working again.

@pappajj have you checked if your data loggers have sent any data to the remote service recently? Last update timestamp in their dashboard has not been updated for quite some time.

jaypikay, 2022-11-25:
My inverter requires a separate device I have to attach in order to have it send data. Like this one https://www.primesolar.eu/shop/solis-lan-datenlogger-stick/

It depends on the inverters ofcourse.

pappajj, 2022-11-25:
Ahhh silly me 
Of course 
I have had a wifi issue at the dwelling the Solis inverter is ion, so no data has been uploaded to the cloud! 

Fixing this tomorrow so hopefully it will solve the integration also 🙂

hultenvp, 2022-11-25:
Thanks for the updates.
Looking forward to hear if this solves the issue.
I think it might:
The integration filters or any inverter that didn't update for now thank 48 hours.
This is because many users reported stale inverters messing up their sensor list ( for example after a data logger replacement)
Now if your logger stopped logging then you end up with an empty list of inverters, which indeed could trigger the exception.
Maybe some extra logging to make people aware of the filtering, plus catching the exception will help here.

pappajj, 2022-11-25:
Working now for me! Thanks so much

closed 2022-11-26 by hultenvp

hultenvp, 2022-11-26:
Created a new version with extra logging.

PR #220 (2022-11-26): 213 cannot login unknown error occurred
Closes #213 

* Add notification when stale inverter is ignored
* Catch StopIteration