hultenvp/solis-sensor #216: Solis Energy Today entity became unavailable

PieterVandenBerge, 2022-11-23:
**Describe the bug**
Solis Energy Today entity became unavailable. I was using this to track how much kWh is generated on the panels.
[image]

Which leads to
[image]

It seems to have started yesterday 22nd of November
[image]

**Versions**
Home Assistant Core Installed version 2022.11.2
Solis integration v3.1.0

***NEXT***
I tried updating HA Core but this seems to have made it worse only:
Home Assistant 2022.11.4
Supervisor 2022.10.2
Operating System 9.3
Frontend 20221108.0 - latest

Now all entities seem to have become unavailable.
[image]

Let me know if any additional info would be required.

PieterVandenBerge, 2022-11-23:
I reverted back to Home Assistant Core 2022.11.2 and now all entities are available again including Solis Energy Today.
However, I'm now not convinced I should update HA Core. 

Looking forward to any feedback.

hultenvp, 2022-11-24:
I'm on 2022.11.3. No problems there. Will update now and see if I can reproduce your issue.

hultenvp, 2022-11-24:
My upgrade was flawless, all works fine. 
I'll need a debug log to investigate.

Chris370-arch, 2022-11-28:
Had this same issue (2022.11.4), performed a reload and it stopped working. A restart of the integration did not resolve. I had to perform a proper power cycle of HA.

hultenvp, 2022-11-29:
Maybe related to the timeout discussed in #183

hultenvp, 2022-12-16:
I'll close this issue, further issues after release 3.2.0 please report in #232

PR #233 (2022-12-16): 216 solis energy today entity became unavailable
Closes #183 , closes #216 , closes #230
closed 2022-12-16 by hultenvp