hultenvp/solis-sensor #222: New Entities

viking2010, 2022-11-27:
Hi,

Just wondering how easy it would be for you to add some additional entities in to what's available currently?  For example, Battery health?  There's a few things that could be good to add but before I compile a list, just wanted to enquire.  

Also, I have noticed that the entity "sensor.solis_inverter_daily_grid_energy_used" isn't related to the grid at all and should be "homeLoadTodayEnergy" (taken from the soliscloud_test.py.  

Many thanks

viking2010, 2022-11-27:
If the following from the API's could be added then that would be great (copy/pasted from the solarman_test.py output):

"gridSellEnergy": 0,
"gridSellEnergyStr": "kWh",
"gridSellMonthEnergy": 0,
"gridSellMonthEnergyStr": "kWh",
"gridSellTodayEnergy": 0.0,
"gridSellTodayEnergyStr": "kWh",
"gridSellTotalEnergy": 1.339,
"gridSellTotalEnergyStr": "MWh", (prob better as kWh??)

"gridPurchasedTodayEnergy": 1.4,
"gridPurchasedTodayEnergyStr": "kWh",
"gridPurchasedTotalEnergy": 823.0,
"gridPurchasedTotalEnergyStr": "kWh",

"storageBatteryCurrent": -0.3,
"storageBatteryCurrentStr": "A",
"storageBatteryVoltage": 49.4,
"storageBatteryVoltageStr": "V",

I'm going to raise a ticket with Solis with regards to these API's as they're always blank for some reason. Seem to be several like that:
"gridPurchasedEnergy": 0,
"gridPurchasedEnergyStr": "kWh",
"gridPurchasedYearEnergy": 0,
"gridPurchasedYearEnergyStr": "kWh",
"gridPurchasedYesterdayEnergy": 0.0,
"gridPurchasedYesterdayEnergyStr": "kWh",
"gridPurchasedMonthEnergy": 0,
"gridPurchasedMonthEnergyStr": "kWh",
"gridSellYearEnergy": 0,
"gridSellYearEnergyStr": "kWh",
"gridSellYesterdayEnergy": 0.0,
"gridSellYesterdayEnergyStr": "kWh",

viking2010, 2022-11-28:
> Also, I have noticed that the entity "sensor.solis_inverter_daily_grid_energy_used" isn't related to the grid at all and should be "homeLoadTodayEnergy" (taken from the soliscloud_test.py.

Looking over the soliscloud_api.py file, it looks to be accurate, so I can only assume it's me that's wrong here lol!

hultenvp, 2022-11-29:
When Lucidity and myself worked on Soliscloud support we wanted to keep the sensor names identical to the Ginlong portal.
We described the mapping here: https://github.com/hultenvp/solis-sensor/discussions/71
A lot of the sensors you want extra can be added. You can add them yourself to experiment if you want and create a pull request if you have it working.
Just add them in soliscloud API to the right list (see API documentation or test.py which endpoint to find them)
If the sensor does not yet exists then you need to also define the key in ginlong_const.py and add the sensor to the list in const.py.

viking2010, 2022-11-29:
Ahh, that makes sense, and I think it also answers my question in the Solis Beta discussion about them not seeming to work when I update the .py scripts. I'm just getting to grips with Python and I don't think I'd updated the ginlon_const.py file as I'm using the solis side of things. But I take it the Ginlong_const.py is also required whether you use solis or not?

hultenvp, 2022-11-29:
It's a list of shared constant keys and maps the API output to the actual sensors. If they key is not supported by ginlong then it's simply skipped, so you can extend for SolisCloud only

viking2010, 2022-11-29:
Cool, thanks for that. Happy to close this off 👍

closed 2022-11-29 by viking2010

jmason, 2022-11-29:
@viking2010 are you going to open a pull request to add those metrics you were requesting? I was planning to add a couple of those too :)

viking2010, 2022-11-29:
@jmason I'm not 100% sure how to do that. Let me have ago at adding them in first and see what results I get. If things go well, then I'll then look at doing a pull request. I have used a few programming languages in my time but I'm still getting to grips with Python. Whilst I can get the gist what the code is doing, knowing exactly how things work is another matter entirely! Don't expect instant success, but I will drop a response here if I do figure it all out!!

jmason, 2022-12-08:
I have had a go at adding some of these here: https://github.com/hultenvp/solis-sensor/pull/227