hultenvp/solis-sensor #223: WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found

barthoefs, 2022-12-01:
`2022-12-01 02:29:26.837 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found
2022-12-01 02:30:26.967 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found
2022-12-01 02:31:27.082 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found
2022-12-01 02:32:27.214 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found
2022-12-01 02:33:27.380 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found
2022-12-01 02:49:22.618 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found
2022-12-01 02:50:22.744 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found
2022-12-01 02:51:22.861 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found
2022-12-01 02:52:22.984 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found
2022-12-01 02:53:23.108 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found
2022-12-01 02:57:24.748 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found
2022-12-01 03:01:25.658 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found
2022-12-01 03:02:25.790 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found
2022-12-01 03:03:25.914 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found
2022-12-01 03:04:26.046 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found
2022-12-01 03:05:26.178 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found
2022-12-01 03:06:26.304 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found
2022-12-01 03:07:26.428 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found
2022-12-01 06:17:13.337 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found
2022-12-01 06:19:22.708 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found
2022-12-01 06:20:22.828 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found
2022-12-01 06:21:22.958 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found
2022-12-01 06:22:23.095 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found
2022-12-01 07:22:23.035 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found
2022-12-01 08:01:39.672 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found`

Is this fixable? What is the default polling for the API?

hultenvp, 2022-12-01:
Would be great if you could dump a debug log, but I assume you have the same issue as others discussing here: https://github.com/hultenvp/solis-sensor/issues/183#issuecomment-1330424320

hultenvp, 2022-12-08:
Interestingly, when I extend timeouts to 100sec I now see the same issue.
```
2022-12-01 19:57:23.463 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Error from URI (/v1/api/inveterList) : 0, message='Attempt to decode JSON with unexpected mimetype: text/html', url=URL('https://www.soliscloud.com:13333/v1/api/inveterList')
2022-12-01 19:57:23.479 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found
2022-12-01 19:57:23.509 WARNING (MainThread) [custom_components.solis.service] Failed to discover, scheduling retry in 300 seconds.
2022-12-01 19:57:23.564 DEBUG (MainThread) [custom_components.solis.service] Scheduling discovery in 300 seconds.
```

jmason, 2022-12-09:
I added some extra logging to log the raw text in order to debug this mime type error.  here's what I got:

```
homeassistant         | 2022-12-09 11:12:29.595 INFO (MainThread) [custom_components.solis.soliscloud_api] Output from URI (/v1/api/inveterList) : <html>
homeassistant         | <head><title>502 Bad Gateway</title></head>
homeassistant         | <body bgcolor="white">
homeassistant         | <center><h1>502 Bad Gateway</h1></center>
homeassistant         | <hr><center>nginx</center>
homeassistant         | </body>
homeassistant         | </html>
```

That's a load-balancer generated error page.  I would say that's a problem on the backend with the Solis Cloud API servers.

barthoefs, 2022-12-09:
Yeah seems like Solis Cloud api issue today

viking2010, 2022-12-09:
I'm seeing the same at my end too. The solistest.py is pulling back data ok, but the integration isn't and reports the same errors as above.

rossdargan, 2022-12-10:
I get this issue following a reboot for a bit (maybe 30 - 60 minutes). I wonder if there is more to this. It does eventually come back up.

hultenvp, 2022-12-12:
@jmason, @viking2010 , @rossdargan : I recently had some connectivity troubles myself. Running now with extra logs to get more clarity, but I saw both 408 and 502 errors.
* I'm pretty sure the time of my VM was off by more than 15 minutes when I got the 408 errors. Some API calls require a timestamp and that timestamp is not allowed to be off for more than 15 minutes from server time. IT's not specified anywhere, but I'm pretty sure they throw a 408 if it is. Once confirmed I will add some decent logging to make it clear in the error log.
* The 502 is more elusive, it comes and goes. The startup sequence fires of 3 calls immediately after each other. It could be that the load balancer can't scale up well, for example if a lot of HA users update their systems around the same time and cause some peak load. I can imagine is that the increased retry delay at some point gives enough breathing room again to deal with the load. another idea could be that the server doesn't like the burst in calls and decides it hates the session doing that and puts it on blacklist for a while. Anyway, all still theory. Any help here with leads, insights, root cause would be appreciated.

jmason, 2022-12-12:
I think you are probably on the money regarding the 408 error being related to the VM time drift.
Also that's a good theory about the 502s being related to the backend disliking repeated API calls too close together; that would explain how the same API calls work ok from the test app, but then give a 502 once called via the library.

hultenvp, 2022-12-13:
Thanks for the response. I'd like to somehow get confirmation on the 502 theory. Let's see if I can use the test app to mimic a single session making to frequent calls.

hultenvp, 2022-12-16:
I modified the test app to also use aiohttp and one single session. If I make 10 calls in parallel I get a 429:
```{"error": "Too Many Requests", "message": "too much request 2 times in 1000 millseconds", "path": "/v1/api/inveterList", "status": 429, "timestamp": 1671179957966}```

So it looks like the 502 is probably not caused by frequent calls.
It still could be a load-balancer issue, but then why it only causes issues with the integration and not the test app.....

hultenvp, 2022-12-16:
I'll close this issue, further issues after release 3.2.0 please report in #232

closed 2022-12-16 by hultenvp