hultenvp/solis-sensor #224: Which sensor should I use for Returned to grid in Home Assistant Energy Dashboard?

finalbillybong, 2022-12-03:
As title.

viking2010, 2022-12-06:
Hi @finalbillybong I believe it should be the entity ending in *_daily_on_grid_energy. It's not clear that it's the export but as I understand it, using the Ginlong names, that's how it's referenced. In the SolisCloud_api.py it's shown as:

[image]

Give that a whirl and see how it goes. I'm not exactly generating enough at the moment to see any export values, but looking back at historical information, that's the one where export figures can be seen.

viking2010, 2022-12-06:
.........Although.....having just had a little bit of excess sunlight today, the figure that outputs is out by a factor of 10. 

@hultenvp the API outputs "gridSellTodayEnergy": 0.3," for example, but the soliscloud_api.py has a section commented with: "# Just temporary till SolisCloud is fixed". I guess this may no longer be required? Be interesting to see if this is different for those with the older data logger.

For now @finalbillybong, if you open the workarounds.yaml file in the folder ..\config\custom_components\solis and change the "correct_daily_on_grid_energy_enabled:" to false, then it should correct the value to what it should be.

closed 2022-12-06 by finalbillybong