hultenvp/solis-sensor #229: Error on stationdetail - Please contact the administrator for abnormal data

jonathanmandrews2022, 2022-12-11:
Seems to login in from what can tell but as soon as tries to get station detail it comes up with a Chinese server error message of 'Please contact the administrator for abnormal data'

This has not worked and checked plant id taken from here 

https://www.soliscloud.com/#/station/stationdetail_1?id=1298491919448******

Home Assistant 2022.12.1
Supervisor 2022.11.2
Operating System 9.4

HACS Version: 1.28.4
Add in version: v3.1.1

2022-12-11 12:39:49.177 DEBUG (MainThread) [custom_components.solis.soliscloud_api] workarounds: {'correct_daily_on_grid_energy_enabled': True}
2022-12-11 12:39:49.378 INFO (MainThread) [custom_components.solis.soliscloud_api] Login successful
2022-12-11 12:39:49.379 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Found inverters: ['60310232282000**']
2022-12-11 12:39:49.379 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: 60310232282000**
2022-12-11 12:39:49.504 INFO (MainThread) [custom_components.solis.soliscloud_api] /v1/api/stationDetail responded with error: 1:数据异常 请联系管理员
2022-12-11 12:39:49.504 DEBUG (MainThread) [custom_components.solis.ginlong_base] AttributeError, plantName does not exist
2022-12-11 12:39:49.505 INFO (MainThread) [custom_components.solis.soliscloud_api] Failed to acquire plant name, login failed

hultenvp, 2022-12-23:
I have created a new version 3.2.2 that possibly fixes the issue, use at own risk.

Please report logging here if you still encounter issues

jonathanmandrews2022, 2022-12-23:
@hultenvp just to confirm this is now working.

thanks for all your time and effort on this.

closed 2022-12-23 by jonathanmandrews2022