hultenvp/solis-sensor #230: Error during setup of component solis

TooBri, 2022-12-12:
**Describe the bug**
Error during setup of component solis

Hi,

I am trying to add a solis inverter according to the solis sensor integration. The integration is installed via HACS. It are 2 inverters on one location. Both via soliscloud and ginlong the names of the inverters turn up, after filling in the initial details. Directly HA shows an error saying the integration cannot be loaded, and HA must be restarted. After restart, the underneath error turns up in the logs, the same for soliscloud and ginlong. 
Im wondering if there is anything I should change or adapt. 

**Versions**
HA Version an deployment: 
Home Assistant 2022.12.3
Supervisor 2022.11.2
Operating System 9.3

HACS version
1.28.4
Integration version
3.1.1.

**Debug logs**

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 253, in _async_setup_component
    result = await task
  File "/config/custom_components/solis/__init__.py", line 44, in async_setup
    if DOMAIN.lower() == entry["platform"]:
TypeError: string indices must be integers

**Screenshots**
If applicable, add screenshots to help explain your problem.

TooBri, 2022-12-14:
Hi @hultenvp, 

it worked like a charm, all values for both inverters coming trough as supposed. Thanks for your fast reply and support :)

hultenvp, 2022-12-16:
Great! I'll catch the error in the new version so that you do not have to edit after every release ;-)

matthewbyrne, 2023-03-24:
@hultenvp , i'm getting this after upgrading to HA 2023.3.6. Any thoughts?

hultenvp, 2023-04-03:
Hi @matthewbyrne, not sure. Code should just ignore any issues. Can you post the logs?

matthewbyrne, 2023-04-03:
@hultenvp , sorry, it magically fixed itself. 🤷

PR #233 (2022-12-16): 216 solis energy today entity became unavailable
Closes #183 , closes #216 , closes #230
closed 2023-05-26 by hultenvp