hultenvp/solis-sensor #235: Unable to login to Solis cloud

jaakleen, 2022-12-16:
### Discussed in https://github.com/hultenvp/solis-sensor/discussions/226

<div type='discussions-op-text'>

<sup>Originally posted by **robconway** December  7, 2022</sup>
I am unable to login to Solis cloud despite following the info, and I have an inverter installed, so I am unclear at the error. Any ideas?

My log show:

Logger: custom_components.solis.soliscloud_api
Source: custom_components/solis/soliscloud_api.py:193 
Integration: Solis Inverter ([documentation](https://github.com/hultenvp/solis-sensor/), [issues](https://github.com/hultenvp/solis-sensor/issues)) 
First occurred: 07:42:06 (1 occurrences) 
Last logged: 07:42:06

No inverters found</div>

jaakleen, 2022-12-16:
async def _get_station_details(self, plant_id: str) -> dict[str, str] | None:
        """
        Fetch Station Details
        """

        params = {
            'id': plant_id
        }
        result = await self._post_data_json(PLANT_DETAIL, params)

        if result[SUCCESS] is True:
            jsondata : dict[str, str] = result[CONTENT]
            if jsondata['code'] == '0':
                return jsondata
            else:
                _LOGGER.info("%s responded with error: %s:%s",PLANT_DETAIL, \
                    jsondata['code'], jsondata['msg'])
                result = { SUCCESS: True, MESSAGE:'OK', STATUS_CODE:200, CONTENT:{'code': '0', 'msg': 'success', 'data': {'sno': plant_id, 'latitude':0.0, 'longitude':0.0, 'cityStr': ''}}}
                jsondata : dict[str, str] = result[CONTENT]
                return jsondata
        else:
            _LOGGER.info('Unable to fetch details for Station with ID: %s', plant_id)
        return None

jaakleen, 2022-12-16:
> Thanks @jaakleen <https://github.com/jaakleen>,
>
> Can you enable debug and post the log here?

The code I poster is a work-a-round for the issue. The error is stil
logged, i am include it here. But because of the work-a-round it continues

alienatedsec, 2022-12-20:
@hultenvp I started experiencing the same issue

[image]

I removed the integration and readded it today; hence why is only one occurrence

spiderpigsam, 2022-12-21:
HI, I am also having the same issue followed documentation to the letter, tried uninstall integration and re-install.

This error originated from a custom integration.

Logger: custom_components.solis.soliscloud_api
Source: custom_components/solis/soliscloud_api.py:204
Integration: Solis Inverter (documentation, issues)
First occurred: 08:15:26 (1 occurrences)
Last logged: 08:15:26

No inverters found

Sam

spiderpigsam, 2022-12-21:
Here is the debug log extract.

`2022-12-21 10:33:40.351 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Response contains unexpected data: {'success': True, 'code': '1', 'msg': 'æ•°æ®å¼‚å¸¸ è¯·è”ç³»ç®¡ç†å‘˜', 'data': None}
2022-12-21 10:33:40.351 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found`

hultenvp, 2022-12-23:
Fixed in v3.2.2. If you still experience issues, please report in #229

closed 2022-12-23 by hultenvp