hultenvp/solis-sensor #247: Negative value in energy dashboard?

jgrgt, 2023-01-08:
I'm using my solis "Total Lifetime energy output" value in my Home Assistant energy dashboard. I've not checked it in detail for a while, but recently I looked back and noticed this large negative peak in my solar power output:

[image]

I'm running all latest versions right now. This could be caused by and old bug? Or am I using the wrong sensor value? Did some kind of overflow happen?

hultenvp, 2023-01-09:
Are you on ginlong.com or soliscloud, @jgrgt ?

I've not seen this before. It's definitely not an overflow. It could be invalid data from the server, or it could be something in how the energy dash uses the data.
In any case maybe best to use "Energy today" for this, it has a higher resolution.

hultenvp, 2023-01-13:
Are you still observing the issue?

jgrgt, 2023-01-13:
As far as I know, I'm using the `soliscloud` integration. 

I've now switched to using "Energy today". Hoping to avoid the issue in the future. So far nothing is showing up in my energy dashboard, hopefully in a few hours I should see some sunshine being reported.

closed 2023-01-13 by jgrgt