hultenvp/solis-sensor #249: sensor.solis_battery_power is always negative

barthoefs, 2023-01-11:
Battery power shows always as negative (charging and discharging). Battery current does show as positive and negative when charging and discharing (see screenshots).
Is there a way to align this? When current show charging, battery power shows charging?

I am using latest everything.

**Screenshots**
sensor.solis_battery_power
[image].
sensor.solis_battery_current
[image]

barthoefs, 2023-01-12:
This is my current "fix":
template:
  - sensor:
      - name: "Battery Power Corrected"
        unit_of_measurement: W
        device_class: power
		icon: mdi:battery
        state: >-
          {% if states('sensor.solis_battery_current') | float <= 0 %}
            {{ states('sensor.solis_battery_power') }}
          {% else %}
            {{ states('sensor.solis_battery_power') | float | abs }}
          {% endif %}

hultenvp, 2023-01-13:
Interestingly, the code actually already has a fix for wrong bat power sign, when it was always positive.
But in your case, the power seems to be always negative, instead of always positive.
I will extend the fix to also check for the inverse, but maybe it's also good to submit a ticket with Solis that they finally fix their API.

```
            # turn batteryPower negative when discharging (fix for https://github.com/hultenvp/solis-sensor/issues/158)
            try:
                if self._data[BAT_CURRENT] < 0 and self._data[BAT_POWER] > 0:
                    self._data[BAT_POWER] = self._data[BAT_POWER] * -1
            except KeyError:
                pass
```

closed 2023-01-13 by hultenvp

matthewbyrne, 2023-02-05:
I'm still getting this... Though battery is always positive for me. This was fine until I upgraded to the latest a couple of days ago.