hultenvp/solis-sensor #252: Solis energy today the same for all 3 inverters ( shows plant total instead of daily per inverter)

Friesje, 2023-01-20:
Hello,

I recently installed the solis addon in HA and connected it with soliscloud.
I have 3 inverters in the plant.

Now it shows  Energy today 1, 2 and 3 with all the same data ( it shows plant total today instead of today total per inverter).
Other data like actual power is different and correct but only the energy today goes wrong.   How can i edit/fix this ?
I already checked the identities and they are all 3 different as it should be.
[image]

hultenvp, 2023-01-20:
You observe correctly. This was introduced a long while ago (https://github.com/hultenvp/solis-sensor/commit/c8367fcef61747175327c6c68c363bdf8069e451).

At that time energy today in inverter detail did simply not work.
If you know how to do it, could you try to remove the comment on line 69 in Soliscloud_api.py and add a comment on line 133 and 138.

That should fix it for you. If you can report back if it actually works then I can revert the change.

Friesje, 2023-01-20:
Hello,

I wasn't sure what you neat so i tried something.

Rule 68:          INVERTER_ENERGY_TODAY:            ['eToday', float, 2], # Moved to PLANT_DETAIL   was disabled,  i enabled this.

I disabled rule:
132:         INVERTER_ENERGY_TODAY:            ['dayEnergy', float, 2],
and
137:         INVERTER_ENERGY_TODAY:            ['dayEnergy', float, 2]

Now it seems to work!
is this what you meant ?

hultenvp, 2023-01-23:
Yes, this is exactly what I meant. Thanks for confirming.

I'll see if I can fix this in the next update in a way that allows fallback for those that possibly still have issues.
In the meantime you can keep this fix in place and enjoy the correct sensor data ;-)

PR #255 (2023-01-27): 252 solis energy today the same for all 3 inverters shows plant total instead of daily per inverter
Closes #252
closed 2023-01-27 by hultenvp