hultenvp/solis-sensor #254: Since Yesterday, the energy_today sensor show "unknown"

SruiSan, 2023-01-24:
**Describe the bug**
Since yesterday, the energy_today sensor doesn't show a value

[image]

I try putting on "True" the correct_daily_on_grid_energy_enabled: value in workarrounds.yaml, but the result was the same.

**Config**
Standard config

**Versions**
HASS Version: Home Assistant 2023.1.7
HACS version 1.30.0
Integration version: v3.3.0

**Debug logs**
24/01/2023	11:09:28.900	DEBUG (MainThread) [custom_components.solis.soliscloud_api] workarounds: {'correct_daily_on_grid_energy_enabled': True}
24/01/2023	11:09:36.415	DEBUG (MainThread) [custom_components.solis.service] Scheduling discovery in 1 seconds.
24/01/2023	11:09:38.344	DEBUG (MainThread) [custom_components.solis.soliscloud_api] Found inverters: ['<MY_INVERTER_ID>']
24/01/2023	11:09:38.344	DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: <MY_INVERTER_ID>
24/01/2023	11:09:42.033	DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: <MY_INVERTER_ID>
24/01/2023	11:09:44.229	DEBUG (MainThread) [custom_components.solis.sensor] Creating inverterpowerstate (<MY_INVERTER_ID>)
24/01/2023	11:09:44.229	DEBUG (MainThread) [custom_components.solis.sensor] Creating inverterstate (<MY_INVERTER_ID>)
24/01/2023	11:09:44.229	DEBUG (MainThread) [custom_components.solis.sensor] Creating timestampmeasurement (<MY_INVERTER_ID>)
24/01/2023	11:09:44.229	DEBUG (MainThread) [custom_components.solis.sensor] Creating temperature (<MY_INVERTER_ID>)
24/01/2023	11:09:44.229	DEBUG (MainThread) [custom_components.solis.sensor] Creating dcinputvoltagepv1 (<MY_INVERTER_ID>)
24/01/2023	11:09:44.230	DEBUG (MainThread) [custom_components.solis.sensor] Creating dcinputvoltagepv2 (<MY_INVERTER_ID>)
24/01/2023	11:09:44.230	DEBUG (MainThread) [custom_components.solis.sensor] Creating dcinputvoltagepv3 (<MY_INVERTER_ID>)
24/01/2023	11:09:44.230	DEBUG (MainThread) [custom_components.solis.sensor] Creating dcinputcurrentpv1 (<MY_INVERTER_ID>)
24/01/2023	11:09:44.230	DEBUG (MainThread) [custom_components.solis.sensor] Creating dcinputcurrentpv2 (<MY_INVERTER_ID>)
24/01/2023	11:09:44.230	DEBUG (MainThread) [custom_components.solis.sensor] Creating dcinputcurrentpv3 (<MY_INVERTER_ID>)
24/01/2023	11:09:44.230	DEBUG (MainThread) [custom_components.solis.sensor] Creating dcinputpowerpv1 (<MY_INVERTER_ID>)
24/01/2023	11:09:44.230	DEBUG (MainThread) [custom_components.solis.sensor] Creating dcinputpowerpv2 (<MY_INVERTER_ID>)
24/01/2023	11:09:44.230	DEBUG (MainThread) [custom_components.solis.sensor] Creating dcinputpowerpv3 (<MY_INVERTER_ID>)
24/01/2023	11:09:44.230	DEBUG (MainThread) [custom_components.solis.sensor] Creating acoutputvoltage1 (<MY_INVERTER_ID>)
24/01/2023	11:09:44.230	DEBUG (MainThread) [custom_components.solis.sensor] Creating acoutputcurrent1 (<MY_INVERTER_ID>)
24/01/2023	11:09:44.231	DEBUG (MainThread) [custom_components.solis.sensor] Creating actualpower (<MY_INVERTER_ID>)
24/01/2023	11:09:44.231	DEBUG (MainThread) [custom_components.solis.sensor] Creating acfrequency (<MY_INVERTER_ID>)
24/01/2023	11:09:44.231	DEBUG (MainThread) [custom_components.solis.sensor] Creating energytoday (<MY_INVERTER_ID>)
24/01/2023	11:09:44.231	DEBUG (MainThread) [custom_components.solis.sensor] Creating energythismonth (<MY_INVERTER_ID>)
24/01/2023	11:09:44.231	DEBUG (MainThread) [custom_components.solis.sensor] Creating energythisyear (<MY_INVERTER_ID>)
24/01/2023	11:09:44.231	DEBUG (MainThread) [custom_components.solis.sensor] Creating energytotal (<MY_INVERTER_ID>)
24/01/2023	11:09:44.231	DEBUG (MainThread) [custom_components.solis.sensor] Creating batpower (<MY_INVERTER_ID>)
24/01/2023	11:09:44.231	DEBUG (MainThread) [custom_components.solis.sensor] Creating batvoltage (<MY_INVERTER_ID>)
24/01/2023	11:09:44.231	DEBUG (MainThread) [custom_components.solis.sensor] Creating batcurrent (<MY_INVERTER_ID>)
24/01/2023	11:09:44.231	DEBUG (MainThread) [custom_components.solis.sensor] Creating batcapacityremaining (<MY_INVERTER_ID>)
24/01/2023	11:09:44.231	DEBUG (MainThread) [custom_components.solis.sensor] Creating battotalenergycharged (<MY_INVERTER_ID>)
24/01/2023	11:09:44.232	DEBUG (MainThread) [custom_components.solis.sensor] Creating battotalenergydischarged (<MY_INVERTER_ID>)
24/01/2023	11:09:44.232	DEBUG (MainThread) [custom_components.solis.sensor] Creating batdailyenergycharged (<MY_INVERTER_ID>)
24/01/2023	11:09:44.232	DEBUG (MainThread) [custom_components.solis.sensor] Creating batdailyenergydischarged (<MY_INVERTER_ID>)
24/01/2023	11:09:44.232	DEBUG (MainThread) [custom_components.solis.sensor] Creating griddailyongridenergy (<MY_INVERTER_ID>)
24/01/2023	11:09:44.232	DEBUG (MainThread) [custom_components.solis.sensor] Creating griddailyenergypurchased (<MY_INVERTER_ID>)
24/01/2023	11:09:44.232	DEBUG (MainThread) [custom_components.solis.sensor] Creating griddailyenergyused (<MY_INVERTER_ID>)
24/01/2023	11:09:44.232	DEBUG (MainThread) [custom_components.solis.sensor] Creating gridmonthlyenergypurchased (<MY_INVERTER_ID>)
24/01/2023	11:09:44.232	DEBUG (MainThread) [custom_components.solis.sensor] Creating gridyearlyenergypurchased (<MY_INVERTER_ID>)
24/01/2023	11:09:44.232	DEBUG (MainThread) [custom_components.solis.sensor] Creating gridtotalongridenergy (<MY_INVERTER_ID>)
24/01/2023	11:09:44.232	DEBUG (MainThread) [custom_components.solis.sensor] Creating gridpowergridtotalpower (<MY_INVERTER_ID>)
24/01/2023	11:09:44.233	DEBUG (MainThread) [custom_components.solis.sensor] Creating gridtotalconsumptionpower (<MY_INVERTER_ID>)
24/01/2023	11:09:44.233	DEBUG (MainThread) [custom_components.solis.sensor] Creating gridtotalenergypurchased (<MY_INVERTER_ID>)
24/01/2023	11:09:44.233	DEBUG (MainThread) [custom_components.solis.sensor] Creating gridtotalenergyused (<MY_INVERTER_ID>)
24/01/2023	11:09:44.233	DEBUG (MainThread) [custom_components.solis.sensor] Creating batstateofhealth (<MY_INVERTER_ID>)
24/01/2023	11:09:44.233	DEBUG (MainThread) [custom_components.solis.sensor] Creating socChargingSet (<MY_INVERTER_ID>)
24/01/2023	11:09:44.233	DEBUG (MainThread) [custom_components.solis.sensor] Creating socDischargeSet (<MY_INVERTER_ID>)
24/01/2023	11:09:44.233	DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 0:01:00, at 2023-01-24 11:10:44.233731+00:00
24/01/2023	11:10:44.234	DEBUG (MainThread) [custom_components.solis.soliscloud_api] Fetching data for serial: <MY_INVERTER_ID>
24/01/2023	11:10:46.697	DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 0:02:53.056150, at 2023-01-24 11:13:39.753456+00:00

**Screenshots**
[image]

Any idea if this is a bug/error or misconfiguration for my side (nothing was changed for me, I only update the component, hass and hacs when they release.

tobes49, 2023-01-24:
Just to add mine is doing the same, (was saying unknown)  but it does update after 3 to 4 hours which then throws the energy dashboard out
[image]

svetj, 2023-01-25:
> A bit of moaning part of my comment:
> 
> The Solis Cloud is so unreliable that using it for any sort of integration is waste of time and energy. As much as this integration is great, and can work for many with no issues, it relies on unstable and unpredictable API, which makes the whole Solis platform the worst I have been working with. They have great products, but horrible support and service. In addition, it could be fine for one and don't work for many at the same time. Fixing those issues requires raising tickets and waiting for several days before updating.
> 
> I stopped using (or I don't rely on) the Solis Cloud (for the same reason as you described above), and moved to native Modbus which is feeding another Home Assistant integration. That connection is stable and free of any API, or cloud reliance.
> 
> [image]

How do you moved to native Modbus?
I'm having the same issues :-(

alienatedsec, 2023-01-25:
> How do you moved to native Modbus?

@svetj you either have a specific data logger e.g., https://www.ginlong.com/accessories9/S2_WL_ST_gl.html or you build a custom connection via supported devices e.g., https://github.com/wills106/homeassistant-solax-modbus/wiki/Compatible-RS485-Adaptors

In both cases, you can use the above Solax Modbus integration (it supports Solis too) or HA native Modbus support through this repo - https://github.com/fboundy/ha_solis_modbus  

When using the S2-WL-ST data logger, it will keep your solis cloud updated and working as designed as well as the local functions from the integration. I don't recommend DLS-W or DLS-L data loggers as those are no longer supported by Solis' remote control function. It looks like they are slowly upgrading their cloud offering and the new data logger will be required.

Using a custom adapter will be local only, but at least you have some options to choose from.

hultenvp, 2023-01-27:
Hi @SruiSan, can you check if you revert to the previous version if it works again or if the problem persists?

Also, please enable debugging and attach log here.
[image]

Darsma, 2023-01-27:
Actually it fetches correct numbers but it seems only from 12AM
[image]

hultenvp, 2023-01-27:
@Darsma : Did you update to latest version somewhere between 25 jan 12:00 and 26 jan 12:00?

hultenvp, 2023-01-27:
Then somehow https://github.com/hultenvp/solis-sensor/pull/240 interferes with the filtering out of illegal energy today data in the morning.
Will investigate

tobes49, 2023-01-27:
Mine started this problem with unknown sensor I then updated to try and fix it. 
Happened on 20th Jan at 8.25pm when it changed to unknown 

[image]

As Darsma found it dumps the morning total data at 12.00 -solis cloud shows it slowly building and totals matches at 12.00 so all the data is there just get loaded in 1 batch at midday.

I've turned on logging but will have to wait till tomorrow for it to happen again.

martin-92, 2023-01-27:
I am experiencing the same problems. It has worked flawlessly for months but since the last update the energy_today is unknown. After a few hours it starts working again. Or after a few reboots of HA.

tobes49, 2023-01-27:
> Mine started this problem with unknown sensor I then updated to try and fix it. Happened on 20th Jan at 8.25pm when it changed to unknown

This is my home one:
To be honest thinking about it I did do lots of updates to HA on the 20th but remember having the problem with the solis not working and updated the solis integration to try and get it working again.

My works HA install is running HA 2023.1.7  and Integration 2.3.5 this still works perfectly
[image]

I cant find how to roll back my home one - So ill update my works one to see if it causes the same error.

closed 2023-01-27 by hultenvp

conallob, 2023-01-28:
Another user with the same symptoms.

When should we expect v3.3.1 to be released?

tobes49, 2023-01-28:
I have updated to 3.3.1 last night working fine again. Thank you for the fast fix.
[image]

Conallob
Update HACS, search for solis and update should be listed.