hultenvp/solis-sensor #256: solis_battery_power is always positive value

matthewbyrne, 2023-02-03:
**Describe the bug**
Up until this week the `solis_battery_power` entity was negative while discharging, and positive while charging. Now it seems as though it is always positive.

**Versions**
HA Version an deployment 2022.12.7
HACS version 1.30.1
Integration version v3.3.1

**Screenshots**
[image]

matthewbyrne, 2023-02-03:
Feels like this change broke it
https://github.com/hultenvp/solis-sensor/pull/250

hultenvp, 2023-02-11:
Can you perhaps make a screenshot with both battery power and battery current?

matthewbyrne, 2023-02-12:
[image]

From ~6am today was me charging the battery, and from 8am was me using the battery. As you can see, both are positive.
[image]

matthewbyrne, 2023-02-13:
Correct. Soliscloud. 

I've had to resort to @barthoefs fix from here:
https://github.com/hultenvp/solis-sensor/issues/249#issuecomment-1379991258

hultenvp, 2023-02-14:
Ughh, you are right, there is a mistake in the code. 🤦 
```
            # turn batteryPower negative when discharging (fix for https://github.com/hultenvp/solis-sensor/issues/158)
            try:
                math.copysign(self._data[BAT_POWER],self._data[BAT_CURRENT])
            except KeyError:
                pass
```
should be:
```
            # turn batteryPower negative when discharging (fix for https://github.com/hultenvp/solis-sensor/issues/158)
            try:
                self._data[BAT_POWER] = math.copysign(self._data[BAT_POWER],self._data[BAT_CURRENT])
            except KeyError:
                pass
```

closed 2023-02-14 by hultenvp