hultenvp/solis-sensor #259: Battery wrong data

kisjeth, 2023-02-14:
from tonight the battery reports unrealistic data, with charge values ​​of 55000kwh but no charge anomaly appears on the application.
[image]

V.3.3.1

alienatedsec, 2023-02-14:
Check #258

barthoefs, 2023-02-14:
Same issue here.
The api in HA shows charging at 208,745.0 and the soliscloud app shows 2087.

kisjeth, 2023-02-14:
> Same issue here. The api in HA shows charging at 208,745.0 and the soliscloud app shows 2087.

I create a sensor and change like #258 suggested

barthoefs, 2023-02-14:
> > Same issue here. The api in HA shows charging at 208,745.0 and the soliscloud app shows 2087.
> 
> I create a sensor and change like #258 suggested

Can you show your code please?

viking2010, 2023-02-14:
Hope this helps. Place this in your config.yaml file. I have mine separated so have a templates.yaml file where I stick this. 

```
  - sensor:
    - name: New Battery Power
      unit_of_measurement: kW
      device_class: power
      state: >
        {% set new_battery_power = states('sensor.solis_inverter_battery_power') | float(0) / 100 %} 
        {{ new_battery_power }}
```

kisjeth, 2023-02-14:
> > > Same issue here. The api in HA shows charging at 208,745.0 and the soliscloud app shows 2087.
> > 
> > 
> > I create a sensor and change like #258 suggested
> 
> Can you show your code please?

  - sensor:
      - name: "Carica/Scarica batteria corretto"
        unit_of_measurement: "W"
        state: "{{ (states('sensor.solis_battery_power_2') | float / 100) | round(3) }}"

viking2010, 2023-02-14:
That's even better @kisjeth!

YanisKyr, 2023-02-14:
Am I being dumb? Probably.. Any help would be much appreciated:

https://i.imgur.com/EYWUefq.png

kisjeth, 2023-02-14:
> Am I being dumb? Probably.. Any help would be much appreciated:
> 
> [image]

you have to create a sensor that reads data with the wrong measurement and transforms them with the correct unit of measurement.
to do this, you need to insert the above code into the configuration.yaml to create a new entity for it

`- sensor:
      - name: "CORRECT SENSOR"
        unit_of_measurement: "W"
        entity_id: "CHARGE/DISCHARGE"
        state: {{ (states('sensor.WRONG_SENSOR') | float / 100) | round(3) }}"`

kisjeth, 2023-02-14:
> Am I being dumb? Probably.. Any help would be much appreciated:
> 
> [image]

you have to create a sensor that reads data with the wrong measurement and transforms them with the correct unit of measurement.
to do this, you need to insert the above code into the configuration.yaml to create a new entity for it

```
- sensor:
     - name: "CORRECT SENSOR
        unit_of_measurement: "W"
        entity_id: "CHARGE/DISCHARGE"
        state: {{ (states('sensor.WRONG_SENSOR') | float / 100) | round(3) }}"
```

YanisKyr, 2023-02-14:
For the record, what worked for me in the end is the below in the configuration.yaml

```
template:
  - sensor:
      - name: "Battery Temp FIX"
        unit_of_measurement: "W"
        state: "{{ (states('sensor.PROBLEMATIC_ONE') | float / 100) | round(2) }}"
```

hultenvp, 2023-02-14:
I need to push a fix for #256 anyhow, so I was considering adding a temporary workaround. In the end I decided not to: I've experienced in the past many times that a fix that works for one person does not necessarily work for others.
For example you all report the API returning a wrong unit of kW where W is meant and a value then a factor 100 off.
But I cannot predict what how the API will behave if your battery power goes above 1000W charge/discharge, will the API then suddenly correctly report 1 kW or follow the same bug and report 100 MW or something else?
So let's use the above template sensors until they fixed it.
r3.3.2 will fix the sign, not this issue.

matthewbyrne, 2023-02-16:
This is fixed now, by Solis themselves. See the discussion for details

hultenvp, 2023-02-21:
I don't have a battery. @kisjeth , can you confirm it also works again for you?

kisjeth, 2023-02-22:
> I don't have a battery. @kisjeth , can you confirm it also works again for you?

Yes, sorry. I forgot to close this issue.

closed 2023-02-22 by kisjeth