hultenvp/solis-sensor #262: energy configuration on Home Assistant

tesco222, 2023-02-17:
I have my inverter set up and working over API which is great but I cannot add the realtime power output of the panels to the energy dashboard in Home assistant. 
Solis AC Output Total Power is not selectable in the dropdown of available sensors.
Do I need to edit the sensor somehow?

viking2010, 2023-02-17:
I have the following sensors setup in my Energy Dashboard.....although I don't actually use the grid consumption sensor from Solis as I use the one from my electric meter directly.

- Grid Consumption - sensor.[INVERTER_NAME]_daily_grid_energy_purchased
- Return to Grid: sensor.[INVERTER_NAME]_daily_on_grid_energy
- Solar Panels: sensor.[INVERTER_NAME]_energy_today
- Battery Storage: 
         - Going To Battery: sensor.[INVERTER_NAME]_daily_energy_charged
         - Energy From Battery: sensor.[INVERTER_NAME]_daily_energy_discharged

Hope that helps

hultenvp, 2023-02-21:
Indeed for the energy dashboard one needs to select an energy sensor. Energy today has the best precision, but if your HA is down then you lose data as it resets at midnight. If you have an unstable HA then better use energy_this_month or enrgy_this_year.

See also: https://github.com/hultenvp/solis-sensor/discussions/245

hultenvp, 2023-02-21:
I'm closing this issue as it is not a bug. Feel free to continue support discussions in https://github.com/hultenvp/solis-sensor/discussions/245

closed 2023-02-21 by hultenvp