hultenvp/solis-sensor #269: Solis Power Grid total power does not align with Solis App power

Paulholloway, 2023-03-25:
**Describe the bug**
Trying to replicate the power distribution display (with gauges) as what I have with the Solis app.  Battery power, Solar power, Grid consumption are all correct and tie up with the Solis app. 'Solis Power Grid total power' sensor is always out.  There doesn't appear to be any more power sensors available.  See screenshots.  

**Versions**
Home Assistant 2023.3.6
Supervisor 2023.03.2
Operating System 9.5
Frontend 20230309.1 - latest
HACS
Integration version: | 1.31.0
Frontend version: | 20220906112053

**Screenshots**
[image]
[image]

hultenvp, 2023-04-05:
It could be that there is a Solis Power Grid total power_2 sensor that works.
If HA finds a certain sensors already present then it creates the sensor with a _2 postfix.
Might happen if there's already history present for example because of a previous config or an interrupted config.
Go to settings->Devices & services-> Find Solis and click on "xx Entities"->Look if you can find the Solis Power Grid total power entity and check its state and check also if there are more entities with the same name.
If the sensor is not being updated (red exclamation mark) then the sensor is not detected during startup.

ilbarone87, 2023-04-05:
@Paulholloway do you pull the data from Solis cloud? Because I seem to have the same issue. There’s nothing recording the “Today consumption” data for me as well.

Paulholloway, 2023-04-07:
> It could be that there is a Solis Power Grid total power_2 sensor that works.
> If HA finds a certain sensors already present then it creates the sensor with a _2 postfix.
> Might happen if there's [...]

Apologies- I'm talking about the **consumption**, i.e. sensor.solis_total_consumption_power (above comment edited).  In the above screenshots, the HA 'Consumption' should be reading 147W.

The sensor records data, but doesn't seem to present the same data as per the solis app.  They both take data from Solis cloud so the timing of the readings will/should (?) always align.  The earlier screenshots show most of the data aligns but the sensor.solis_total_consumption_power is often out.  

[image]

There are no '_2' power sensors listed as entities.  All hidden and disabled entities are shown.
[image]

@ilbarone87 Yes I'm pulling this from Solis Cloud.

hultenvp, 2023-04-17:
I'm unsure what can cause that. 
If the data is missing in the API response then de data is indeed 'out'. But why the API does not respond and if that's really what's happening is unclear. 
The integration does regular API calls. So far all issues on SolisCloud are around the API responding with errors, never with incomplete datasets.
I would not be surprised by the way if their own app does not use the same API but a private layer.

closed 2024-01-27 by Paulholloway