hultenvp/solis-sensor #273: Solis Energy Today sometimes has strange values

stecya, 2023-04-13:
Hi, 

Lately, I started experiencing strange results in Solis Energy Total entity values.:
- receiving 0 value after 23:00 which leads to HA Energy Dashboard messing up the plot, as the next value will be counted as energy production. This leads to a huge bar before the midnight
- receiving (what it looks like) yesterday's total value, which once again totally messes up HA Energy Dashboard

The issue started happening after we changed the timezone due to Daylight Saving Time, however, the issue is not reproduced every night, just here and there.

Both HA and Solis have the same timezone in their settings

[image]
[image]

blackspiritus, 2023-04-13:
I'm having the same issue, this morning it was worse then before, it counted the full yield from yesterday multiple hours this morning and my graph is now f*d :( tried to find the values in the statistics database but I'm not sure which one I need to adjust.
[image]
[image]

ilbarone87, 2023-04-13:
There was a maintenance yesterday night in SolisCloud platform and also I can see the dashboard on their website is changed. 
In particular the household consumption has changed.
Can that be the cause?
Might be that the integration need a review to match their changes?

hultenvp, 2023-04-17:
It is fair to believe the maintenance is causing this. Very annoying...

hultenvp, 2023-04-17:
Also be sure your inverter itself has the right time and timezone!

viking2010, 2023-04-17:
I actually had the very same issue a couple of weeks ago, but it only happened on 2 occasions between 11pm and midnight.  I was going to raise an issue on here about it, but given it hasn't happened again, I've not done so, but can now see others have had a similar issue. I just removed the erroneous data using the statistics editor. I did check the data from the Solis Cloud side of things and there were no issues with the data at that time of day. But one evening when I was expecting it to happen again, I was monitoring the API data using the python script. But sods law there were no issues at all, and there haven't been since. I don't think this is an API issue though and more of an HA one....but then that's just a hunch and I have nothing to back that up!

From a conversation I had with a member of the support team at Solis, they don't have very much knowledge about their API. The support guy asked if I had any API implementation instructions as they would love to see it. He then said, and I quote, "our API support sucks!".

blackspiritus, 2023-04-17:
@viking2010 I was now also able to change the values in the editor, I wasn't aware I could change the values in the developer tools Statistics tab, so thank you for the tip, my graph is now looking fine again!

eaasnes, 2023-04-20:
I’m having some trouble with the Energy Today as well. I got two inverters and the Energy Today attribute for each is showing the sum for both inverters instead of showing the separate inverter energy.

hultenvp, 2023-04-21:
@viking2010: The integration forces energy today to '0' until the inverter switches on again in the morning. It should do so from midnight onward. This is done because the APIs report the energy today from the previous day for as long as the inverter is off. This messes up the energy dashboard.
Why it is starting at 23:00 I'm unsure. I use datetime.now(), which gives back local time. The only thing I can imagine is that there is a daylight saving time issue on your local install. Mine resets nicely at midnight (I'm on CEST)
[image]

hultenvp, 2023-04-21:
@eaasnes : please check if in the workarounds.yaml you have : 
```use_energy_today_from_plant: false```

eaasnes, 2023-04-21:
@hultenvp : I haven't changed anything in my workarounds.yaml file. 
Content of my /config/custom_components/solis/workarounds.yaml
```
# Switch on/off SolisCloud workarounds 

#SolisCloud returns daily on grid energy a factor 10 off
correct_daily_on_grid_energy_enabled: false
#Default energy today is taken from inverter detail. Some experience issues, 
#alternatively use energy today from plant, as it is the sum of all #inverters 
#registered under the plant it can only be used with single inverter per plant
use_energy_today_from_plant: false

```
Any other ideas? :-)

curry684, 2023-05-02:
Happened again to me last night, the root cause is clear at least:

[image]

Easiest workaround could be to simply ignore 0-values followed by the exact same value as the last one read - as this issue only happens at midnight it's not ever plausible to be different.

eaasnes, 2023-05-02:
Not quite the same problem for me. Usually both my inverters are reporting the total energy (both inverters together), but sometimes it drops down to the right value for the individual inverter.
[image]

viking2010, 2023-05-17:
Are folks still getting this issue at all? I don't seem to be suffering with this anymore. I did double check my inverters date/time and made sure it was accurate. Whether this made a difference or not, I don't know, but things seem to be more consistent now. I'm also running the latest version of HA so that could also be a factor.

curry684, 2023-05-17:
Saw it only once this month thus far. Seems to be a weird upstream issue.

ilbarone87, 2023-05-20:
I had it as well on the 18-19/05, 3 times has registered a production of 10kwh between 23.00 and 01.00

hultenvp, 2023-05-26:
Is it stable now?

curry684, 2023-05-30:
Installing the update now, will keep a watch on it.

blackspiritus, 2023-05-30:
I installed the update a few days ago and still got the issue this morning, production was registered at midnight between 12 and 1

hultenvp, 2023-05-30:
Thanks for letting know the issue is still there. It's not tied to the update, by the way, just me not being able to follow up in the past weeks.

Just to summarize (please correct if wrong)
* It appears to be an issue for people living in GMT timezone
* Issue appears on Soliscloud, not m.ginlong.com
* HA, SolisCloud and the actual inverter settings are all GMT, not UTC
* The issue started to occur after start of daylight saving time
* Symptoms: 
  * Energy today toggling between '0' and actual values between 23:00 and midnight
  * @eaasnes reports the energy today value to be alternatingly right or the total value of both inverters. Unclear if this is the same issue
  * The initial reporting of wrong values throughout the day appear to be the result of maintenance issues and has resolved itself

I'd love to get my hands on raw values for energy today.
If you're willing to help, could you please try the following change line 158 in service.py from 
```if attribute == INVERTER_ENERGY_TODAY or attribute == INVERTER_ACPOWER:```
to:
```
if attribute == INVERTER_ENERGY_TODAY or attribute == INVERTER_ACPOWER:
    if attribute == INVERTER_ENERGY_TODAY: _LOGGER.debug("Server value EnergyToday: %s", value)
```

And run during the problem hours with debug on? (You'll need to restart HA after the code changes)

curry684, 2023-05-30:
I am in the Netherlands and my inverter, Soliscloud and Home Assistant server are all set to CET with DST enabled, therefore currently CEST.

[image]

So yes, it is suspicious that the issues only seem to occur in the final hour before midnight as that is the hour in which UTC/GMT are a different date than my entire local setup. It may currently be the final 2 hours before midnight but I have no anecdotal evidence to support that (seen it occur at times between 23:10 and 23:45 CEST).

bluemoon83, 2023-06-16:
I had this issue on a couple of days in April but nothing until last week or so, since then its happened on 4 days out of 8, always between 2300-0000 and the value is always what my daily production was for that day.  Again, my time settings are UTC +0000 and times appear correct.  
[image]
[image]

cafedelmar01, 2023-06-16:
Same here! Had the same idea to open a issue, keep up the good work!

curry684, 2023-06-19:
Same, multiple corrections needed over the last week. Would even dare to say the same days as @bluemoon83 above.

hultenvp, 2023-06-20:
I'm having the issue myself as well. Messing up the production totals, but the energy dash for some reason does compensate for the daily totals.

Please confirm you see the same as I do in the history (screenshot 1: 2 dips on June 6, adding 2x 10.1 kWh to the solar totals for the day.)
This is SolisCloud messing up the data. I might be able to filter it out, but the design is inherently stateless, so restarting HA when the SolisCloud is misbehaving is something I probably will not be able to fix.

Please understand I don't have a lot of time at the moment, so can take a while to fix it. Help is welcome.

[image]

[image]

ilbarone87, 2023-06-20:
I can confirm is exactly the same behaviour, mostly happens at the end of the day but have seen that happening also in other random frame times.

hultenvp, 2023-07-12:
This issue should now be fixed as part of #296. Please reopen if still occurring

closed 2023-07-12 by hultenvp