hultenvp/solis-sensor #274: At night I get huge incorrect readings.

PieterVandenBerge, 2023-04-13:
For some strange reason I got huge readings at night today.
Not sure why and I'm also not sure if this data can in fact be deleted?

[image]

Soliscloud is showing correct data:
[image]

What should I do?

PlutonicCloud, 2023-04-13:
Experienced the same last night: suddenly 10kw solar energy was produced multiple times throughout the night. Changed the incorrect data in the energy dashboard using the explanation from this page: https://github.com/macxq/foxess-ha/wiki/How-to-fix-Energy-Dashboard-data-(statistic-data).

PieterVandenBerge, 2023-04-14:
Thanks for the feedback.
Very good guide on how to fix Energy Dashboard data!

ilbarone87, 2023-04-15:
> Experienced the same last night: suddenly 10kw solar energy was produced multiple times throughout the night. Changed the incorrect data in the energy dashboard using the explanation from this page: https://github.com/macxq/foxess-ha/wiki/How-to-fix-Energy-Dashboard-data-(statistic-data).

Would this fix the issue i have posted in the link below?
https://github.com/hultenvp/solis-sensor/discussions/271

hultenvp, 2023-04-17:
If the time and locale (timezone) of the inverter itself, soliscloud and your home assistant are all correct then there cannot be any overflow of energy from the previous day.
If you have excluded this as an potential root cause then it's soliscloud providing wrong data. The integration simply provides what soliscloud API offers.
Not much I can do about it.

curry684, 2023-04-25:
I have had a similar issue occur from 20th to 23rd of April - on every day at some point between 23:00 and 00:00 I would get a spike identical to either *exactly* the day's total production, or once also exactly *twice* that number. Within the hour the minutes at which it occurred seemed random.

Didn't happen yesterday so for now assuming it's a fluke, but it's something to look into if it keeps cropping up.

viking2010, 2023-04-25:
@curry684 That sounds more like this issue: https://github.com/hultenvp/solis-sensor/issues/273

marcselis, 2023-05-01:
Problem seems to be solved by replacing the "Solis Solar Today" sensor by the "Solis Solar Total" sensor.

curry684, 2023-05-02:
"Total" doesn't count fractions so is pretty much useless at the day or even week level.

hultenvp, 2023-07-12:
This issue should now be fixed as part of #296 . Please reopen if still occurring

closed 2023-07-12 by hultenvp