hultenvp/solis-sensor #276: FYI: Solis Home/Pro will be shut down on 30th April 2023

bartgabriels, 2023-04-14:
Dear User,
Solis Home/Pro will be shut down on 30th April 2023. Please download the “SolisCloud” APP or log in to the web version of SolisCloud “[www.soliscloud.com](http://www.soliscloud.com/)” as soon as possible. On the main page of SolisCloud, you can directly log in with the original Solis Home/Pro account password, without any other operation. It is expected that the migration will be completed automatically within 2 hours. If there are still issues after the migration is completed, please contact us at ：https://www.ginlong.com/global/contactus.html.
If you already migrated your account and are using SolisCloud, please ignore this email. Thank you for your understanding and support.

snigehere, 2023-04-17:
I have a slightly different message but same reason .... I am now waiting my API approval so I can reinstall the integration.

Dear Users,
The platform “Solis Home” https://m.ginlong.com and “Solis Pro” http://pv.ginlong.com will shut down on April 30, 2023. 

We respectfully request that you download the "SolisCloud" app or log in to www.SolisCloud.com as soon as possible to migrate data to the new generation monitoring platform.

BENEFITS:
The new monitoring platform is a major upgrade from the previous monitoring platform.  The response time is faster, the data is more secure and overall platform is more stable.  Furthermore the platform boast features like remote control features, advanced reporting, organization management and soon we will bring service management.  

INSTRUCTIONS:
Video Instruction: https://youtu.be/FU0YywC5--4
If you have issues, please contact your local service support. 
https://www.ginlong.com/global/contactus.html

How to Migrate toSoliscloud.
https://oss.soliscloud.com/Solis%20Home%26Pro%20Shutdown%20Announcement%20-%20How%20to%20Migrate%20toSoliscloud.pdf

This is a system email. Please do not reply to this email. Thank you.

nefh, 2023-05-06:
I've just received the mail (06.05.2023...) and soliscloud do not recognize my plant from m.ginlong.com and suggest creation of completely new account. However, I can still log in onto m.ginlong.com and my plant is still visible there.

snigehere, 2023-05-06:
When migration started you just logged onto solis cloud with the same credentials and the inverter was moved over but as its past the close down date I wonder if this is still being done... If you have not done so I would open a ticket with solis https://solis-service.solisinverters.com/ (you will need to register for this as its a separate system) and ask them to solve it .. if you just create a new account you will loose all your history

Blokkert, 2023-05-31:
Today it is official removed! My ginlong connection is dead now :)

bartgabriels, 2023-05-31:
Same here … works fine on Soliscloud

hultenvp, 2024-01-05:
Left notification open for a while. Guess everyone should know by now ;-)

closed 2024-01-05 by hultenvp