hultenvp/solis-sensor #277: No inverters found

bartgabriels, 2023-04-15:
I just migrated from m.ginlong.com to SolisCloud. Migration went well, API Access requested (within 10 minutes approved) but integration with HA is not working. 

Logger: custom_components.solis.soliscloud_api
Source: custom_components/solis/soliscloud_api.py:211 
Integration: Solis Inverter ([documentation](https://github.com/hultenvp/solis-sensor/), [issues](https://github.com/hultenvp/solis-sensor/issues)) 
First occurred: 08:27:12 (2 occurrences) 
Last logged: 08:27:46

No inverters found

2023-04-15 08:55:15.270 DEBUG (MainThread) [custom_components.solis.ginlong_api] Fetching data for serial: 000370017107302
2023-04-15 08:55:16.265 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 0:02:23.774206, at 2023-04-15 06:57:40.039427+00:00
2023-04-15 08:55:33.882 DEBUG (MainThread) [custom_components.solis.soliscloud_api] workarounds: {'correct_daily_on_grid_energy_enabled': False, 'use_energy_today_from_plant': False}
2023-04-15 08:55:34.313 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found

**Screenshots**
If applicable, add screenshots to help explain your problem.

michaelsargent, 2023-04-16:
I did the same last month.  You can keep using the ginlong (old portal) for HA.  I updated my password in the new soliscloud.com portal and it still updates a month after.  So I can use the old app on my iphone or the new soliscloud app.

bartgabriels, 2023-04-16:
Can I? Support of the ginlong portal stops 30/4 …

michaelsargent, 2023-04-16:
Damn, so I have to find a solution too.  Mind you, they may not support it, but I am hoping that it keeps running though. I'll make a copy of my HA instance and swap over to soliscloud and see what happens.  What version of solis-sensor are you on?  Can you log on to soliscloud.com portal through a browser?  Did you update the portal_domain and portal_plant_id?

hultenvp, 2023-04-17:
Hi @bartgabriels , @michaelsargent 

Please follow the wiki: https://github.com/hultenvp/solis-sensor/wiki/Soliscloud-issues#testing-issues
And let me know the results here.

Pretty sure it is (again) a backend issue.

My workload currently unfortunately doesn't allow me to be so active on GitHub, so expect a delay in my response.

curry684, 2023-04-18:
Mental note: Solis portal mentions `https://www.soliscloud.com:13333/` as the API URL. If you copy/paste this into HA integration *including trailing slash* it gives the `No inverters found` error. Removing the trailing slash made it work for me right away.

Disclaimer: didn't double check, it may have been API flukes on their end, but maintainer can check this easier than me.

bartgabriels, 2023-04-18:
This fixed it .... Thanks

hultenvp, 2023-04-21:
Try the test script @Blokkert

dpgh947, 2023-04-21:
I converted to soliscloud a few days ago and it has been working fine, but today I have been getting loads of "no inverters found" errors logged, over 100 today. In between it is still working fine. I wonder if lots of people have been prompted (like I was) to convert from ginlong to soliscloud by the recent email saying that the "old" ginlong interface is being withdrawn, and the new soliscloud can't cope with the increasing load?

LucidityCrash, 2023-04-23:
I've been seening this and tried running the test code just now ... looks like the inverterslist api endpoint is returning a 502.

dpgh947, 2023-04-25:
I've tried the test code (I downloaded the whole zip file) and I get loads of (for example)
```
Failed with: API returned an error: 0, message='Attempt to decode JSON with unexpected mimetype: text/html', url=URL('https://www.soliscloud.com:13333/v1/api/stationAll'), error code: None, response None
```
I can't get a good response out of it at all.

I am getting hundreds of "no inverters found" logged per day now, it seems that MOST requests fail but every now and then it gets through, because the entities in HA do seem to be getting updated mostly every 10-20 minutes. This seemed to start for me around 7:20am on Saturday. It had been fine before that (I only converted to soliscloud a few days before Saturday)

HamsenBender, 2023-04-26:
Have been trying to switch to SolisCloud for some weeks now, but I always see the same error in the log:
"No inverters found"

Any ideas?

viking2010, 2023-04-27:
I can confirm the above @dpgh947.  Solis seem to have fixed their API issues now. Hopefully that will fix all these issues people have had with getting setup.

HamsenBender, 2023-04-27:
Indeed, the test script is now working while it was still failing all steps yesterday.
I still get the same error however in HA when setting up the integration, "Cannot login with provided URL and credentials" and "no inverters found" in the logfile

LucidityCrash, 2023-04-27:
That would suggest you've not got things configured correctly ... I'm going to assume that since the test script works you have the username, key, and secret correct ... exactly what URL have you entered ?  It pretty much HAS to be exactly : 
https://www.soliscloud.com:13333

Note NO trailing / 
and select the soliscloud radio button

HamsenBender, 2023-04-27:
Pretty sure I used the correct settings, since the test script was working.
But I think I finally found the cause, I noticed in the Soliscloud portal, the plant ID was empty. 
(The ID in grey on top, after the plant name)
I changed the plant name to something new, and after that an ID was generated for the plant.
Few minutes after that, I am now able to add the integration.
And after adding the integration, it mentions the ID which was previously empty.
So I'm pretty sure the issue was related to the Plant not having an ID

nheetesonne, 2023-04-27:
> Pretty sure I used the correct settings, since the test script was working.
> But I think I finally found the cause, I noticed in the Soliscloud portal, the plant ID was empty.
> (The ID in grey on top, [...]

The plant id in the overview page (https://soliscloud.com/#/station/stationdetail_1?id=xxxxxxxxxxx) was empty for me too.
After changing the name of the installation (Choose "change information" on the top right of the overview page) a plantid was generated. After that the intergration worked immediately. The error I got before was "no inverters found"

dpgh947, 2023-04-27:
I have had 6 errors logged through today, as opposed to several hundred per day at its worst, so massively improved, but still a slight intermittent problem, although not noticeable unless you look for it.

curry684, 2023-04-30:
For the record the official SolisCloud app is also frequently showing HTTP 503 and other errors. Their service is currently hardly stable and it's unlikely that one can make a really stable integration with it yet.

bartgabriels, 2023-04-30:
Works like a charm here now …

atryda22, 2023-05-11:
> That would suggest you've not got things configured correctly ... I'm going to assume that since the test script works you have the username, key, secret, and plant ID correct ... exactly what URL have you entered ? It pretty much HAS to be exactly : https://www.soliscloud.com:13333
> 
> Note NO trailing / and select the soliscloud radio button

I had the same problem with unnecessary / after port number
I think some info in the manual would be useful.
I was also not sure which field to enter what value because they are not described and the order is different in the manual.
In my experience, the order is as follows:
- our ID (some suggest that it may be the ID of the inverter)
- secret
- key id
- station id

Regards

closed 2023-05-26 by hultenvp