hultenvp/solis-sensor #278: Is the intention to add control via the /v2/api/control endpoints?

stevegal, 2023-04-17:
I've asked solis for the control api documentation and they sent me a pdf of the control api.

Would this be something that would be considered for support in the future?

viking2010, 2023-04-22:
I didn't know there was a control api document. Would it be possible to share it?

stevegal, 2023-04-28:
I'm not sure I'm allowed to share it publically as it says "classification: Secret". Much of the security seems to be the same as you have in this repo (the hashed content header etc). Interesingly the password it requires in the v2/api/login {"userInfo":"username", "passWord": "md5hexdigestofpassword"} where the UserInfo is your username (or email?) and teh passWord is the md5 hex digest of the password.

So far I've got it to login and give me a token (the login gives a lot more than just the token), I'm just not sure where to add the token as section 3.2 just says "need to use Token".

I'll try to just add a "Token" header and see if that works 🤷

stevegal, 2023-04-29:
> I didn't know there was a control api document. Would it be possible to share it?

As I said, not sure if I'm allowed. However I just raised a request and asked for details of how to control via the api and they sent me the document in a few days.

stevegal, 2023-05-10:
got a working example...

https://github.com/stevegal/solis_control/blob/main/solis_control.py

Note this this quite brittle and is more of a WIP at this stage! (note must have 3 time slots in the config for example!)

there are a lot more control "cid" available as well, but being able to control the charge & discharge times is a good start.

hultenvp, 2023-05-26:
Nice! I've been thinking what entities to use if this is to be integrated with home assistant. I need to look deeper into it, but I guess it should be a derivative of a base entity. Then it would also need its own custom card or so to be able to control

cafedelmar01, 2023-06-03:
> uld also need its own custom

> got a working example...
>
> https://github.com/stevegal/solis_control/blob/main/solis_control.py
>
> Note this this quite brittle and is more of a WIP at this stage! (note must have 3 time slots in the [...]

Mind Sharing how to get this working? (Looking for an solution to disable and enable powerexport remote by HA)

stevegal, 2023-06-10:
> Mind Sharing how to get this working? (Looking for an solution to disable and enable powerexport remote by HA)

Sure, you need to install pyscript HACS plugin first, then just paste the service code into the a file called solis_control.py in the pyscript directory and you should be ready to add it into your automations... or just it as a base for other scripts to call.

https://github.com/stevegal/solis_control readme has the info you need. The plant id etc can be found in the same way as in awesome plugin, but it needs your username and password in addition to api keys. I'm not sure it needs turning on from solis for you. I asked if tehy had an api where I could change times and they responded with some documentation, which I got working without further contact with them, so I'm not sure if it gets turned on when you ask for your API keys (this would be my guess)

cafedelmar01, 2023-06-10:
> Sure, you need to install pyscript HACS plugin first, then just paste the service code into the a file called solis_control.py in the pyscript directory and you should be ready to add it into your [...]

That would be awesome! Is it on some how possible to retrieve a list of push/pull api index? The is i would love to control remotely is the "backflow/export grid setting"

stevegal, 2023-09-02:
The api that I was using seems to have changed. There is, however, a new tikcet type of `End User Control Access` now available on the solis cloud platform. I've raised a ticket with them and I'm waiting a response.

TGRolando, 2023-09-30:
I don't suppose you have had any further luck with this.
I've been in contact and signed their NDA. I've got the pdf document but haven't got any API keys yet.
I'm also raising a ticket for "End User Control Access"

RutgerBeyen, 2023-11-09:
I'm very interested in this functionality as well.
Seems Solis did deactivate my API access after requesting the "end user control access", so waiting for all access to be working again before I can test anything.
I'd be happy to assist when working on this feature

stevegal, 2024-01-16:
the control in the linked repo seems to be working fine. There was a bit of instability as solis rolled out their (unpublished) next version that they use in their web app, but the v2 control functionality seems to be stable.

graemejross, 2024-05-23:
Hi @stevegal - I requested remote access, signed the nda, installed pyscript HACs plugin, installed solis_control.py and setup a script to call the service with my credentials.  When I run the script, no errors are generated but there is also no effect to my inverter settings.  Do you know if this is still working?  Do you have any tips on debugging, for example how can I tell if the api has been called and if any error messages were issued?  Thanks for the good work!

steveworkman, 2024-07-08:
@stevegal I asked Solis for the API this morning and I was not asked to sign an NDA. The docs are on their OSS server.

- API https://oss.soliscloud.com/doc/SolisCloud%20Device%20Control%20API%20V2.0.pdf
- List of commands https://oss.soliscloud.com/doc/SolisCloud_control_api_command_list.xlsx

ausloki, 2024-07-08:
What email pls. is only think stopping me from using it 😊
Refuse to sign an NDA

steveworkman, 2024-07-08:
I contacted Solis support through their official channels and the above was what I got

calentadorsolarpasoapaso, 2024-09-22:
I got it working today, maybe the API changed recently and stevegal branch was not working to me. I´ll create a branch with my changes.

fboundy, 2024-12-07:
I've added this via PR #398

jmason, 2025-03-26:
Given that #398 is merged, I would say this could be closed?

jmason, 2025-05-08:
Going to close this

closed 2025-05-08 by jmason