hultenvp/solis-sensor #279: Solis Daily Grid Energy Used shows no values

jseppeli, 2023-04-20:
Setting up Home assistant Energy dashboard but "Solis Daily Grid Energy Used" does not show any values

Have set this as Grid consumption and "Solis Daily On-grid Energy" as Return to Grid

hultenvp, 2023-04-21:
Does the sensor report any values at all? Not all inverters report this data. AFAIK you need an inverter with battery. Mine also does not and only reports 0.0 kWh.

jseppeli, 2023-04-21:
These does not show any values at all.
- Solis Daily Grid Energy Purchased
- Solis Daily Grid Energy Used
- Solis Daily On-grid Energy
- Solis Monthly Grid Energy Purchased
- Solis Yearly Grid Energy Purchased

In Solis cloud there is values for these that would be the most interesting to have. The monthly and yearly is not in my opinion that important as HA however show these calulated when the timeline is selected.
- Today from grid / Today energy imported
- Today to grid / Today to grid

Aren't these fetchable by the Solis API or does Solis have differencies in what can be fetched by the API and what is shown in Solis cloud?

hultenvp, 2023-05-26:
Hi, as mentioned in another thread I'm in the middle of a home renovation, so apologies for any delay in response.

I fetch the following data via the API:
```
        INVERTER_SERIAL:                  ['sn', str, None],
        INVERTER_PLANT_ID:                ['stationId', str, None],
        INVERTER_DEVICE_ID:               ['id', str, None],
        INVERTER_DATALOGGER_SERIAL:       ['collectorId', str, None],
        # Timestamp of measurement
        INVERTER_TIMESTAMP_UPDATE:        ['dataTimestamp', int, None],
        INVERTER_STATE:                   ['state', int, None],
        INVERTER_TEMPERATURE:             ['inverterTemperature', float, 1],
        INVERTER_POWER_STATE:             ['currentState', int, None],
        INVERTER_ACPOWER:                 ['pac', float, 3],
        INVERTER_ACPOWER_STR:             ['pacStr', str, None],
        INVERTER_ACFREQUENCY:             ['fac', float, 2],
        INVERTER_ENERGY_TODAY:            ['eToday', float, 2], # Default
        INVERTER_ENERGY_THIS_MONTH:       ['eMonth', float, 2],
        INVERTER_ENERGY_THIS_MONTH_STR:   ['eMonthStr', str, None],
        INVERTER_ENERGY_THIS_YEAR:        ['eYear', float, 2],
        INVERTER_ENERGY_THIS_YEAR_STR:    ['eYearStr', str, None],
        INVERTER_ENERGY_TOTAL_LIFE:       ['eTotal', float, 2],
        INVERTER_ENERGY_TOTAL_LIFE_STR:   ['eTotalStr', str, None],
        STRING_COUNT:                     ['dcInputtype', int, None],
        STRING1_VOLTAGE:                  ['uPv1', float, 2],
        STRING2_VOLTAGE:                  ['uPv2', float, 2],
        STRING3_VOLTAGE:                  ['uPv3', float, 2],
        STRING4_VOLTAGE:                  ['uPv4', float, 2],
        STRING1_CURRENT:                  ['iPv1', float, 2],
        STRING2_CURRENT:                  ['iPv2', float, 2],
        STRING3_CURRENT:                  ['iPv3', float, 2],
        STRING4_CURRENT:                  ['iPv4', float, 2],
        STRING1_POWER:                    ['pow1', float, 2], # Undocumented
        STRING2_POWER:                    ['pow2', float, 2], # Undocumented
        STRING3_POWER:                    ['pow3', float, 2], # Undocumented
        STRING4_POWER:                    ['pow4', float, 2], # Undocumented
        PHASE1_VOLTAGE:                   ['uAc1', float, 2],
        PHASE2_VOLTAGE:                   ['uAc2', float, 2],
        PHASE3_VOLTAGE:                   ['uAc3', float, 2],
        PHASE1_CURRENT:                   ['iAc1', float, 2],
        PHASE2_CURRENT:                   ['iAc2', float, 2],
        PHASE3_CURRENT:                   ['iAc3', float, 2],
        BAT_POWER:                        ['batteryPower', float, 3],
        BAT_POWER_STR:                    ['batteryPowerStr', str, None],
        BAT_REMAINING_CAPACITY:           ['batteryCapacitySoc', float, 2],
        BAT_STATE_OF_HEALTH:              ['batteryHealthSoh', float, 2],
        BAT_CURRENT:                      ['storageBatteryCurrent', float, 2],
        BAT_CURRENT_STR:                  ['storageBatteryCurrentStr', str, None],
        BAT_VOLTAGE:                      ['storageBatteryVoltage', float, 2],
        BAT_VOLTAGE_STR:                  ['storageBatteryVoltageStr', str, None],
        BAT_TOTAL_ENERGY_CHARGED:         ['batteryTotalChargeEnergy', float, 3],
        BAT_TOTAL_ENERGY_CHARGED_STR:     ['batteryTotalChargeEnergyStr', str, None],
        BAT_TOTAL_ENERGY_DISCHARGED:      ['batteryTotalDischargeEnergy', float, 3],
        BAT_TOTAL_ENERGY_DISCHARGED_STR:  ['batteryTotalDischargeEnergyStr', str, None],
        BAT_DAILY_ENERGY_CHARGED:         ['batteryTodayChargeEnergy', float, 2],
        BAT_DAILY_ENERGY_DISCHARGED:      ['batteryTodayDischargeEnergy', float, 2],
        GRID_DAILY_ON_GRID_ENERGY:        ['gridSellTodayEnergy', float, 2],
        GRID_DAILY_ON_GRID_ENERGY_STR:    ['gridSellTodayEnergyStr', str, None],
        GRID_DAILY_ENERGY_PURCHASED:      ['gridPurchasedTodayEnergy', float, 2],
        GRID_DAILY_ENERGY_USED:           ['homeLoadTodayEnergy', float, 2],
        GRID_MONTHLY_ENERGY_PURCHASED:    ['gridPurchasedMonthEnergy', float, 2],
        GRID_YEARLY_ENERGY_PURCHASED:     ['gridPurchasedYearEnergy', float, 2],
        GRID_TOTAL_ENERGY_PURCHASED:      ['gridPurchasedTotalEnergy', float, 2],
        GRID_TOTAL_ENERGY_PURCHASED_STR:  ['gridPurchasedTotalEnergyStr', str, None],
        GRID_TOTAL_ON_GRID_ENERGY:        ['gridSellTotalEnergy', float, 2],
        GRID_TOTAL_ON_GRID_ENERGY_STR:    ['gridSellTotalEnergyStr', str, None],
        GRID_TOTAL_POWER:                 ['psum', float, 3],
        GRID_TOTAL_POWER_STR:             ['psumStr', str, None],
        GRID_TOTAL_CONSUMPTION_POWER:     ['familyLoadPower', float, 3],
        GRID_TOTAL_CONSUMPTION_POWER_STR: ['familyLoadPowerStr', str, None],
        GRID_TOTAL_ENERGY_USED:           ['homeLoadTotalEnergy', float, 3],
        GRID_TOTAL_ENERGY_USED_STR:       ['homeLoadTotalEnergyStr', str, None],
        SOC_CHARGING_SET:                 ['socChargingSet', float, 0],
        SOC_DISCHARGE_SET:                ['socDischargeSet', float, 0]
```
So all the requested data _is_ fetched. If the data does not show up then sadly that is another example of the poor stability of the Soliscloud API's.
I don't have the right inverter to test these fields myself (they're always 0 in my case), but if you want to be sure it's a server-side issue, and you know how to edit code, you could uncomment line 279 in soliscloud.api and look at the raw response from the server in the *debug* logs. If it's having a non-zero value there then please drop the response here in the comments and I can have a look.

alceasan, 2023-06-02:
I'm on the same situation, I have lost lots of information with the new integration. I think in the new API, they have moved this information to other endpoint. Right now, almost all data is being recovered from the `userStationList` endpoint, but this endpoint lacks information regarding consumption and on-grid energy.

In the `stationDetail` endpoint there is much more information that's not being recovered, such as:
- dayEnergy - Today Generation
- gridPurchasedDayEnergy - Meter today purchase energy
- gridSellDayEnergy - Meter today sell energy

This is the API reference: [https://oss.soliscloud.com/templet/%5BExternal%5D%20SolisCloud%20Monitoring%20API.pdf](https://oss.soliscloud.com/templet/%5BExternal%5D%20SolisCloud%20Monitoring%20API.pdf)

PR #302 (2023-07-09): 279 solis daily grid energy used shows no values
Closes #279
closed 2023-07-09 by hultenvp