hultenvp/solis-sensor #281: I am seeing "ContentTypeError" errors in the log

snigehere, 2023-04-22:
**Describe the bug**
Since moving from the https://m.ginlong.com/ to https://www.soliscloud.com/ I notice it takes an hour or more before data appears following a restart.  I have turned on debug and checking the log I see a lot of records:
DEBUG (MainThread) [custom_components.solis.soliscloud_api] Error from URI (/v1/api/inverterList) : ContentTypeError(RequestInfo(url=URL('https://www.soliscloud.com:13333/v1/api/inverterList'), method='POST', headers=<CIMultiDictProxy('Host': 'www.soliscloud.com:13333', 'User-Agent': 'HomeAssistant/2023.4.6 aiohttp/3.8.4 Python/3.10', 'Content-MD5': 'Pf6YJAUHEPtLXSHoRiWfrg==', 'Content-Type': 'application/json', 'Date': 'Sat, 22 Apr 2023 14:40:21 GMT', 'Authorization': 'API *******************:**************+******=', 'Accept': '*/*', 'Accept-Encoding': 'gzip, deflate', 'Content-Length': '35')>, real_url=URL('https://www.soliscloud.com:13333/v1/api/inverterList')), (), message='Attempt to decode JSON with unexpected mimetype: text/html', headers=<CIMultiDictProxy('Date': 'Sat, 22 Apr 2023 14:40:23 GMT', 'Content-Type': 'text/html', 'Content-Length': '166', 'Connection': 'keep-alive', 'Via': 'HTTP/1.1 SLB.9')>)
2023-04-22 15:40:24.293 WARNING (MainThread) [custom_components.solis.soliscloud_api] No inverters found
2023-04-22 15:40:24.310 WARNING (MainThread) [custom_components.solis.service] Failed to discover, scheduling retry in 60 seconds.
2023-04-22 15:40:24.313 DEBUG (MainThread) [custom_components.solis.service] Scheduling discovery in 60 seconds.

**Config**
Your anonymized config entry

**Versions**
HA Version an deployment 
HACS version 1.32.1
Integration version v3.3.2

**Debug logs**
Attached
```

[solis cloud log.txt](https://github.com/hultenvp/solis-sensor/files/11301873/solis.cloud.log.txt)

viking2010, 2023-04-22:
This could be related to the API issues that seem to be occurring at the moment. I'm only getting sporadic connectivity to the Solis Cloud API.

LucidityCrash, 2023-04-23:
Looks like the inverterslist api endpoint is returning a 502 at the moment.

viking2010, 2023-04-25:
Could be related to the API errors that are still occurring at the moment - https://github.com/hultenvp/solis-sensor/discussions/280

viking2010, 2023-04-27:
Solis have fixed their API issues now it would seem, so could you try again and see if this has now fixed your problem?  If it has, hopefully this open issue can be closed off.

snigehere, 2023-04-27:
looks like the high failure rate stopped at 05:38 GMT today ... in the 6 hours since then I have only seen 4 errors .. this is typical of the 4 I have seen:

2023-04-27 08:37:07.082 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Error from URI (/v1/api/userStationList) : ContentTypeError(RequestInfo(url=URL('https://www.soliscloud.com:13333/v1/api/userStationList'), method='POST', headers=<CIMultiDictProxy('Host': 'www.soliscloud.com:13333', 'User-Agent': 'HomeAssistant/2023.4.6 aiohttp/3.8.4 Python/3.10', 'Content-MD5': 'xxxxxxx', 'Content-Type': 'application/json', 'Date': 'Thu, 27 Apr 2023 07:37:07 GMT', 'Authorization': 'API xxxxxxxxxx:xxxxxxxxxx+xxxxxxxxxx=', 'Accept': '/', 'Accept-Encoding': 'gzip, deflate', 'Content-Length': '2')>, real_url=URL('https://www.soliscloud.com:13333/v1/api/userStationList')), (), message='Attempt to decode JSON with unexpected mimetype: text/html', headers=<CIMultiDictProxy('Date': 'Thu, 27 Apr 2023 07:37:07 GMT', 'Content-Type': 'text/html', 'Content-Length': '206', 'Connection': 'keep-alive', 'Via': 'HTTP/1.1 SLB.9')>)
2023-04-27 08:37:07.084 INFO (MainThread) [custom_components.solis.soliscloud_api] Unable to fetch details for Station with ID: xxxxxxxxxxxxx
2023-04-27 08:37:07.084 DEBUG (MainThread) [custom_components.solis.soliscloud_api] Using inverterDetail for energy_today
2023-04-27 08:37:07.085 DEBUG (MainThread) [custom_components.solis.service] Scheduling next update in 0:05:00, at 2023-04-27 07:42:07.085305+00:00

I can live with that as a failure rate .. its way better than before.

Closing bug report

closed 2023-04-27 by snigehere