hultenvp/solis-sensor #284: Feature: Self use energy

wawyed, 2023-05-05:
Would it be possible to display self use energy? I believe it should be available in the API?

[image]

wawyed, 2023-05-05:
Nvm that's exposed, not sure if self use is available...

wawyed, 2023-05-05:
I've found oneSelf but not sure (I think it's total?):
```
Name    Type   Description 
oneSelf Double Self-consumption
```

viking2010, 2023-05-05:
With regards to the Self-Use, I spotted a post on the Facebook group for Solis the other day. One of the Solis engineers posted:

 _"You need to ignore Self-Use as at the moment it is not valid. It's supposed to be Solar generation minus export. However, other values of export/import energy (dynamic export, time charging/discharging, force charging) will cause it to be wrong."_

So it sounds like it's best not worrying about the self use value at the moment.

hultenvp, 2023-05-26:
Given @viking2010's answer (thanks!) I'm going to close this issue as not a bug.

closed 2023-05-26 by hultenvp