hultenvp/solis-sensor #285: Multiple inverters on Station showing as a single Device

swests, 2023-05-15:
**Describe the bug**
My setup has two inverters, one hybrid and one to grid.  I successfully added the plant and see 1 device. I was expecting the entities to be shown per inverter (Device).  If I run the test scripts I see you're extracting each logger and inverter for the station.  Is this a bug in the integration or as intended - for all entities to show under a single device?

**Config**
GUI Setup for SolisCloud

**Versions**
HA Version an deployment - HA 2023.4.6, HAOS running as VM
HACS version - 1.32.1
Integration version - v3.3.2

**Debug logs**
Add to configuration.yaml:
```
logger:
  default: warning
  logs:
    custom_components.solis: debug
```
And attach the resulting log to the issue. If yo do not know where to find logs you missed an important step in the install of HACS. ;-)

**Screenshots**
[image]
[image]

swests, 2023-05-22:
@hultenvp  - would you please comment on this.  Many thanks.

hultenvp, 2023-05-26:
Hi, renovating the house, so had no time to work on this project so far. Even didn't have the panels on the roof for a while ;-)

This is probably a bug. Intent is to show each individual inverter as a device under the station.
I do not see any issue in my code that could cause this. Can you add debug output here? Please follow the instructions in the wiki if you don't know how to do it.

swests, 2023-05-30:
Hi.

Good luck with the renovations - I remember that experience.

For debug, I guess you'd like me to remove the integration, and then re-add as I guess the issue is somewhere during the initial setup.  Please confirm.

 Is there a chance that there are some stale artefacts that could do with deletion? And if so how would I find them.

hultenvp, 2023-06-01:
Let's first start with switching debug on and then reloading the integration. It should give me some initial data. Would be great if you could provide me with that info.
Based on that I might have to make a custom version to get more targeted debug output.

hultenvp, 2023-07-10:
I can (finally) reproduce the issue, no solution yet

hultenvp, 2023-07-11:
Fixed

closed 2023-07-11 by hultenvp