hultenvp/solis-sensor #288: Error after upgrade to V3.3.3

RCGrahamWilmott, 2023-05-31:
After upgrading to V3.3.3 via HACS, everything is showing as "Unavailable"

I've tried restarting the integration as well as rebooting the whole host.

The only thing in logs (I have debugging on) is:

`This error originated from a custom integration.

Logger: custom_components.solis.service
Source: custom_components/solis/service.py:116
Integration: Solis Inverter (documentation, issues)
First occurred: 10:22:22 (25 occurrences)
Last logged: 11:42:43

Failed to discover, scheduling retry in 540 seconds.
Failed to discover, scheduling retry in 600 seconds.
Failed to discover, scheduling retry in 660 seconds.
Failed to discover, scheduling retry in 720 seconds.
Failed to discover, scheduling retry in 780 seconds.`

juro11, 2023-05-31:
same here, also I did notice, that since today data obtained via NodeRed from ginlong platform is no longer available

tomkolp, 2023-05-31:
I have the same.  I can't log in to the website or reset my password.

dhoogfr, 2023-05-31:
Upgraded to 3.3.3 this morning (also via HACS) and I'm still able to retrieve the data (soliscloud)
So, its not affecting everyone

Question is: what's the difference between the setups or accounts?

RCGrahamWilmott, 2023-05-31:
@conallob - whats the easiest way to roll back? (Never need to before)

@dhoogfr - good question. I've been using this plugin for a very long time. I started with https://m.ginlong.com/ and config in YAML, and moved to soliscloud and UI config as the plugin advanced. Not sure if you are a new(ish) user?

conallob, 2023-05-31:
Correction, rolling back to v3.3.2 worked for a while, before all sensors report "Unavailable" after a few min.

@RCGrahamWilmott to rollback:

1. Go to the HACS Page in your home assistant instance
2. Select Integrations
3. In the menu for any HACS integration
4. Select Redownload, then use html dropdown to choose an explicit version, usually the one immediately before the suggested release

dhoogfr, 2023-05-31:
@RCGrahamWilmott I started with the YAML config, but always was on soliscloud.
The API key I use was created during the period that you could generate your own key in the Solis portal.
It is no longer visible, but still works.

RCGrahamWilmott, 2023-05-31:
Thanks @conallob - now downgraded to V3.3.2 but still not working.

I wonder if I am somehow still using ginlong.com, and it has finally been turned off. I am trying to find my API key for soliscloud to re-add the integration and see if it works again.

edbanger, 2023-05-31:
@RCGrahamWilmott : I was on a V2.something, and certainly using m.ginglong.com + yaml (unfortunately, I only update when something breaks...)

That stopped working around 04:00 GMT, so it seems like ginglong.com was indeed finally turned off. 

Now I have to wait for my Soliscloud API access to arrive before I can move on....

dhoogfr, 2023-05-31:
Wondering if this has anything to do with it

[image]

aguscaba, 2023-06-01:
> After upgrading to V3.3.3 via HACS, everything is showing as "Unavailable"
>
> I've tried restarting the integration as well as rebooting the whole host.
>
> The only thing in logs (I have debugging on) [...]

+1

aguscaba, 2023-06-01:
I tried by switching from 3.3.2 to 3.3.1 and 3.3.3 but no luck.

Do you think It could be something to do with Platform v2 (I have configured the v2)? probably  the thing got updated to SolisCloud?

aguscaba, 2023-06-01:
Please check this:
https://www.ginlong.com/uploads/file/Solis_Home&Pro_Shutdown_Announcement-How_to_Migrate_to_Soliscloud-01.pdf

RCGrahamWilmott, 2023-06-01:
I noticed that my API key was no longer visible on SolisCloud, so I contacted support, who re-activated the API on my account. However, it gave me the same key ID and secret, and HomeAssistant still shows everything as "Unavailable".

However, completely deleting the integration from HomeAssistant, rebooting, reinstalling the integration, rebooting, setting up again, and a final reboot, means it is now showing data again. Not sure of the exact steps to fix it, but this worked for me...

(Unfortunately, this means the ID of my sensors has changed, but I can deal with that)

hultenvp, 2023-06-01:
> I'm seeing the same issues with v3.3.3. Rolling back to v3.3.2 resolves the issue, so with differential diagnosis, the issue is a recent change in v.3.3.3

That's a different issue than the rest in this issue thread. Are you already on SolisCloud or are you still on Platform v2?. Could you provide debug logging?

edbanger, 2023-06-01:
Just to follow up (Or add to the Noise):

Situation: Was on V2.x using Platform v2, stopped working yesterday at 02:00 UTC
Requested Soliscloud API access, which was granted just now.
Added V3.3.3 integration to HACS just now, using SOLISCLOUD API access, and I'm getting data.

So a "Clean" V3.3.3 install based on Soliscloud access works.

gielie, 2023-06-01:
> I noticed that my API key was no longer visible on SolisCloud, so I contacted support, who re-activated the API on my account. However, it gave me the same key ID and secret, and HomeAssistant still [...]

I followed your handlings but still get an error, (cannot login with provide url and credentials) 
Can’t figure out what is wrong

gielie, 2023-06-01:
I figured it out, thanks to this comment https://github.com/hultenvp/solis-sensor/issues/232#issuecomment-1534410770

in the Soliscloud page there was no number after my id, only —-, in the top left corner. I changed my name and there an id number appeared. Did the whole setup again and bingo. 

It’s working now

conallob, 2023-06-02:
> That's a different issue than the rest in this issue thread. Are you already on SolisCloud or are you still on Platform v2?. Could you provide debug logging?

@hultenvp I thought I was on SolisCloud, but it turns out I was on Platform v2. I've migrated to SolisCloud now and everything is working again. The release of v3.3.3 had unfortunate timing with the maintenance window which decommissioned Platform v2.

Could the [Home Assistant Repairs](https://developers.home-assistant.io/docs/core/platform/repairs/) be used to track actions in future please? From the history here, a number of us were surprised we were still dependent on the v2 platform.

Zeeflyboy, 2023-06-05:
I’m getting all the data as usual except Solis energy today is showing unknown since  I updated last night. Any one else?

hultenvp, 2024-01-05:
Closing due to inactivity

closed 2024-01-05 by hultenvp